sm0svx/svxlink #128: Stereo output

ajorg, 2015-01-21:
The ARS Repeaters USB EchoLink adapter has a VOX circuit that only triggers properly on stereo output. svxlink only supports mono output. On Windows EchoLink it's not a problem, because Windows copies mono output to all channels by default. Linux doesn't do this.

I've worked around the issue by writing an ALSA config that routes channel 0 to both 0 and 1, but it was hard to discover how to do this. It would be easier to work around if svxlink supported audio output to multiple channels / stereo.

sm0svx, 2015-01-21:
Linux is not to blame for this. It's just that SvxLink is hard coded to always open the audio device in stereo mode. It would probably be quite simple to add a new config variable, CARD_CHANNELS, that can be set to 1 to open the audio device in mono mode. That will send the audio to both left and right connector.

ajorg, 2015-01-22:
Sounds good. Do you want to paste me a patch to try?

sm0svx, 2015-01-23:
The CARD_CHANNELS config variable is now available in latest git master.

ajorg, 2015-01-23:
Excellent! I'll give it a try this weekend.

ajorg, 2015-01-24:
I backported the patch to 14.08 and it seems to work nicely there. I'm rebuilding from HEAD now as well.

sm0svx, 2015-01-25:
OK, sounds good. Closing issue.

closed 2015-01-25 by sm0svx