sm0svx/svxlink #130: Raspberry Pi and SVXLINK v14.08

n1pce, 2015-02-11:
Trying to install on my B+. I have tried to contact others that have done it but have received no help. The only reply I did receive told me to talk to this group. 

I would like to use my Raspberry Pi B+ and connect it to my repeater so I can add EchoLink.

I was following the Installation Instructions note for note and the following is what I get:

pi@raspberrypi ~/Downloads/svxlink-14.08/src/build $ cmake ..
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Reading versions file...
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'sigc++-2.0'
--   package 'sigc++-2.0' not found
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:97 (MESSAGE):
  Could NOT find SIGC2 (missing: SIGC2_LIBRARY SIGC2_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:288 (_FPHSA_FAILURE_MESSAGE)
  cmake/Modules/FindSIGC2.cmake:31 (find_package_handle_standard_args)
  CMakeLists.txt:305 (find_package)

-- Configuring incomplete, errors occurred!
pi@raspberrypi ~/Downloads/svxlink-14.08/src/build $

I running the most up to date version of Raspin.

I am new to Linux. So any help will be greatly appreciated.

John - N1PCE

stratfan, 2015-02-11:
Hi John,

I have a Raspberry pi running a repeater so I can help, especially when you get into the gpio config.  Looks like you need to install sigc++  try entering:

sudo apt-get install sigc++

Gene Eighmy
W4EAE

f5vmr, 2015-02-11:
Hi Guys.
I successfully installed the latest version from the svxlink-master from this page, which should always be the most up-to-date. The F5ZGM repeater would normally be on echolink, but we have an internet issue that is currently being worked on. You can look up my Facebook svxlink, and join in for help, despite being in France, I post in English for the most part despite located in france. Regards. Chris F5VMR./ G4NAB (see qrz.com)

n1pce, 2015-02-12:
Hello Gene and Chris.

Thank you for getting back to me.

I will check out the Facebook page Chris. I don't hand out on FB much these days but I will stop by and Join.

Now on with the show...

I used the command line and now have sgc++ installed. :-)

Here is what I have now:

pi@raspberrypi ~/Downloads/svxlink-14.08/src/build $ cmake ..
-- Reading versions file...
-- Found components for RT
-- RT_INCLUDES = /usr/include
-- RT_LIBRARIES     = /usr/lib/arm-linux-gnueabihf/librt.so
-- Could NOT find SPEEX (missing:  Speex_LIBRARY Speex_INCLUDE_DIR) 
-- Could NOT find OPUS (missing:  Opus_LIBRARY Opus_INCLUDE_DIR) 
-- checking for module 'alsa'
--   package 'alsa' not found
ALSA_INCLUDE_DIR=ALSA_INCLUDE_DIR-NOTFOUND
ALSA_LIBRARY=ALSA_LIBRARY-NOTFOUND
CMake Error at cmake/Modules/LibFindMacros.cmake:74 (message):
  Required library ALSA NOT FOUND.

  Install the library (dev version) and try again.  If the library is already
  installed, use ccmake to set the missing variables manually.
Call Stack (most recent call first):
  cmake/Modules/FindALSA.cmake:35 (libfind_process)
  async/audio/CMakeLists.txt:77 (find_package)

-- Configuring incomplete, errors occurred!
pi@raspberrypi ~/Downloads/svxlink-14.08/src/build $ sudo apt-get install alsa
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'alsa-base' instead of 'alsa'
alsa-base is already the newest version.
alsa-base set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi ~/Downloads/svxlink-14.08/src/build $ sudo apt-get install speek
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package speek
pi@raspberrypi ~/Downloads/svxlink-14.08/src/build $ sudo apt-get install RT
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package RT
pi@raspberrypi ~/Downloads/svxlink-14.08/src/build $ 

I was trying anything. hahaha... (no idea what RT is)
** I just found the typo of "speek" I used "speex" It says it installed OK. I tried "sudo apt-get install opus" but it says it could not be found. and then ran things and still get the same error messages. 

I do have a USB Sound card plugged in. (one of those cheep ones)
I can confirm that Alsa and Alsa-base are on the system. I found them under "/usr/share"
What do I do now?

Thank you 

John - N1PCE

f5vmr, 2015-02-12:
Hi john, well done so far. On this forum you will find the instructions for install at the location of your choice. There is one section in CAPS whic I think is the preferred option, so that when installed, you should find your configuration file, /etc/svxlink/svxlink.conf. Your sound files then should be in /usr/share/svxlink/sound, and the software module configuration in /usr/share/svxlink/svxlink.d/. I recommend a good read of the man files on here before you proceed. They will also help you understand the configuration of the GPIO ports. We have had someone design a port interface for a repeater with link that works. The Usb audio card work quite well, some more noisy than others.
Keep in touch. Chris.

ch0mik, 2015-02-12:
try this - its works fine in my cubieboard A10 with Cubian (Echolink Server
only)

sudo apt-get update
sudo apt-get install -y g++ make libsigc++-dev sigc++ libgsm1-dev
libpopt-dev tcl8.5-dev libgcrypt-dev libspeex-dev libasound2-dev alsa-utils
build-essential cmake

sudo adduser svxlink
sudo adduser svxlink daemon

cd ~/
git clone https://github.com/sm0svx/svxlink.git
cd svxlink/src
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc
-DLOCAL_STATE_DIR=/var -DUSE_QT=no ..
make NO_CHOWN=1 install
sudo make install
sudo ldconfig

#modify first the /etc/svxlink/svxlink.conf and similar conf files
sudo svxlink --config /etc/svxlink/svxlink.conf

n1pce, 2015-02-13:
ch0mik. All was good through to the "cd build". The next line starting with "cmake"   Gave me these results:

pi@raspberrypi ~/svxlink/src/build $ cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc -DLOCAL_STATE_DIR=/var -DUSE_QT=no .. 
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Reading versions file...
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'sigc++-2.0'
--   found sigc++-2.0, version 2.2.10
-- Found SIGC2: /usr/lib/arm-linux-gnueabihf/libsigc-2.0.so  
-- Found chown: /bin/chown
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found.
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found components for RT
-- RT_INCLUDES = /usr/include
-- RT_LIBRARIES     = /usr/lib/arm-linux-gnueabihf/librt.so
-- Found SPEEX: /usr/lib/arm-linux-gnueabihf/libspeex.so  
-- Could NOT find OPUS (missing:  Opus_LIBRARY Opus_INCLUDE_DIR) 
-- checking for module 'alsa'
--   found alsa, version 1.0.25
-- Found ALSA 1.0.25
-- checking for module 'popt'
--   found popt, version 1.16
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so
-- Looking for HIDIOCGRAWINFO
-- Looking for HIDIOCGRAWINFO - found
-- Found GCrypt: /lib/arm-linux-gnueabihf/libgcrypt.so  
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so
-- Found Tclsh: /usr/bin/tclsh (found version "8.5") 
-- Found TCL: /usr/lib/libtcl8.5.so  
-- Could NOT find TCLTK (missing:  TK_LIBRARY TK_INCLUDE_PATH) 
-- Could NOT find TK (missing:  TK_LIBRARY TK_INCLUDE_PATH) 
-- Found components for DL
-- DL_INCLUDES = /usr/include
-- DL_LIBRARIES     = /usr/lib/arm-linux-gnueabihf/libdl.so
-- Could NOT find TCLTK (missing:  TK_LIBRARY TK_INCLUDE_PATH) 
-- Could NOT find TK (missing:  TK_LIBRARY TK_INCLUDE_PATH) 
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Found gzip: /bin/gzip
-- Found groff: /usr/bin/groff
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/svxlink/src/build
pi@raspberrypi ~/svxlink/src/build $ 

I went ahead and did the "make NO_CHOWN=1 install". It took about 40min to do. I didn't see anything wrong until just after 100% This is what showed up:

Install the project...
-- Install configuration: "Release"
-- Installing: /usr/lib/libasynccore.so.1.3.99.3
CMake Error at async/core/cmake_install.cmake:56 (FILE):
  file INSTALL cannot copy file
  "/home/pi/svxlink/src/build/lib/libasynccore.so.1.3.99.3" to
  "/usr/lib/libasynccore.so.1.3.99.3".
Call Stack (most recent call first):
  async/cmake_install.cmake:37 (INCLUDE)
  cmake_install.cmake:37 (INCLUDE)

Makefile:62: recipe for target 'install' failed
make: *** [install] Error 1
pi@raspberrypi ~/svxlink/src/build $ 

Any ideas?

Thank you

John - N1PCE

n1pce, 2015-02-13:
Gene - W4EAE

Would it be possible to chat via email? The email on QRZ.com doesn't work. Mine is current and correct>

Thank you

John - N1PCE

stratfan, 2015-02-13:
Updated my email address on QRZ and uploaded a photo of my first version of svxlink hardware which was mostly alligator clip jumpers with a few pots and IC's thrown in.

Gene Eighmy
W4EAE

sm0svx, 2015-02-21:
Some additional information:
* The installation documentation for [Debian install](https://github.com/sm0svx/svxlink/wiki/InstallSrcDebian) should also work for RPi if you're running Rasbian or some other Debian based distro. It's not completely up-to-date but may work.
* NO_CHOWN=1 does not do anything. It was a thing in the old build system.
* The primary way to get help is to register with the svxlink-devel mailing list. More information on the [Getting Support page](https://github.com/sm0svx/svxlink/wiki/GettingSupport).

sm0svx, 2015-04-26:
Closing this issue since it's neither a bug report nor a feature request.

closed 2015-04-26 by sm0svx