sm0svx/svxlink #131: IRLP board support

vk3jed, 2015-02-18:
The IRLP board is a simple parallel port interface to connect a radio to a PC.  Some hams have given up on IRLP, and have a board kicking around.  The interface is also simple to homebrew, which gives another means to interface a PC to a radio.  The IRLP board supports COS, PTT, hardware DTMF decoding (all digits except D), and 3 user programmable auxiliary outputs, which can be used to control external devices.  In my case, I eventually hope to have SvxLink control my main transmitter and receiver directly.  My setup uses an IRLP board, due to its heritage.  While it is possible to swap interfaces, that could cause issues with some of the other software components, which expect to talk to this style of interface.  I am currently using thelinkbox to control the radios, but want to make use of SvxLink's remotetrx and voting features.

There is also an equivalent setup for the Raspberry Pi, which uses a number of GPIO pins instead of a parallel port.  I haven't played with this variant...  Yet. :)

sm0svx, 2015-02-21:
* *Problem 1*: I don't own an IRLP board, so I cannot test the code properly.
* *Problem 2*: I don't even own a computer that have a parallel port anymore. Just recently thew away the last one.
* *Problem 3*: The IRLP board cannot decode the D digit as you say. This make it impossible to use macros.

That being said, I would accept a patch implementing support for the IRLP board. I'd require using ppdev for parallel port access so that SvxLink does not have to be run as root. The implementation would probably involve a singleton class that talk to the parallel port so that access can be shared between different parts of SvxLink, unless ppdev support multi open. Then three classes would have to be implemented: PttIrlp, SquelchIrlp and IrlpDtmfDecoder.

Another approarch is to implement an external utility that talk to the parallel port and then use PTY:s to communicate to SvxLink (e.g. SQL_DET=PTY, DTMF_DEC_TYPE=PTY and PTT_TYPE=PTY).

vk3jed, 2015-02-21:
On 21/02/2015 8:38 PM, Tobias Blomberg wrote:
> * /Problem 1/: I don't own an IRLP board, so I cannot test the code
>     properly.
>   * /Problem 2/: I don't even own a computer that have a parallel port
>     anymore. Just recently thew away the last [...]
Both IRLP and thelinkbox are able to do it without being run as root.  
Every IRLP node is proof of the former, and my own system is proof of 
the latter. :)  That said, your issues certainly are significant.  
Unfortunately, I don't write code myself, otherwise I'd have already 
started working on the said patch. :)
>
> Another approarch is to implement an external utility that talk to the 
> parallel port and then use PTY:s to communicate to SvxLink (e.g. 
> SQL_DET=PTY, DTMF_DEC_TYPE=PTY and PTT_TYPE=PTY).
>
That method would make sense too (though I still have the same issue as 
above re code :) ).

-- 
73 de Tony VK3JED/VK3IRL
http://vkradio.com

sm0svx, 2015-02-21:
Using ppdev make it possible to run an application accessing the parallel port as a non-root user. That is why I mentioned it. Maybe that is what IRLP is using. There may be other methods but the only other method I know of is using direct hardware IO port access through the iopl, ioperm, inb, outb etc system calls but they require running the application as user root.

vk3jed, 2015-02-22:
On 21/02/2015 9:42 PM, Tobias Blomberg wrote:
> Using ppdev make it possible to run an application accessing the
> parallel port as a non-root user. That is why I mentioned it. Maybe
> that is what IRLP is using. There may be other methods but the [...]
Could be, IRLP requires the /dev/parport device to work.

-- 
73 de Tony VK3JED/VK3IRL
http://vkradio.com