sm0svx/svxlink #132: Enhanced SQL_DET with logical OR/AND

pe1chl, 2015-02-25:
Our Tait TB8100 repeater hardware has an interesting configuration capability:
When selecting the receiver opening condition, one cannot only select a single
source (CTCSS, RSSI, SINAD) but these sources can be combined in an AND/OR network.

We are running our 2 meter repeater with the config: CTCSS present OR RSSI above -100dBm.
This means that weak out of area stations (e.g. working on a different repeater) will not open
the squelch when they operate a different CTCSS tone, but stronger fixed stations closer to the
repeater can use it even when they have no CTCSS transmit capability.

Now that we are deploying receivers of different types, that do not have this feature, we would
like to configure it in SvxLink.

E.g. the value of SQL_DET could be a logical expression like SERIAL|CTCSS or SIGLEV|CTCSS.

I checked the source and it does not appear to be an easy change.  But I am not fluent in the way
the code is written, maybe it would be possible to have a "logic" sql detector that can be "pushed"
on top of a number of existing detectors.

gustavc39, 2015-02-25:
Hi Rob,

   I think that svxlink already implements some sort of squelch 
combining.  Maybe not exactly what you want.  Inside a RepeaterLogic 
section you can enable different options to open the repeater at the 
same time:

OPEN_ON_1750=1000
OPEN_ON_CTCSS=136:2000
OPEN_ON_DTMF=*
OPEN_ON_SQL=5000
OPEN_ON_SEL5=01234
OPEN_SQL_FLANK=OPEN
OPEN_ON_SQL_AFTER_RPT_CLOSE=10

The functionality that you need can possibly be achieved by using 
SQL_TYPE=SIGLEV at the Rx section, and in the RepeaterLogic combining 
OPEN_ON_SQL and OPEN_ON_CTCSS.

73!

Gustavo

pe1chl, 2015-02-25:
Ok that is for opening the repeater, it is indeed more versatile.
What I describe is to be working for the squelch during a QSO as well.
So, when someone releases the mike and that that moment a signal is received without CTCSS
and a weak signal level, the squelch closes and the roger beep sounds.  But when a strong
signal OR a signal with CTCSS is present, the squelch is open.  Especially this combo appears
to work well when lots of users have equipment without CTCSS but the receivers are placed
on high locations and receive signals from outside the area.   When it is too much work to
implement a completely flexible solution, a new detector that is hardwired to open on
SIGLEV | CTCSS would be sufficient in our use case, as SIGLEV provides the method to open
only on relatively strong signals.

sm0svx, 2015-02-25:
I think the best and most straight forward way to implement this is to implement a SigLevDetLogicOr class. I think that implementing something more general, like a full logic expression parser, is overkill.

The logic or case seem perfectly valid. What about logic and?  When would this make sense?  A mix with logic and/or?  When would that make sense?

pe1chl, 2015-02-25:
Well, in the case of the Tait TB8100 one can configure things like:
CTCSS OR (RSSI > -120dBm AND SINAD > 12dB)
In SvxLink that could translate to:
CTCSS OR (SERIAL AND SIGLEV)
where SERIAL is wired to a RSSI threshold indicator, and CTCSS and SIGLEV (similar to SINAD)
are evaluated in software.
However, this is overkill in most cases, and in the use case I described it would only need the OR
operator between two variables, because we have no RSSI available anyway.

pe1chl, 2015-02-26:
There is one extra use for a more complex expression with AND and NOT.
This would be mainly of use when the CTCSS detector can run multiple instances.
(it can in the Tait repeater but currently not in SvxLink, I would think that is possible to add)

Say you run a repeater with 77Hz CTCSS and there is another repeater on the same freq
outside of normal range with 88.5Hz CTCSS.  During tropo users of the remote repeater
appear on the input.   In that case you might want a logical expression like:
CTCSS-77 OR (SIGLEV AND NOT CTCSS-88.5)
That will mean that users sending CTCSS at 88.5Hz will not open your squelch even when
they are quite strong, users sending 77Hz will always open the squelch, and users with no
or another CTCSS will open it only when quite strong.
Of course this is only a dream :-)

sm0svx, 2015-02-27:
Ok thanks. That sound like a valid case as well. I'll think about it a little bit more. However, the latter case complicate things a bit sine we need multiple config sections for the different squelch types.

pe1chl, 2015-02-27:
It would be fine when there at first is just a simple "OR" logic between existing detector mechanisms
and the possibility for more complex expressions and multiple instances of CTCSS and maybe other
detectors is left as a wishlist item for later.

pe1chl, 2018-07-21:
I have another use case for this: set a squelch type like "SIGLEV AND VOX" (maybe combined with the other examples given above).
This would be attractive because it would prevent the voter selecting receivers where a plain carrier is received, with quick response to normal situations (a VOX always has some hangtime but when SIGLEV drops the squelch closes immediately).

We are still anxiously awaiting this feature but we cannot find anyone who can understand the structure of the code well enough to program this.

sm0svx, 2020-07-27:
So... five years later...

Hope you still have any use for this feature because it was a lot of work to implement :-)

As of commit be4d911302d1ebf2d1218767075722c1fa655abe you can now set up squelch combinations using a configuration structure like the one below.
```
RxRtl:CTCSS_BLOCK]
SQL_DET=CTCSS
CTCSS_FQ=77.0
CTCSS_SNR_OFFSET=-3.22

[RxRtl:CTCSS]
SQL_DET=CTCSS
CTCSS_FQ=127.3,136.5
CTCSS_SNR_OFFSET=-3.22

[RxRtl:SIGLEV]
SQL_DET=SIGLEV
SQL_HANGTIME=100
SIGLEV_RX_NAME=RxRtl
SIGLEV_OPEN_THRESH=10
SIGLEV_CLOSE_THRESH=2
SQL_DELAY=300

[RxRtl]
...
SQL_DET=COMBINE
SQL_COMBINE=(RxRtl:SIGLEV | RxRtl:CTCSS) & !RxRtl:CTCSS_BLOCK
...
```
Since the CTCSS detector is a bit slow I had to add quite a big SQL_DELAY to the SIGLEV squelch. Otherwise the squelch will self oscillate.

closed 2020-07-27 by sm0svx

Dloranger, 2020-07-27:
Is there an option to have ctcss be a gpio input like the sql?  Many radios offer both a sql and ctcss valid signal as gpio, it would be helpful to support multiple gpio in the logics if possible

pe1chl, 2020-07-27:
Yes it is still interesting to have this!  Our big repeater with 25 receivers and 5 transmitters is no longer operational but we have another repeater which suffers from unfortunate CTCSS decisions: it operates on CTCSS 77 Hz but a neighboring repeater uses CTCSS 74.4 Hz and sometimes users of that repeater key our receiver.  So I would want to experiment with 2 tone detectors and the AND/NOT construct.  And of course also the other examples given above.
What makes it a bit more difficult to test it quickly is the incompatibility introduced in the remotetrx protocol some years ago, we still run the version before that everywhere so it is difficult to upgrade part of the sites.
Nice that there is still activity in the project, I will see how I can plan to upgrade the software (what sequence to use).

sm0svx, 2020-07-27:
@Dloranger: I guess it would be possible to set up two GPIO squelch detectors and use an expression like:

SQL_COMBINE=Rx1:GPIO_SQL | Rx1:GPIO_CTCSS

sm0svx, 2020-07-27:
@pe1chl: Interesting to hear if the new squelch can solve your problem but the risk is that you trigger the 74.4Hz "anti-detector" when transmitting on 77.0Hz since they are so close.