sm0svx/svxlink #134: Cor, cos,cas carrier detect

isladelobos, 2015-03-02:
Hello.
Im configuring the svxlink v1.2.0 to carrier detect from the busy radio.

My radio have a pin output it have always +5v High wen busy off, and +0v Low wen busy on (receiving).
It is a Icom radio.

I put one optocoupler pc123: http://www.antimath.info/wp-content/uploads/2011/06/pc123.jpg
Connect +5V to Anode and GND to Cathode in the radio side.
Connect (pin1 - DCD) to Collector and (pin5 - GND) to Emitter in the Computer side.

It is my configuration TX1,

The lines modified or added are:
SQL_DET=SERIAL
SQL_PORT=/dev/ttyS0
SQL_PIN=DCD:CLEAR

Im tested with CLEAR and SET but same, not activate the carrier.

[Rx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:0
AUDIO_CHANNEL=0
#SQL_DET=VOX
SQL_DET=SERIAL
SQL_PORT=/dev/ttyS0
SQL_PIN=DCD:CLEAR
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=2000
#SQL_EXTENDED_HANGTIME=1000
#SQL_EXTENDED_HANGTIME_THRESH=15
#SQL_TIMEOUT=600
VOX_FILTER_DEPTH=20
VOX_THRESH=1000
#CTCSS_MODE=2
#CTCSS_FQ=77.0
#CTCSS_SNR_OFFSET=0
#CTCSS_OPEN_THRESH=15
#CTCSS_CLOSE_THRESH=9
#CTCSS_BPF_LOW=60
#CTCSS_BPF_HIGH=270
SERIAL_PORT=/dev/ttyS0
SERIAL_PIN=CTS:SET
#SERIAL_SET_PINS=DTR!RTS
#EVDEV_DEVNAME=/dev/input/by-id/usb-SYNIC_SYNIC_Wireless_Audio-event-if03
#EVDEV_OPEN=1,163,1
#EVDEV_CLOSE=1,163,0
#SIGLEV_DET=TONE
SIGLEV_SLOPE=1
SIGLEV_OFFSET=0
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
SIGLEV_OPEN_THRESH=30
SIGLEV_CLOSE_THRESH=10
DEEMPHASIS=0
#SQL_TAIL_ELIM=300
#PREAMP=6
PEAK_METER=1
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=100
DTMF_SERIAL=/dev/ttyS0
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1
#SEL5_DEC_TYPE=INTERNAL
#SEL5_TYPE=ZVEI1

[Tx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:0
AUDIO_CHANNEL=0
PTT_PORT=/dev/ttyS0
PTT_PIN=DTRRTS
#SERIAL_SET_PINS=DTR!RTS
#PTT_HANGTIME=1000
TIMEOUT=300
TX_DELAY=500
#CTCSS_FQ=136.5
#CTCSS_LEVEL=9
PREEMPHASIS=0
DTMF_TONE_LENGTH=100
DTMF_TONE_SPACING=50
DTMF_TONE_AMP=-18

The cuestion is if i need more configurations add or change.

Thanks.
Roque Santana.

dl1hrc, 2015-03-02:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

> Hello. Im configuring the svxlink v1.2.0 to carrier detect from the
> busy radio.
>
> My radio have a pin output it have always +5v High wen busy off,
> and +0v Low wen busy on (receiving). It is a Icom [...]

Hi,

you need +12V / -12V for a standard RS232. 0V is not defined.

vy 73's de Adi, DL1HRC

> It is my configuration TX1,
>
> The lines modified or added are: SQL_DET=SERIAL
> SQL_PORT=/dev/ttyS0 SQL_PIN=DCD:CLEAR
>
> Im tested with CLEAR and SET but same, not activate the carrier.
>
> [Rx1] [...]
#EVDEV_OPEN=1,163,1
> #EVDEV_CLOSE=1,163,0 #SIGLEV_DET=TONE SIGLEV_SLOPE=1
> SIGLEV_OFFSET=0 #TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
> SIGLEV_OPEN_THRESH=30 SIGLEV_CLOSE_THRESH=10 DEEMPHASIS=0
> #SQL_TAIL_ELIM=300 [...]

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v1.4.11 (GNU/Linux)

iQEcBAEBAgAGBQJU9MITAAoJEIBzxqy+/CmtFhcH/2WAKeUL+S6NTXzMwKwK/XOz
5nb2GFVS/KuXR7VcPwiNJkbbrvYORcqROmM3M3l18M6ErNeEGP/knVH7sWX3YJxX
xGJKr2geHI8bHAPiS2wprt6LB8hVVWksVUO6JIeuREWkctPEkg9YZj2JXlyN9bkg
E8c/QrwC0qE7rdE6GaeK+1u5XVVGd4Gf+CEWSIzBxoOUucNPeQPSLNrwWfhXggfp
fUZVUSOzXyj0wh8YaBEZNs0k2ronBf8AiTRID8GYESqzFW7hKgBMoBcmnbaUkE1S
dE+Xg96AV4q2w3wk0zJmHeERR/lrac6a1EJrf68T3/BoJtQ3sr9S/qwAaXSmVtM=
=/h7U
-----END PGP SIGNATURE-----

isladelobos, 2015-03-02:
Thanks Adi. 

Im reading this article: http://www.svxlink.net/?page_id=294
But  suspect is between two radios or similar.

I can use this configuration:
http://320volt.com/wp-content/uploads/2008/12/opto-coupler-lojik-5volt-test.gif

73's
Roque

isladelobos, 2015-03-04:
Ok.
I see the configuration now, sorry.

For readers: These lines NO need to be added...
SQL_PORT=/dev/ttyS0
SQL_PIN=DCD:CLEAR

Only need modify the line:
#SQL_DET=VOX
SQL_DET=SERIAL

And take this info:
SERIAL_PIN=CTS:SET

Next the SQL SERIAL PIN in /dev/ttyS0 port is CTS, activated wen SET.

The electrical configuration was:
http://320volt.com/wp-content/uploads/2008/12/opto-coupler-lojik-5volt-test.gif
Because this radio is SET wen stand-bye. and CLEAR on busy.
Working fine.

closed 2015-03-05 by isladelobos