sm0svx/svxlink #135: NetUpLinkTrx issues: TCP Transmit buffer Overflow, TCP Transmit error and Heartbeat timeout,

f5vmr, 2015-03-30:
Having embarked on my first experiments with remotetrx, I am encountering these issues.
The first two occur when transmitting to the remote UHF receiver using a handi, EIther one or the other error occur after the squelch of the remote receiver opens, then disconnected the link with the VHF repeater. The heartbeat timeout occurs randomly.
We are fortunate to have a fixed IP address on the VHF repeater and have selected a port in a certain range open through the providers modem. One issue we have that inclement weather is impacting the Wifi link which is on 2.4 GHz and over 50 Kms, so we do have an occasional dropout.
I am connecting a UHF transceiver pair (as a repeater) on my personal IP address whih unfortunately is a dynamic IP address, so I have to check it regularly and update the distant Raspberry with the change of address in the lines in svxlink.conf. My own bandwidth and speed is also very restricted.
I am downlinking the audio and transmitting it on the UHF TX without difficulty, but as soon as I attempt a transmitting to the remote UHF rx, the faults appear.

Mon Mar 30 18:33:35 2015: NetUplinkTrx: Activating fallback repeater mode
Mon Mar 30 18:33:35 2015: *** ERROR: TCP transmit error in NetUplink NetUplinkTrx.
Mon Mar 30 18:33:56 2015: NetUplinkTrx: Client connected: 80.13.132.130:55679
Mon Mar 30 18:33:56 2015: NetUplinkTrx: Deactivating fallback repeater mode
Mon Mar 30 18:33:56 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:33:56 2015: Rx1: AddToneDetector(1750, 50, 150)
Mon Mar 30 18:33:56 2015: Rx1: AddToneDetector(88, 4, 1200)
Mon Mar 30 18:33:56 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio
Mon Mar 30 18:33:56 2015: Tx1: Turning the transmitter ON
Mon Mar 30 18:33:56 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio
Mon Mar 30 18:34:06 2015: Tx1: Turning the transmitter OFF
Mon Mar 30 18:34:37 2015: Rx1: SetMuteState(CONTENT)
Mon Mar 30 18:34:38 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:34:44 2015: Tx1: Turning the transmitter ON
Mon Mar 30 18:34:44 2015: Rx1: SetMuteState(CONTENT)
Mon Mar 30 18:34:48 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:35:04 2015: Tx1: Turning the transmitter OFF
Mon Mar 30 18:35:56 2015: *** ERROR: Heartbeat timeout in NetUplink NetUplinkTrx
Mon Mar 30 18:35:56 2015: NetUplinkTrx: Client disconnected: 80.13.132.130:55679
Mon Mar 30 18:35:56 2015: NetUplinkTrx: Activating fallback repeater mode
Mon Mar 30 18:36:27 2015: NetUplinkTrx: Client connected: 80.13.132.130:55682
Mon Mar 30 18:36:27 2015: NetUplinkTrx: Deactivating fallback repeater mode
Mon Mar 30 18:36:27 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:36:28 2015: Rx1: AddToneDetector(1750, 50, 150)
Mon Mar 30 18:36:28 2015: Rx1: AddToneDetector(88, 4, 1200)
Mon Mar 30 18:36:28 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio
Mon Mar 30 18:36:28 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio
Mon Mar 30 18:39:17 2015: NetUplinkTrx: Client disconnected: 80.13.132.130:55682
Mon Mar 30 18:39:17 2015: NetUplinkTrx: Activating fallback repeater mode
Mon Mar 30 18:39:17 2015: *** ERROR: TCP transmit error in NetUplink NetUplinkTrx.
Mon Mar 30 18:39:40 2015: NetUplinkTrx: Client connected: 80.13.132.130:55683
Mon Mar 30 18:39:40 2015: NetUplinkTrx: Deactivating fallback repeater mode
Mon Mar 30 18:39:42 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:39:42 2015: Rx1: AddToneDetector(1750, 50, 150)
Mon Mar 30 18:39:42 2015: Rx1: AddToneDetector(88, 4, 1200)
Mon Mar 30 18:39:42 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio
Mon Mar 30 18:39:42 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio
Mon Mar 30 18:40:00 2015: Tx1: Turning the transmitter ON
Mon Mar 30 18:40:30 2015: *** ERROR: Heartbeat timeout in NetUplink NetUplinkTrx
Mon Mar 30 18:40:30 2015: NetUplinkTrx: Client disconnected: 80.13.132.130:55683
Mon Mar 30 18:40:30 2015: Tx1: Turning the transmitter OFF
Mon Mar 30 18:40:30 2015: NetUplinkTrx: Activating fallback repeater mode
Mon Mar 30 18:40:31 2015: NetUplinkTrx: Client connected: 80.13.132.130:55684
Mon Mar 30 18:40:31 2015: NetUplinkTrx: Deactivating fallback repeater mode
Mon Mar 30 18:40:31 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:40:32 2015: Rx1: AddToneDetector(1750, 50, 150)
Mon Mar 30 18:40:32 2015: Rx1: AddToneDetector(88, 4, 1200)
Mon Mar 30 18:40:32 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio
Mon Mar 30 18:40:32 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio
Mon Mar 30 18:41:21 2015: *** ERROR: Heartbeat timeout in NetUplink NetUplinkTrx
Mon Mar 30 18:41:21 2015: NetUplinkTrx: Client disconnected: 80.13.132.130:55684
Mon Mar 30 18:41:21 2015: NetUplinkTrx: Activating fallback repeater mode
Mon Mar 30 18:41:51 2015: NetUplinkTrx: Client connected: 80.13.132.130:55686
Mon Mar 30 18:41:51 2015: NetUplinkTrx: Deactivating fallback repeater mode
Mon Mar 30 18:41:51 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:41:51 2015: Rx1: AddToneDetector(1750, 50, 150)
Mon Mar 30 18:41:51 2015: Rx1: AddToneDetector(88, 4, 1200)
Mon Mar 30 18:41:51 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio
Mon Mar 30 18:41:51 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio
Mon Mar 30 18:43:15 2015: Rx1: The squelch is OPEN (5.58395)
Mon Mar 30 18:43:15 2015: NetUplinkTrx: Tone detected: 1750
Mon Mar 30 18:43:15 2015: *** ERROR: TCP transmit buffer overflow in NetUplink NetUplinkTrx.
Mon Mar 30 18:43:15 2015: NetUplinkTrx: Client disconnected: 80.13.132.130:55686
Mon Mar 30 18:43:15 2015: Rx1: The squelch is CLOSED (5.33395)
Mon Mar 30 18:43:15 2015: NetUplinkTrx: Activating fallback repeater mode
Mon Mar 30 18:43:15 2015: Rx1: The squelch is OPEN (4.62038)
Mon Mar 30 18:43:15 2015: Tx1: Turning the transmitter ON
Mon Mar 30 18:43:25 2015: Rx1: The squelch is CLOSED (5.07821)
Mon Mar 30 18:43:25 2015: Tx1: Turning the transmitter OFF
Mon Mar 30 18:43:32 2015: Rx1: The squelch is OPEN (5.48053)
Mon Mar 30 18:43:33 2015: Tx1: Turning the transmitter ON
Mon Mar 30 18:43:37 2015: NetUplinkTrx: Client connected: 80.13.132.130:55687
Mon Mar 30 18:43:37 2015: NetUplinkTrx: Deactivating fallback repeater mode
Mon Mar 30 18:43:37 2015: Rx1: The squelch is CLOSED (5.19495)
Mon Mar 30 18:43:37 2015: Tx1: Turning the transmitter OFF
Mon Mar 30 18:43:37 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:43:37 2015: Rx1: AddToneDetector(1750, 50, 150)
Mon Mar 30 18:43:37 2015: Rx1: AddToneDetector(88, 4, 1200)
Mon Mar 30 18:43:37 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio
Mon Mar 30 18:43:37 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio
Mon Mar 30 18:43:41 2015: Rx1: The squelch is OPEN (4.53735)
Mon Mar 30 18:43:41 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:43:42 2015: *** ERROR: TCP transmit buffer overflow in NetUplink NetUplinkTrx.
Mon Mar 30 18:43:42 2015: NetUplinkTrx: Client disconnected: 80.13.132.130:55687
Mon Mar 30 18:43:42 2015: Rx1: The squelch is CLOSED (5.33001)
Mon Mar 30 18:43:42 2015: NetUplinkTrx: Activating fallback repeater mode
Mon Mar 30 18:44:02 2015: NetUplinkTrx: Client connected: 80.13.132.130:55688
Mon Mar 30 18:44:02 2015: NetUplinkTrx: Deactivating fallback repeater mode
Mon Mar 30 18:44:02 2015: Rx1: SetMuteState(NONE)
Mon Mar 30 18:44:02 2015: Rx1: AddToneDetector(1750, 50, 150)
Mon Mar 30 18:44:02 2015: Rx1: AddToneDetector(88, 4, 1200)
Mon Mar 30 18:44:02 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio
Mon Mar 30 18:44:02 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio

The NetTx port for listening I have selected is 19123, So I am a little confused to see the other ports appear shown alongside NetUpLinkTrx like 55688 etc.
I know many of you have already overcome these issues, so I would appreciate advice.

Chris F5VMR

f5vmr, 2015-03-30:
Subsequent to this I see that each time this occurs the following line is produced:
Mon Mar 30 18:04:37 2015: NetUplinkTrx: Client connected: 80.13.132.130:55654 but each time this port number is incremented. We are up to 55700 now! What is going on here? Is this a port generated elsewhere in the APP?

pe1chl, 2015-03-31:
It looks like your network is failing.  Maybe the link fails as soon as you send some traffic.
When the repeater is idle there is only a regular heartbeat, when there is audio there is more
traffic to send.
The incrementing port number is normal when a connection fails and is re-established.  The
destination port number is always the same but the source port number is incrementing.
(this is done by Linux, not by SvxLink)

When you have a dynamic address, use a VPN (e.g. OpenVPN) to establish a tunnel over the
connection with a fixed IP address.  This can be done by setting up an OpenVPN server at the
side with the fixed address, and connect from the dynamic address side to there using a config
that will give a fixed address to the client (a bit too much to explain here, Google for it).
Then you have a fixed address on both sides and can use SvxLink on top of that.  We do this
all the time, and it works quite well also e.g. for temporary repeater access over UMTS.

f5vmr, 2015-03-31:
Hi Rob, 

Well I had already considered the problem thus, with a ddns address for my
end (the dynamic address) placed on the distant end in the lines for the
configuration for the remote receiver, I do have an apparent stable
connection, but the transmit problem is still the same, that is when I
transmit by my handi to the remote receiver, the link is broken off by
remotetrx.  I can see this happening on both sides, by simultaneously
looking at the activity by putty, and tail -f both logs (svxlink.log of the
repeater and remotetrx.log at my end). The repeater log simply indicates
that the link is terminated, whereas the remotetrx log shows me the TCP
errors. It is certainly indicative of a network problem. The ports are open,
the speed is adequate, but somewhere my config is not so hot. You have seen
the results of the first logs.

 

Stable connection:

Tue Mar 31 11:20:13 2015: NetUplinkTrx: Activating fallback repeater mode

Tue Mar 31 11:20:32 2015: NetUplinkTrx: Client connected:
80.13.132.130:56408

Tue Mar 31 11:20:32 2015: NetUplinkTrx: Deactivating fallback repeater mode

Tue Mar 31 11:20:32 2015: Rx1: SetMuteState(NONE)

Tue Mar 31 11:20:32 2015: Rx1: AddToneDetector(1750, 50, 150)

Tue Mar 31 11:20:32 2015: Rx1: AddToneDetector(88, 4, 1200)

Tue Mar 31 11:20:32 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio

Tue Mar 31 11:20:32 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio

Tue Mar 31 11:21:55 2015: Rx1: SetMuteState(CONTENT)

Tue Mar 31 11:21:56 2015: Rx1: SetMuteState(NONE)

 

Then immediately after my transmission to the remote receiver:

Tue Mar 31 11:25:29 2015: Rx1: The squelch is OPEN (5.76166)

Tue Mar 31 11:25:29 2015: Rx1: SetMuteState(NONE)

Tue Mar 31 11:25:30 2015: *** ERROR: TCP transmit buffer overflow in
NetUplink NetUplinkTrx.

Tue Mar 31 11:25:30 2015: NetUplinkTrx: Client disconnected:
80.13.132.130:56409

Tue Mar 31 11:25:30 2015: Rx1: The squelch is CLOSED (5.51166)

Tue Mar 31 11:25:30 2015: NetUplinkTrx: Activating fallback repeater mode

 

Then a few moments later everything resets:

Tue Mar 31 11:25:50 2015: NetUplinkTrx: Client connected:
80.13.132.130:56411

Tue Mar 31 11:25:50 2015: NetUplinkTrx: Deactivating fallback repeater mode

Tue Mar 31 11:25:50 2015: Rx1: SetMuteState(NONE)

Tue Mar 31 11:25:50 2015: Rx1: AddToneDetector(1750, 50, 150)

Tue Mar 31 11:25:50 2015: Rx1: AddToneDetector(88, 4, 1200)

Tue Mar 31 11:25:50 2015: NetUplinkTrx: Using CODEC "S16" to encode RX audio

Tue Mar 31 11:25:50 2015: NetUplinkTrx: Using CODEC "S16" to decode TX audio

 

The security phrases are identical as is the listening port. 

 

Any further ideas would be gratefully received.

 

Chris F5VMR

pe1chl, 2015-03-31:
It could be that you have a MTU problem somewhere.
As long as there are only small packets like connection establishment and heartbeat everything
is OK, but when you send larger amounts of data there is a too large packet and it is not correctly
fragmented, the link gets stuck and it fails.
You should have the same problem when you SSH from one machine to the other and do something
that creates a lot of output.

f5vmr, 2015-03-31:
Hi Again Rob,

It seems that changing the MTU to 1492 on the raspberry did have an effect,
but not anywhere near good. It now drops out on receive. I will explore the
VPN option as see what occurs.

Regards

Chris

pe1chl, 2015-03-31:
Ok in general issues like this are caused by badly configured firewall (dropping all ICMP packets)
but it can be caused by bad routers as well.  When the real MTU is 1492 you are probably using
PPPoE to the provider.  Make sure the MTU is changed at both ends when there indeed is an
issue with a bad firewall that you cannot correct.  It is possible to set a lower MTU on the ethernet
interface but this can also be put in a route:
ip route add 10.20.30.40 via 192.168.1.1 dev eth0  mtu 1492

VPN will make this particular issue only worse (or more complicated), it is only useful as a solution
of the problem of varying IP address of a remotetrx.

f5vmr, 2015-03-31:
Hi Rob,

Firstly there are no firewall issues, as there is no firewall in place on
the addresses of both raspberries.

The internet here is running on PPPoE, and both double sets of routers are
set to 1492 MTU. I forced the issue on both raspberries as you described
earlier today but no difference has been observed. I also enlarged the
buffer size but notice no change there either.

So to describe the network:

The remotetrx raspberry is hardwired to a secondary router, acting only as a
wifi extender in a difficult to reach part of the house hardwired to the
Livebox. The DNS of the router is disabled, so all connections take their IP
addresses from the DNS of the Orange Livebox. I run with the firewall off,
and let each of my computers monitor their own aspects. An Orange Livebox is
also the other side of the network in Limoges that then feeds a bullet to
our 50 KMs Wifi to a second bullet that then feeds a router also fully open
to the SVXLink Raspberry. Despite this complexity, I am receiving the audio
from the SVXLink Repeater audio which is then transmitted on my local UHF TX
from the remotetrx Raspberry. The instant I transmit to the receiver
attached to this raspberry, the faults are indicated. Monitoring the main
repeater on a VHF set nearby I hear my transmission initiate then drop out.
Only the remotetrx raspberry demonstrates the errors, and after a pause the
link re-establishes, frequently without the received audio from the repeater
dropping at all.  We have opened a range of ports for our use only on all
the distant routers.

This is the first effort to trial a remote receiver. I have had no previous
issue with running EchoLink or other data streams from my home, although
streaming video is not entirely without problems, Speed here is around 2Mbs
RX and not more than 250Kbs TX.

Thanks for your help so far.

 

Chris.

pe1chl, 2015-04-01:
Ok it is always difficult to debug network problems from a description.
When you have set the MTU to a value that should be acceptable end-to-end, that should not
be the issue.  I only warned for this because especially when you have a special MTU smaller
than 1500 and on the path there are badly configured firewalls that drop the ICMP destination
unreachable (fragmentation needed but DF set) message, the connection is going to fail in
exactly the way you are experiencing.  And this is unfortunately quite common, even at ISPs,
because of bad information on the internet (ICMP=ping=bad, you should block it).  When clueless
people talk about "you should be stealth" you know you are in the wrong place for information.
(unfortunately some people highy regard the hogwash site where this is proclaimed)

But ok, when your MTU problem is because of PPPoE and you have already set MTU 1492 that
should not be the cause.  Still, I think there is something that fails on large packets.
I would recommend to install the wireshark package (terminal mode), using:
apt-get install tshark
and then trace the network traffic using something like:
tshark -n -ta -i eth0
When you are connected over the same interface you want to exclude your own connection, e.g.:
tshark -n -ta -i eth0 not tcp port 22
or
tshark -n -ta -i eth0 not host 11.22.33.44

Then observe what is going on.   You can see what is going back and forth and see where it
hangs and what happens there (is the connection closed, is it re-trying until it fails, what is it
re-trying?  a large packet?)
When you do this on both ends you can see what does and what does not make it through the
link.
You can add the -x option to see the packet in hex as well, or the -V option to get a complete
description of the entire packet (a screen full of data for every packet).
If desired, re-direct it into a file to study later.

I can assure you that remotetrx basically works.   We use it all the time, and others do so as well.

f5vmr, 2015-04-01:
Thank you Rob,

 

That is very comprehensive. I did try one thing, eliminating the second
router at my house, going straight to the livebox, but with the same errors.

I will give the wireshark a try after my lunch.

I know the remotetrx works as you say you have great success.

Many thanks.

Chris

f5vmr, 2015-04-01:
Hi Rob,

Just a brief update to say the Raspberry at the VHF repeater is doing the
same thing TCP Transmit Buffer overflow. 

Tell me - would the audio level present at the time have any effect on the
size of the packet? Should I leave the size of the buffer as default?

I have looked at some results of the tshark but I cannot see anything of
value, the fact is I don't understand most of it. J I will discuss the
resulting file with a friend who might do.

Both ends of our network indicate the MTU at 1492.

I am not happy with our personal internet at home due to persistent noise
errors and sometimes intermittent connection, but Orange say there is
nothing wrong when they conduct tests. I have lost count of the number of
times our IP address changes over consecutives 24 hour blocks. As they are
also the provider on the distant end as well, it might be a challenge to
resolve our problem. It is quite likely not our systems at all.

 

I shall next try two raspberrys on my home network, to prove my system,
before I start going further. Fortunately I have several J ready for the
next repeaters to be installed.

Regards

Chris

f5vmr, 2015-04-01:
Hi Rob,

I think I have proved conclusively that the Internet Infrastructure in our
sector is less than acceptable.

 

I Created a raspberry model B as "a repeater", that is to say made svxlink
as the prime mover (without radios I might add), but configured NetRx and
NetTx in the Voter and MultiTx sections respectively, and used my real UHF
repeater as the "remotetrx", and there was no problem whatsoever over my own
network. I am almost delighted that this is the case, but it does not
overcome my initial problem with the scheme in place. I shall have to tackle
the IP to see what they have to say about it.

 

Many thanks for your assistance.

Chris F5VMR

pe1chl, 2015-04-01:
Ok you need to have some knowledge of TCP to know if the network is working right.
I have used TCP/IP over packet radio in the past so I know the protocol from the trace screens
and I assume to easily that others can understand it as well from the tshark output.
Hopefully your friend can help you diagnose the problem.  However, checking if what is sent at
one side is indeed appearing at the other side should not be difficult.
I have MTU issues myself since I switched over to VDSL where PPPoE is unfortunately used as
well, so I know what the problems can be.   I use a default route with MTU 1492 myself.  With this
correctly configured, it should be workable without problems.
Increasing the transmit buffer is not the way to go, this is just a symptom.  What happens I think
is that the TCP stream gets stuck, and the buffer will fill no matter how large you make it.  When
the connection is working OK there should not be more than a few KB buffered at any time.
Try to use SSH, SCP, FTP or whatever you find convenient to debug the connection problem first,
and then go back to SvxLink.

pe1chl, 2015-04-21:
As you may have read on the development mailing list, we got problems like that as well now.
A WiFi link that is sometimes a bit weak causes TCP buffer overflows, and the handling caused
the service to crash.  Tobias has implemented a fix, and we are now testing it (it is in the master
branch on the git server)

f5vmr, 2015-12-11:
I have updated the main repeater to 1.5.0 but yet to do it on the remote receiver, where the fault still appears. I will try and upgrade that but there are other issues with it at the moment in relation to voting and audio quality when it is voted in. Maybe the upgrade will cure it.

f5vmr, 2017-11-11:
No re-occurence of this issue

closed 2017-11-11 by f5vmr