sm0svx/svxlink #136: Supervisory web page

brettwi, 2015-04-10:
As an enhancement, would it be possible to have svxlink generate a Supervisory web page indicating who has connected via Echolink, what the current status of the repeater is (up or down), and ability to control various features within svxlink? 

My repeater will be rather remote and hard to access, but with an interactive web page I can control (even reboot via password, or disable the transmitter if illegal activity is observed) the server. Primarily it would be beneficial to be able to monitor server activity (how many times has someone used the repeater and totals per day, week, month, total transmit time (a nice bar graph would be useful) and other supervisory functions. It would be nice to see how many times someone is 'kerchunking' the repeater. It would be extremely helpful to be able to see all the svxlink,conf variables remotely.

This feature would be very beneficial for a club or the average repeater user to be able to view, with guest and supervisory levels of access.

That way I can see what my power usage might be, and tell if I am overworking my poor amplifier.

Would also be a nice way to schedule on the air nets if I can see when the busiest times of day are for traffic. Would also be nice to have a log of how many times features like echolink, metarinfo, propagation, and voicemail are being used. Would be nice to see a list of voicemails waiting to be heard, and what is happening with mailboxes. 

A very nice feature might also be the ability to send test tones (to check RX levels) or even to be able to change the normal ID to a text message which could be read out via the speech engine ("repeater xxx will be down for maintenance at 2pm for one hour").

Seems like the killer app for this software. Since we all have computers running svxlink, why not make use of them.

Has anyone gone this far? I know this is far reaching, but it seems to be the next step.

Thank you, 

Brett Williams
WA6SXU

sm0svx, 2015-04-19:
This is quite a big job and developing web UI:s in not one of my areas of expertise. Maybe one day I'd like to learn it and then this would be an obvious project.

It is possible to get some status data out of SvxLink using the publishStateEvent TCL function. You could get all sorts of data out this way and publish it to the state event PTY. A web UI could listen to this PTY and display the information in a user friendly way.

There is no good way today to remotely control SvxLink so implementing a web UI to control things is a bit harder.

There is a project called [OpenRepeater](https://github.com/OpenRepeater/OpenRepeater) that aim to wrap SvxLink in a web UI but I have not seen any activity for a while. You may ask around there.

pe1chl, 2015-04-21:
We have been considering this as well, and probably we have a team member who could create
that webpage, and I already had been thinking about how to implement control of the software
from there.  It appears now the only way is to send DTMF command sequences to stdin, which
is a bit limited.  It would be nice when there was some API (maybe via TCL?) where certain aspects
of the repeater could be controlled via calls made from the webserver.  Could be via a PTY or
some standard RPC mechanism.
We could envision things like:
- enable/disable the entire repeater
- enable/disable DTMF (can be done by setting a TCL variable)
- enable/disable an rx
- enable/disable a tx

With a feature like that, and indeed the eventsource mechanism for output (also a process could
read the logfile), it would be possible to construct something resembling the above request.

Note that one of the things brett is requesting is already there.  When you enable the
LocationInfo feature you can get the usage statistics of the repeater from an APRS system that
collects them, like aprs.fi.   Also you can use existing packages for system management and
monitoring to do some of these things.  We use nagios.  "control panel" implementations for
system management that allow you to reboot, and perform simple management tasks via a
webpage also exist.

f5nlg, 2017-06-16:
Hi,
Spotnik who is a hotpsot project have a GUI writen in javascript , in SvxReflector mode you can see the "node or relay" connected , who is talking , configure the hotspot , a lot of automatism are implemented like search the better proxy for EchoLink , catch Location from the web , get Ip Adress etc ...
it works on SvxReflector , Echolink , Free Radio Network ( but not tested i can't get a valid account )
this is an active project .
http://www.spotnik.org and https://github.com/spotnik-ham

thanks for reading

JP Piers / F5NLG