sm0svx/svxlink #139: PlayTone enhancement:

DG1BBQ, 2015-05-17:
**PlayTone enhancement:**

Pure sine tones sound nice and smooth, but there’s a drawback. They can easily be overheard when reflected audio from walls at home or inside your car causes local destructive interference, so the tone becomes almost inaudible depending on your position relative to the source (dead zones. Very early on, people have recognized this and made emergency signals (sirens, claxons) to have lots of harmonics to rule out that local destructive interference causes dead zones).

I know it's possible to have PlayTone produce a square wave by cranking up the volume (e.g. PlayTone 1000 900 100). Unfortunately the produced tone is invariably loud.

I'd like to propose that PlayTone be modified to optionally produce square waves (possibly even triangle waves?)

Maybe like this :

PlayTone 1000 50 100 sq  # plays 1kHz square wave at volume 50 for 100ms

Or:

playSquareTone 1000 50 100

Axel, DG1BBQ

sm0svx, 2015-06-06:
Thanks for the suggestion. I'll mark it as a feature request.

f5vmr, 2024-04-29:
playTone 440 500 100 seems to have no output - Should this be 440 50 100 which does?