sm0svx/svxlink #141: autocon behaviour

DG1BBQ, 2015-05-17:
**autocon behaviour**

We at DB0BHV-R do not have a timeout set, so stations may connect and stay as long as they wish. The current behavior is problematic as it will only autoconnect if there is currently no connect to anywhere. Also, it would be favorable that autoconnect would reconnect only after there has been no **local** (meaning only stuff comming from the local RX(s)) activity for a **configurable** period of time.
Short bursts that open the squelch (<2s) which cannot be meaningfull local traffic should be ignored by the inactivity timer.

Look at the following scenario:

Autoconnect has previously established a link to a partner repeater. Now someone wants to talk local only. He/She disconnects (##) and has his/her QSO. The way autoconnect works now, if the local QSO takes longer than AUTOCON_TIME, it will suddenly reconnect to the partner repeater, and if there is a local QSO in progress there at the time, the two QSOs will 'collide' and can result in frustration due to it being bad manners.

With the changed behavior, autoconnect would hold back as long as there is local activity and then execute the autoconnect after waiting a configured amount of time (AFAIK, D-Star and DMR do it in a similar fashion when you change a repeater to a different reflector).
 
Also, the option to autoconnect more than one destination would *REALLY REALLY REALLY* be appreciated! Seriously.

Axel, DG1BBQ

sm0svx, 2015-06-06:
I'd suggest that you split this issue in two since it's really two things (timeout handling and multi autoconnect).

on4bcp, 2015-11-21:
Hello,

We are using svxlink on about 6 repeaters with a autonnection to Echolink conference BELREP. With this function we can make one big repeater and is there still a possibility to isolate each repeater by the user ( with dtmf #). With the autocon_time set to 900, the connection to Belrep is automatically made after 15 min. The problem is that the Autoconnection is also made when there is local activity at the repeater and does interfere with possible other qso's at the other repeaters connected to the same conference box.
We would like to set the Autocon_time as low as possible (for example at 2 min.) and this timer to start when squelch closed and resets when there is a squelch on again. So, when there is no activity anymore within the set autocon_time, the connection to echolink may be established again.

I hope that someone can help us with this ?

Bart Peeters, on4bcp

DG1BBQ, 2015-11-21:
That's actually exactly what I proposed. If more need the proposed changes, it'll maybe bump up the priority :]

I also hereby drop the multicon request part, so there's only the autocon behavior change request left.

Axel, DG1BBQ

DG1BBQ, 2017-02-05:
I modified the autocon behavior by writing an external bash script, that reads the status via PTY (repeater up/down, transmit on/off, Echolink connects and disconnects)  through messages from publishStateEvent commands added to the appropriate procedures. it replaces the inbuilt autoconnect completely, which can subsequently be disabled. No changes to the source code necessary.
It has been in operation for more than a year on all repeaters of our linked  reapeater system in the north west of Germany.
(DB0BHV, DB0WTV, DB0EMS, DH0BOH, DB0PDF, DB0HFT, DB0CU, and more comming) and it's working great.

73's, Axel DB1BBQ

on4bcp, 2017-02-06:
Hello,

Looks verry good!
Can you send me more detail about how to inplement this on our repeaters?

Kind regards,

Bart Peeters, on4bcp

DG1BBQ, 2017-02-06:
Hi Bart,

 

I’ll Whip up a guide on what changes need to be made to the TCLs and I’ll also send the required scripts.

Give me a few days.

 

73’s, Axel (DG1BBQ at g m x . d e)

 

P.S.: Didn’t want to close the issue. Must have clicked a wrong button. It would still be nice, if this were implemented in SVXLink.

sm0svx, 2017-02-06:
Great that you found a working solution but as you say, the correct solution would be to change the existing C++ implementation. Reopening issue.