sm0svx/svxlink #142: IP-Mismatch Problem

DG1BBQ, 2015-05-17:
**IP-Mismatch Problem**

The current fix doesn't really solve the problem as it doesnt' fix both sides.

**Scenario:**
Repeater A (henceforth A) only has a DSL line with a dynamic public IP address. The provider automatically disconnects after 24hours (that's how it is in Germany). At the time, A was connected to Repeater B. A's router will resync and immediately reconnect, but will be assigned a new and different public IP address! A's SVXLink will lose the connection to B. After it times out, A's SVXLINK twill try to reconnect to B.

At this point, the question is: "did SVXLink update it's current IP with the EchoLink-server after the DSL connection was reastablished and before it tries to reconnect to B?".

If it did, everything should be fine, provided the Echolink server passes this IP change on in a timely fashion, so that when B receives the reconnect, it first sees the IP mismatch and thus retrieves the current station list from the Echolink server. If the server passes on the info quickly, all is good. The next incomming try from A will succeed! If not, A's reconnect tries will max out (not sure, how many tries A makes to reconnect to B. (I think it's 4 or 5)).

At this point the reconnect is unsuccessful and is abandoned by A.

In comes the current autocon behaviour on A. It will now wait untill its AUTOCON_TIME runs out. If it's set to something like 30mins, it takes a long time until the connection between A an B is back up.

To make sure, the reconnect tries from A should be increased to a number that safely bridges the time it takes for  A's IP address change to be passed on through the Echolink server to B and thus allowing for a successfull reconnect whith as little delay as possible.

I hope I was able to clearly convey the problem.

I am aware that this will only (probably) fix the problem between SVXLink stations. The Echolink clients for Windows/Android/iOS are rarely maintained and would not benefit from this.

Axel, DG1BBQ

pe1chl, 2015-06-02:
In an environment like that, it is best to setup a VPN with fixed addresses so you hide the varying IP addreses from the application.
That way you also avoid the problem of having to insert port forwards in the routers.
We are doing that in our network whenever we have nodes on internet, because we also do lots of monitoring and do not always have direct access to the internet routers we can use for the connection.
E.g. use OpenVPN.

sm0svx, 2015-06-06:
The registration should be refreshed within 5 minutes from when the public IP address changes. The real problem may be the autoconnect behavior. It should probably not wait the AUTOCON_TIME upon an unsuccessful connection but rather keep trying to connect. Maybe then this would not be such a big problem.