sm0svx/svxlink #143: Timeout parameter in logic linking

dl7ata, 2015-05-26:
when I use following section *with* TIMEOUT

 [NetLink]
 CONNECT_LOGICS=SimplexLogic:95:NET1,NetLogic:94:NET9
 DEFAULT_ACTIVE=1
 TIMEOUT=180
 
 I got the error
 
 svxlink: /home/cubie/svxlink/src/svxlink/svxlink/LinkManager.cpp:769: void
 LinkManager::checkTimeoutTimer(LinkManager::Link&): Assertion `lmit !=
 logic_map.end()' failed.
 
Abgebrochen

If I delete the TIMEOUT param, everything works fine.

dl1hrc, 2015-05-26:
we stay in contact, will check it on dl7ata's system this week.

73s Adi

sm0svx, 2015-06-06:
Were you able to pinpoint the problem?

dl7ata, 2015-06-06:
No we were not. Adi had has success to eliminate the message, but the parameter itself doesn’t work on my machine.

sm0svx, 2015-06-06:
When does it crash?  Is it directly when SvxLink is started or some time later?

dl7ata, 2015-06-06:
It crashed on start.

sm0svx, 2015-06-06:
OK. I think I found the problem.

Have you modified the "startup" function in Logic.tcl to play a sound?  Some other modification of TCL scripts that may fire off during startup?  If that is the case, the latest master branch will solve the problem.

dl7ata, 2015-06-07:
Yes I have. I am running two logics and have different sounds during start up.

I’ll compile the latest branch and try out – thanks!

dl7ata, 2015-06-07:
Is something changed in github?

 

me@Cubian:~/svxlink$ git pull

remote: Counting objects: 18, done.

remote: Compressing objects: 100% (18/18), done.

error: insufficient permission for adding an object to repository database .git/objects

 

fatal: failed to write object

fatal: unpack-objects failed

dl7ata, 2015-06-07:
Found a solution …. Compiler running.

 

$ git stash

$ git pull <remote name> <remote branch name> (or) switch branch

$ git stash apply --index

 

_______________________________

 

Is something changed in github?

 

me@Cubian:~/svxlink$ git pull

remote: Counting objects: 18, done.

remote: Compressing objects: 100% (18/18), done.

error: insufficient permission for adding an object to repository database .git/objects

 

fatal: failed to write object

fatal: unpack-objects failed

dl7ata, 2015-06-07:
Unfortunately the timer doesn’t work:

 

[NetLink]

CONNECT_LOGICS=SimplexLogic:95:NL1,NetLogic:94:N9X

DEFAULT_ACTIVE=1

TIMEOUT=300

 

My understanding is, that 5 minutes after deactivating without any activity, the link should reactivate. It does not, there is also nothing in the log. I have to do it manually.

sm0svx, 2015-06-07:
Yes, it should reactivate and it do so here in my test setup. Note that announcements also count as activity so both logic cores must be completely quiet (no traffic, no announcements) for the full five minutes. If you have a lot of Tcl customizations, try to temporarily disable them to see if any of them are causing a problem.

(Note: When adding comments to the issue via email, please trim them down. Do not include the message history/quoting. It makes the issue thread very hard to read. I have edited all your comments above to make the thread readable. Even better is to use the web interface directly: #143)

dl7ata, 2015-06-08:
Test with SvxLink v1.4.99.21 (Jun  7 2015)  is ok. I’ve deactivate all my locale-TCLs and after timeout, the link is reactivated. So the bug is inside the TCLs.

But I made another interesting observation:
As you mentioned, every new input restarts the timeout-timer. Unfortunately also if the input (sql-open/close) comes from the deactivated Netlink. That means, if don't hear anything on your local receiver, it will take some more time when reactivating, depending also on the traffic at the deactivated side.

dl1hrc, 2015-06-10:
The problem occurred obviously related to my SvxServer-branch. The SQL-close command was not sent back to the belonging NetLogic so the SQL state remain open for ever because Logic::getIdleState returned always false. 
Has been fixed, will sent an update to my branch soon.

dl1hrc, 2015-06-10:
BTW: When using LogicLinking together with my SvxServer-branch I think it's better to contact me directly since the special configuration and handling of the NetTrxMsg protocol and the behaviour of the connected svxlink-nodes is different to the normal svxlink<->remotetrx configuration. Maybe the other users could be confused by discussions of this special case.

closed 2015-09-27 by sm0svx