sm0svx/svxlink #144: current cmake issue debian jessie

rneese45, 2015-06-01:
make install failes with following errors

-- Setting owner of /var/spool/svxlink/qso_recorder to svxlink.daemon...
/bin/chown: invalid user: ‘svxlink.daemon’
CMake Error at svxlink/svxlink/cmake_install.cmake:100 (MESSAGE):
  Error while changing owner of file /var/spool/svxlink/qso_recorder
Call Stack (most recent call first):
  svxlink/cmake_install.cmake:38 (include)
  cmake_install.cmake:41 (include)

Makefile:66: recipe for target 'install' failed
make: *** [install] Error 1
root@kb3vgw-repeater:/usr/src/svxlink-14.08.1/src/build#

pe1chl, 2015-06-02:
Please read the installation instructions on the WiKi when installing from source.
https://github.com/sm0svx/svxlink/wiki/InstallationInstructions

rneese45, 2015-06-03:
I followed that to the T and doted the i's and I still get the same error on debian jessie armhf

rneese45, 2015-06-03:
/bin/chown: invalid user: ‘svxlink.daemon’
CMake Error at svxlink/svxlink/cmake_install.cmake:100 (MESSAGE):
Error while changing owner of file /var/spool/svxlink/qso_recorder
Call Stack (most recent call first):
svxlink/cmake_install.cmake:38 (include)
cmake_install.cmake:41 (include)

Makefile:66: recipe for target 'install' failed
make: *** [install] Error 1
root@kb3vgw-repeater:/usr/src/svxlink-14.08.1/src/build#

fresh install today and ran it again per the wiki page it still fails

rneese45, 2015-06-03:
here is the step by step 

apt-get install -y g++ make cmake libsigc++-2.0-dev libgsm1-dev libpopt-dev libgcrypt11-dev \
			libspeex-dev libasound2-dev alsa-utils vorbis-tools sox libsox-fmt-mp3 sqlite3 \
			unzip opus-tools tcl8.6-dev 

	mkdir /usr/src/downloads
	cd /usr/src/downloads
	wget https://github.com/sm0svx/svxlink/archive/14.08.1.tar.gz
	tar xvzf 14.08.1.tar.gz -C /usr/src/
	cd /usr/src/svxlink-14.08.1/src

        mkdir build && cd build

root@kb3vgw-repeater:/usr/src/svxlink-14.08.1/src/build# cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc -DLOCAL_STATE_DIR=/var -DUSE_QT=NO ..
-- Reading versions file...
-- Found components for RT
-- RT_INCLUDES = /usr/include
-- RT_LIBRARIES     = /usr/lib/arm-linux-gnueabihf/librt.so
-- Could NOT find Opus (missing:  Opus_LIBRARY Opus_INCLUDE_DIR)
-- Found ALSA 1.0.28
-- checking for module 'popt'
--   found popt, version 1.16
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so
-- Found GCrypt: /usr/lib/arm-linux-gnueabihf/libgcrypt.so
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so
-- Found Tclsh: /usr/bin/tclsh8.6 (found version "8.6")
-- Found TCL: /usr/lib/arm-linux-gnueabihf/libtcl8.6.so
-- Could NOT find TCLTK (missing:  TK_LIBRARY TK_INCLUDE_PATH)
-- Could NOT find TK (missing:  TK_LIBRARY TK_INCLUDE_PATH)
-- Found components for DL
-- DL_INCLUDES = /usr/include
-- DL_LIBRARIES     = /usr/lib/arm-linux-gnueabihf/libdl.so
-- Could NOT find TCLTK (missing:  TK_LIBRARY TK_INCLUDE_PATH)
-- Could NOT find TK (missing:  TK_LIBRARY TK_INCLUDE_PATH)
-- Found Popt: /usr/lib/arm-linux-gnueabihf/libpopt.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/src/svxlink-14.08.1/src/build

root@kb3vgw-repeater:/usr/src/svxlink-14.08.1/src/build# make
[  0%] [  0%] [  1%] Scanning dependencies of target asynccore
Generating ../../../include/version/SVXLINK.h
Generating ../../../include/version/SVXLINK.h
Generating ../../include/version/SVXLINK.h
[  3%] [  3%] Scanning dependencies of target trx
Generating ../../../include/version/MODULE_HELP.h
Generating ../../../include/version/MODULE_PARROT.h
[  3%] Scanning dependencies of target ModuleHelp
Scanning dependencies of target ModuleParrot
Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncApplication.cpp.o
[  5%] [  5%] Building CXX object svxlink/modules/help/CMakeFiles/ModuleHelp.dir/ModuleHelp.cpp.o
Building CXX object svxlink/modules/parrot/CMakeFiles/ModuleParrot.dir/ModuleParrot.cpp.o
[  5%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/ToneDetector.cpp.o
[  6%] Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncFdWatch.cpp.o
[  7%] Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncTimer.cpp.o
Linking CXX shared module ../../../lib/ModuleHelp.so
[  8%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/SwDtmfDecoder.cpp.o
[  8%] Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncIpAddress.cpp.o
[  8%] Built target ModuleHelp
[  9%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/Rx.cpp.o
Linking CXX shared module ../../../lib/ModuleParrot.so
[  9%] Built target ModuleParrot
[  9%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/LocalRx.cpp.o
[ 10%] Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncDnsLookup.cpp.o
[ 11%] Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncTcpClient.cpp.o
[ 11%] Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncUdpSocket.cpp.o
[ 13%] Building CXX object async/core/CMakeFiles/asynccore.dir/AsyncTcpServer.cpp.o
[ 13%] Generating ../../../include/version/MODULE_TCL.h
[... 296 lines cut ...]
-- Installing: /usr/include/svxlink/AsyncAudioDevice.h
-- Installing: /usr/include/svxlink/common.h
-- Installing: /usr/lib/libecholib.so.1.3.1
-- Installing: /usr/lib/libecholib.so.1.3
-- Installing: /usr/lib/libecholib.so
-- Removed runtime path from "/usr/lib/libecholib.so.1.3.1"
-- Installing: /usr/include/svxlink/EchoLinkDirectory.h
-- Installing: /usr/include/svxlink/EchoLinkDispatcher.h
-- Installing: /usr/include/svxlink/EchoLinkQso.h
-- Installing: /usr/include/svxlink/EchoLinkStationData.h
-- Installing: /usr/include/svxlink/EchoLinkProxy.h
-- Installing: /usr/bin/svxlink
-- Removed runtime path from "/usr/bin/svxlink"
-- Installing: /etc/svxlink/svxlink.d
-- Installing: /var/spool/svxlink/qso_recorder
-- Setting owner of /var/spool/svxlink/qso_recorder to svxlink.daemon...
/bin/chown: invalid user: ‘svxlink.daemon’
CMake Error at svxlink/svxlink/cmake_install.cmake:100 (MESSAGE):
  Error while changing owner of file /var/spool/svxlink/qso_recorder
Call Stack (most recent call first):
  svxlink/cmake_install.cmake:38 (include)
  cmake_install.cmake:41 (include)

Makefile:66: recipe for target 'install' failed
make: *** [install] Error 1

pe1chl, 2015-06-03:
You did not take this (quoted from the docs) into account:

You will also need to create a system user called svxlink before installing if it does not already exist. The daemon group is also needed so if that does not exist it must also be created.

rneese45, 2015-06-03:
I am willing to put this arm board on a dmz port and let some one in and look and test.

rneese45, 2015-06-03:
I just tried the git repo version same error 
-- Setting owner of /var/spool/svxlink/qso_recorder to svxlink.daemon...
/bin/chown: invalid user: ‘svxlink.daemon’
CMake Error at svxlink/svxlink/cmake_install.cmake:100 (MESSAGE):
  Error while changing owner of file /var/spool/svxlink/qso_recorder
Call Stack (most recent call first):
  svxlink/cmake_install.cmake:38 (include)
  cmake_install.cmake:41 (include)

rneese45, 2015-06-04:
This should be done in the install script. Also I am working on system
startup scripts and I will be looking at working on Debian pkgs

rneese45, 2015-06-04:
you should add a line to the wiki for the full line command for adding in the user properly. 
that you currently use so people learning to use the software and build do it properly.

also the init scripts are not systemd compatible 

oot@kb3vgw-repeater:/etc/init.d# ./svxlink start
[....] Starting svxlink (via systemctl): svxlink.serviceFailed to start svxlink.service: Unit svxlink.service failed to load: No such file or directory.

I should have them done this week for you . as now that systemd is the default for jessie

pe1chl, 2015-06-04:
Please don't keep hammering on the same issue here in the issue tracker, when you want to discuss something please use the mailinglist.  The issue tracker is (mainly) for registering issues and the steps taken to resolve them.
Do you have difficulties reading English, do you have limited knowledge of Linux, or are there other circumstances that we should know when trying to help you?
It is characteristic for Linux that it is diffucult to almost impossible to provide ready recipes for this because in this area (directory structure, system administration including software installation and user maintenance, service startup) there are differences between Linux distributions and what may be working on Debian may not work on other distributions and probably not even on other Debian versions.
So the instructions are only to the level where you as the installer yourself should be able to pick them up and apply them to the system you have in front of you.  When you cannot do that, you should either learn how to do it, or use prepared binary packages made by people who can.

rneese45, 2015-06-04:
well the new issue with systemd I will move to a diff post as its a diff issue . I have already begun writting/updating the 13.x debian that some one forked off and trying to bring it to par so debian can have pkgs. 

but the systemd missing files will hurt ubuntu and debian and any oither system using the new systemd 

I have figured out the string for making/adding the user for now and have added it to our install script.

sm0svx, 2015-06-06:
Not a bug. Closing issue.

closed 2015-06-06 by sm0svx