sm0svx/svxlink #145: RemoteTrx assertion on network failure

sm0svx, 2015-06-08:
This issue has been created in response to [this discussion thread on the mailing list](http://sourceforge.net/p/svxlink/mailman/svxlink-devel/thread/552A9FFF.7080609%40amsat.org/#msg33792472) started by @pe1chl.

RemoteTrx will fail with an assertion when a disconnect occur due to network problems.

pa1okz, 2015-09-15:
Last sunday, during a large Dutch HAM event, our repeaters PI3UTR and PI2NOS had a major role and have been in use for over 8 hours full stop (~700 QSO's). During this event, Remotetrx crashed several times due to network congestion at three (or four) different RemoteTrx's. Some are in use by the prime operators, causing all traffic to stop. We'd like to ask for some attention to this issue since it influences the repeater performance dramatically.

sm0svx, 2015-09-27:
This problem should be fixable since I have a way to reproduce it. I have just been very busy with other stuff since the beginning of the summer. I'll try to have a look at it today or at least the coming week.

sm0svx, 2015-09-27:
Please try the new issue145 git branch on some RemoteTrx node where the problem occur often. The problem should be solved in that branch.

pe1chl, 2015-09-27:
Ok, switching to that version on the remotetrxes now to see if we still observe that issue.

pe1chl, 2015-09-28:
Up to now it is looking good!   Thanks for your effort.  We'll keep monitoring, but one remote
had several network failures today and the remotetrx has not crashed.

sm0svx, 2015-09-29:
Sounds good! We'll let it run a couple of more days like that and if it doesn't crash, and no new ill effects are shown, we can close this issue at last.

pa1okz, 2015-09-29:
Unfortunately I found a way to reproduce the crash at a particular location last night, but lets first repeat this some times and figure out some more details, including a way to enable you to reproduce it yourself as well...

pe1chl, 2015-09-29:
It looks like the crash still happens when the network is lost while that particular remotetrx is the currently selected receiver.  The previous report (yesterday) was about a receiver that quite often lost the link but was not actively receiving at that time.  It signaled a heartbeat timeout but recovered OK:

Sun Sep 27 21:24:42 2015: RxNOS: The squelch is OPEN (47.1745)
Sun Sep 27 21:24:43 2015: RxNOS: The squelch is CLOSED (45.1478)
Sun Sep 27 21:24:49 2015: *** ERROR: Heartbeat timeout in NetUplink NetRxNOS
Sun Sep 27 21:24:49 2015: NetRxNOS: Client disconnected: 44.137.42.60:56343
Sun Sep 27 23:19:18 2015: NetRxNOS: Client connected: 44.137.42.60:59890
Sun Sep 27 23:19:18 2015: RxNOS: SetMuteState(CONTENT)
Sun Sep 27 23:19:18 2015: NetRxNOS: Using CODEC "OPUS" to encode RX audio
Sun Sep 27 23:19:18 2015: ------ Opus encoder parameters ------
Sun Sep 27 23:19:18 2015: Frame size           = 320
Sun Sep 27 23:19:18 2015: Complexity           = 10
Sun Sep 27 23:19:18 2015: Bitrate              = 32000
Sun Sep 27 23:19:18 2015: VBR                  = YES
Sun Sep 27 23:19:18 2015: Constrained VBR      = YES
Sun Sep 27 23:19:18 2015: Maximum audio bw     = MEDIUMBAND
Sun Sep 27 23:19:18 2015: Audio bw             = FULLBAND
Sun Sep 27 23:19:18 2015: Signal type          = VOICE
Sun Sep 27 23:19:18 2015: Application type     = AUDIO
Sun Sep 27 23:19:18 2015: Inband FEC           = NO
Sun Sep 27 23:19:18 2015: Expected Packet Loss = 0%
Sun Sep 27 23:19:18 2015: DTX                  = NO
Sun Sep 27 23:19:18 2015: --------------------------------------
Sun Sep 27 23:19:41 2015: RxNOS: SetMuteState(NONE)
Sun Sep 27 23:19:42 2015: RxNOS: SetMuteState(CONTENT)

The receiver that Mischa was reporting on lost the link while it was the active receiver.  One time it was OK and the second time it crashed:

Mon Sep 28 23:09:05 2015: RxNOS: The squelch is OPEN (113.654)
Mon Sep 28 23:09:05 2015: RxNOS: SetMuteState(CONTENT)
Mon Sep 28 23:09:05 2015: RxNOS: SetMuteState(NONE)
Mon Sep 28 23:09:05 2015: RxNOS: SetMuteState(NONE)
Mon Sep 28 23:09:49 2015: NetRxNOS: Client disconnected: 44.137.42.60:50128
Mon Sep 28 23:09:49 2015: RxNOS: The squelch is CLOSED (113.448)
Mon Sep 28 23:09:49 2015: *** ERROR: TCP transmit error in NetUplink "NetRxNOS": 
Resource temporarily unavailable.
Mon Sep 28 23:09:49 2015: RxNOS: The squelch is OPEN (118.956)
Mon Sep 28 23:10:09 2015: NetRxNOS: Client connected: 44.137.42.60:50748
Mon Sep 28 23:10:09 2015: RxNOS: SetMuteState(CONTENT)
Mon Sep 28 23:10:09 2015: NetRxNOS: Using CODEC "OPUS" to encode RX audio
Mon Sep 28 23:10:09 2015: ------ Opus encoder parameters ------
Mon Sep 28 23:10:09 2015: Frame size           = 320
Mon Sep 28 23:10:09 2015: Complexity           = 10
Mon Sep 28 23:10:09 2015: Bitrate              = 32000
Mon Sep 28 23:10:09 2015: VBR                  = YES
Mon Sep 28 23:10:09 2015: Constrained VBR      = YES
Mon Sep 28 23:10:09 2015: Maximum audio bw     = MEDIUMBAND
Mon Sep 28 23:10:09 2015: Audio bw             = FULLBAND
Mon Sep 28 23:10:09 2015: Signal type          = VOICE
Mon Sep 28 23:10:09 2015: Application type     = AUDIO
Mon Sep 28 23:10:09 2015: Inband FEC           = NO
Mon Sep 28 23:10:09 2015: Expected Packet Loss = 0%
Mon Sep 28 23:10:09 2015: DTX                  = NO
Mon Sep 28 23:10:09 2015: --------------------------------------
Mon Sep 28 23:10:15 2015: NetRxNOS: Client disconnected: 44.137.42.60:50748
Mon Sep 28 23:10:15 2015: RxNOS: The squelch is CLOSED (115.27)
Mon Sep 28 23:10:15 2015: RxNOS: The squelch is OPEN (112.606)
Mon Sep 28 23:10:22 2015: RxNOS: The squelch is CLOSED (112.08)
Mon Sep 28 23:10:32 2015: NetRxNOS: Client connected: 44.137.42.60:50759
Mon Sep 28 23:10:32 2015: RxNOS: SetMuteState(NONE)
Mon Sep 28 23:10:32 2015: NetRxNOS: Using CODEC "OPUS" to encode RX audio
Mon Sep 28 23:10:32 2015: ------ Opus encoder parameters ------
Mon Sep 28 23:10:32 2015: Frame size           = 320
Mon Sep 28 23:10:32 2015: Complexity           = 10
Mon Sep 28 23:10:32 2015: Bitrate              = 32000
Mon Sep 28 23:10:32 2015: VBR                  = YES
Mon Sep 28 23:10:32 2015: Constrained VBR      = YES
Mon Sep 28 23:10:32 2015: Maximum audio bw     = MEDIUMBAND
Mon Sep 28 23:10:32 2015: Audio bw             = FULLBAND
Mon Sep 28 23:10:32 2015: Signal type          = VOICE
[... 24 lines cut ...]
Mon Sep 28 23:11:04 2015: Signal type          = VOICE
Mon Sep 28 23:11:04 2015: Application type     = AUDIO
Mon Sep 28 23:11:04 2015: Inband FEC           = NO
Mon Sep 28 23:11:04 2015: Expected Packet Loss = 0%
Mon Sep 28 23:11:04 2015: DTX                  = NO
Mon Sep 28 23:11:04 2015: --------------------------------------
Mon Sep 28 23:11:05 2015: RxUTR: SetMuteState(NONE)
Mon Sep 28 23:11:09 2015: RxNOS: SetMuteState(NONE)
Mon Sep 28 23:14:26 2015: RemoteTrx v1.1.99.23 (Sep 27 2015) Copyright (C) 2003-2015 Tobias Blomberg / SM0SVX

(there it crashed and was restarted...  of course no abort message because it was logging to file)

What was done here to trigger it is: insert/delete a firewall rule at the svxlink site that reject traffic to that particular remotetrx:

iptables -A OUTPUT -d $HOST -p tcp --dport $TCP_PORT -j REJECT --reject-with tcp-reset

This causes the connection at the svxlink site to be lost and the remotetrx will get a connection reset when it sends some data.  That is not much different from a link loss:

[... cut ...]
At the big event we had earlier in the month I wrote a small script that can insert/delete the above firewall rules as a crude way of managing the remote receivers, e.g. when some jerk would keyup a strong signal on one receiver and would block the entire system during the event (and in fact that happened, so it was useful).   Of course we don't use that in normal operation, but here it can be used to simulate network problems.   And we already noticed during the test that sometimes the remotetrx crashed when this firewall trick was used.

What we really need is some dynamic management of the system, but that is another topic.

sm0svx, 2015-09-29:
Doh!
Trying to reproduce it here but no luck so far :-/

sm0svx, 2015-10-03:
Your last description sounds very much like the bug I fixed. That is, crashes happening when the receiver is the active/selected one and a TCP disconnection occur. I have not been able to reproduce such a crash after I fixed the code.

Are you really really sure that you are running the correct software version? I have made some more commits on the issue145 branch so update your code and recompile. Then check the version for RemoteTrx. It should be 1.1.99.25-issue145. You may need to run "cmake" manually before "make" to make the correct version show up.

There are no additional fixes in the new code but I have added a simple simulator for receivers. It's in two parts, a new receiver type "LocalSim" and a new signal level detector "SIM". The LocalSim receiver type just play a tone continuously but otherwise behave like a Local receiver type. The SIM signal level detector has a little more functionality. It can randomize signal level changes in a couple of different ways. Have a look in the manual page. If you can reproduce a bug using the simulator I should also be able to reproduce it here.

pe1chl, 2015-10-03:
When the problem happened again I double-checked that we are running the updated code.
git branch
* issue145
  master
we then recompile using a script:

    #!/bin/bash
    # rebuild svxlink after "git pull" or "git checkout" and install/start it.
    # this script is to be invoked while in the svxlink directory or in the src
    # directory below that.  when a parameter is passed, the build directory
    # is removed first so a clean rebuild is done.
    # Rob PE1CHL
    
    case `pwd` in
        */svxlink/src/build)
        cd ..
        ;;
        */svxlink/src)
        ;;
        */svxlink)
        cd src
        ;;
        *)
        echo "Please change to svxlink directory first"
        exit 1
        ;;
    esac
    
    if [ ! -r CMakeLists.txt ]
    then
        echo "Somehow we ended up in the wrong directory:" `pwd`
        exit 1
    fi
    
    if [ -n "$1" ]
    then
        echo "Removing old build directory"
        rm -rf build
    fi
    
    if [ ! -d build ]
    then
        echo "Creating build directory"
        mkdir build
    fi
    
    cd build || exit 2
    
    echo "Running cmake..."
    cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc -DLOCAL_STATE_DIR=/var -DUSE_QT=no -DCMAKE_BUILD_TYPE=RelWithDebInfo .. || exit 2
    
    echo "Building new version..."
    make || exit 2
    
    echo -n "Install the new version now? (Y/n): "
    read answer
    if [ "$answer" == "n" ]
    then
        exit 1
    fi
    
    echo "Stop running services..."
    svxruns=`pidof svxlink`
    trxruns=`pidof remotetrx`
    
    /etc/init.d/svxlink stop
    /etc/init.d/remotetrx stop
    
    killall svxlink
    killall remotetrx
    
    echo "Removing old version..."
    
    rm -fv /etc/security/console.perms.d/90-svxlink.perms
    rm -fv /etc/udev/rules.d/10-svxlink.rules
    rm -fv /usr/bin/qtel /usr/bin/remotetrx /usr/bin/siglevdetcal /usr/bin/svxlink
    rm -fvr /usr/include/svxlink
    rm -fv /usr/lib/libasyncaudio* /usr/lib/libasynccore* /usr/lib/libasynccpp* /usr/lib/libasyncqt*
    rm -fv /usr/lib/libecholib* /usr/lib/liblocationinfo.a /usr/lib/libtrx.a
    rm -fvr /usr/lib/svxlink
    rm -fv /usr/share/applications/qtel.desktop /usr/share/icons/link.xpm
    rm -fv /usr/share/man/man1/qtel.1.gz /usr/share/man/man1/remotetrx.1.gz
    rm -fv /usr/share/man/man1/siglevdetcal.1.gz /usr/share/man/man1/svxlink.1.gz
    rm -fv /usr/share/man/man5/ModuleDtmfRepeater.conf.5.gz
    rm -fv /usr/share/man/man5/ModuleEchoLink.conf.5.gz
    rm -fv /usr/share/man/man5/ModuleHelp.conf.5.gz
    rm -fv /usr/share/man/man5/ModuleParrot.conf.5.gz
    rm -fv /usr/share/man/man5/ModulePropagationMonitor.conf.5.gz
    rm -fv /usr/share/man/man5/ModuleSelCallEnc.conf.5.gz
    rm -fv /usr/share/man/man5/ModuleTclVoiceMail.conf.5.gz
    rm -fv /usr/share/man/man5/remotetrx.conf.5.gz
    rm -fv /usr/share/man/man5/svxlink.conf.5.gz
    rm -fvr /usr/share/qtel
    rm -fv /usr/share/svxlink/events.d/*.tcl* /usr/share/svxlink/modules.d/*.tcl*
    rmdir -v /usr/share/svxlink/* /usr/share/svxlink
    rmdir -v /var/spool/svxlink/* /var/spool/svxlink
    
    if ! grep -q '^svxlink:' /etc/passwd
    then
        echo "Creating svxlink group and user..."
        groupadd -g 496 svxlink
        useradd -u 496 -g 496 -m -c "SvxLink Daemon" svxlink
    fi
    
    echo "Installing new version..."
    
    make install || exit 3
    ldconfig
    
    echo "Starting new version..."
    
    if [ -n "$svxruns" ]
    then
        /etc/init.d/svxlink start
    fi
    
    if [ -n "$trxruns" ]
    then
        /etc/init.d/remotetrx start
    fi
    
    echo "Done.  Please check the results!"
    exit 0

this time I invoked it without the "clean build" parameter.  do you advise to try that?
(it was cleanly built from the branch version just before issue145 and it did quite some compiles so I think it is OK)
will try the newer version soon.

sm0svx, 2015-10-03:
Starting with a clean build should not be necessary but it may be wise to do so anyway under the circumstances. If RemoteTrx show the correct version during startup it should be ok so double check that.

pe1chl, 2015-10-03:
Ok...  I count that we have 17 machines running svxlink now so it is slowly becoming time to think about some automatic admin like puppet or saltstack...
But I installed the newest version with a clean compile now on all systems.  Will keep an eye on it and try some stresstesting when it no longer fails by itself (the mentioned iptables thing).

sm0svx, 2015-10-04:
I finally got RemoteTrx to crash in my simulator using the issue145 branch. Unfortunately the generated stack trace scrolled off the screen and the core dump was not usable so I need to wait for another crash to happen. However, while waiting for that, valgrind seem to be giving me some interesting information to work with.

sm0svx, 2015-10-04:
OK. Let's try this. I found one more problem and after fixing that, RemoteTrx has been stable and valgrind has been quiet in my simulated environment.

pe1chl, 2015-10-04:
Ok thanks for your effort!  I have (incrementally) compiled it and installed.
Let's run it for a while and later try some induced network failures :-)

pe1chl, 2015-10-10:
No crashes yet...  not even with forced test :-)

sm0svx, 2015-10-10:
OK. That sounds good. I'd say we can merge this fix then since you reported no other ill-effects.

pe1chl, 2015-10-10:
Indeed, certainly no ill-effects.  To be certain it resolved the issue we have to wait some more, but I think it is OK.

sm0svx, 2015-10-11:
So now it's merged to master. I'm closing this issue. Reopen it if the problem reappears.

closed 2015-10-11 by sm0svx