sm0svx/svxlink #147: 10M repeater remote TX

JimZAH, 2015-06-11:
Hi Tobias,

I hope everything is well?

I'm currently setting up two Raspberry PI's for a new 10M project. One PI will be for receive and the other will be for the transmitter site. 

I'm having some difficulty getting the NetTx and NetRx to work locally. 

I receive the following error when running the SVXLink software:

NetTx: Disconnected from remote transmitter at 10.144.53.18:5210: Connection refused

The auth keys are correct and I can't see any firewalls running.

I hope you can help.

All the best

JimZAH, 2015-06-13:
Hi Tobias,

I have solved this after finding a fault with my settings.

Have a great weekend :-)

closed 2015-06-13 by JimZAH

f5vmr, 2015-07-18:
Jim, Please how did you resolve this, as I have the same problem except I have too raspberry side by side on the network, adjacent IP address, and the connections are refused. Passwords and ports are identical, and those ports are available in iptables in/out.

JimZAH, 2015-07-18:
Hi Sir,

At first I was running the wrong command "svxlink" and not "remotetrx". I then found it was a problem with my VPN network. Both sites have a VPN connection between them to stop unwanted snooping and prevent hijacking. 

Have you tried temporary disabling iptables just to be sure it's not a rule problem? 

I'm not an expert but I'd be happy to have a look at the config. My email address is me@m0zah.com.