sm0svx/svxlink #149: Support for BIND_ADDR in QTel

pe1chl, 2015-06-14:
Feature request: support a BIND_ADDR configuration parameter in QTel.
My system has several IP addresses.  One is on the 44.0.0.0/8 network and is connected via
radio, another one is connected to Internet via an ISP.  I would like to be on EchoLink
using my HAM address.
I have searched through the code and found that all the code for that is in there, and indeed it
is supported in the EchoLink module, but QTel is missing the code to get this setting from the
config and pass it to the relevant functions of echolib.
It is probably not a good idea to put a confusing "bind address" field in the configuration dialogs
seen by everone including the many users who have no need for this and don't understand it.
It is not clear to me if I can add a parameter that I can put in the config file using an editor.  Will
it remain in there when the settings are updated through the dialog?
An alternative would be to use a commandline parameter for this feature.

sm0svx, 2015-09-27:
I'm not sure I understand the need for this. You have both a "normal" internet connection and a 44-internet connection. If you connect to the standard EchoLink directory servers, everything will work fine through your ISP. If you connect through an EchoLink proxy on the 44-net, everything will go that way.

Is the problem that you want to connect to the standard EchoLink directory servers so that your traffic is routed through the 44-net?  Why?

About the config file, I would expect config written manually to the config file to stay there even after changes has been done using the GUI. I don't think it's a good idea though. All possible configurations should be reachable from the GUI. A proper tool-tip will hopefully keep people from changing the config if they don't know what they are doing.

pe1chl, 2015-09-27:
We have our repeaters on the AMPRnet network (what is called HAMNET here and in some other countries) which is routed via radio links.  For example, I have a WiFi link to the nearest repeater site (about 8km) and I route a /28 subnet over that.  When I connect to echolink using my address on that network, the link to the EchoLink directory servers will be routed to internet over a connection we have between HAMNET and internet, but the actual QSO will be all over HAMNET when the repeater I connect to is on HAMNET as well.
When I connect to the EchoLink directory server using my normal internet connection, the QSO will be routed over that connection as well and then back into HAMNET over that connection we have.
I prefer to use the HAMNET path (it is about amateur radio, isn't it?) so I would like to let QTel bind do my HAMNET address to achieve that.  For a while I had replaced the DNS name of the EchoLink directory server with a fixed IP in QTel, then hardwired a route into Linux that routes traffic to that internet destination over my HAMNET link (default is to route only 44.0.0.0/8 destinations over HAMNET) and thus selected my HAMNET source address for it, and that worked OK.  However, that way I missed that EchoLink had deployed extra directory servers at new addresses located much closer.  So I prefer not to have such hardwired assumptions about external servers.
As I agree it is not a good idea to trick the GUI config I suggested that alternative of parsing a command-line option for the bind address and passing that down into EchoLib.

sm0svx, 2015-09-27:
Ok. I understand.

Maybe you could implement a proof-of-concept by modifying https://github.com/sm0svx/svxlink/blob/master/src/qtel/MainWindow.cpp#L522-L526, hardcoding your bind address, just to make sure it will work as expected. It should be a simple change.

pe1chl, 2015-09-27:
I am convinced that it would work OK because it works OK in the echolink module that calls the same library with that extra parameter:

    ModuleEchoLink.cpp:  dir = new Directory(servers, mycall, password, location, bind_addr);

    MainWindow.cpp:   dir = new Directory(
      servers,
      settings->callsign().toStdString(),
      settings->password().toStdString(),
      settings->location().toStdString());

What makes it a bit tricky for me is to get the literal (ascii) address parsed into an IpAddress item in C++.

sm0svx, 2015-09-27:
The Async::IpAddress class will handle the conversion for you. Just add last in the constructor call: Async::IpAddress("44.1.2.3"). So the complete call would look something like:

```
  dir = new Directory(
      servers,
      settings->callsign().toStdString(),
      settings->password().toStdString(),
      settings->location().toStdString(),
      Async::IpAddress("44.1.2.3"));
```

pe1chl, 2015-09-27:
Ok thanks for the hint!
I tried that and it works.  I am now connected on my HAMNET address.
When you see no general need for such an option I can live with this local change.

sm0svx, 2015-09-27:
OK. Thanks for testing. It's always good to do a quick test before implementing something properly to really make sure it will work.

It should not be hard to implement but as you may imagine, with the voter problems at hand, this enhancement have lower priority. Maybe I'll just do it for fun some day though :-)

pe1chl, 2015-09-27:
Yes, of course it should be low priority :-)

sm0svx, 2015-10-18:
Now in master. It's not perfect but will have to do for now. The IP address to bind to can be selected in the configuration dialog from a combobox. The selected IP address is stored in the configuration file. The problem is if the IP address changes, like when using DHCP. Then the configured bind address cannot be found and the setting will be reset to use no binding.

closed 2015-10-18 by sm0svx

pe1chl, 2015-10-18:
Very nice!  It works fine, it shows all my addresses and interfaces, 8 items in the list :-)

sm0svx, 2015-10-18:
Impressive :-)

pe1chl, 2015-10-21:
There is one thing that could be enhanced: when selecting a bind address it does not use it for the UDP sockets.  It serves the purpose I had, but when this was added it would be possible to run svxlink and qtel alongside on a single system with two addresses.   I simply added this line:

    Dispatcher::setBindAddr(Async::IpAddress(settings->bindAddress().toStdString()));

before the line:

    Dispatcher *disp = Dispatcher::instance();

This does the job, but you might want to use a temporary variable to store the converted address that is also used in new Directory().

sm0svx, 2015-10-25:
Done!