sm0svx/svxlink #150: Ident from audio file

Emamisu, 2015-07-01:
Hi,
I'm Ema from Italy and I'm a newbie in the linux world and in the Svxlink software.
Is possible to play a custom ident from audio file?

Bye
E. IU2CIQ

f5vmr, 2015-07-02:
Check out the "mailing list"  in the main menu, where the answer to your problem may be found, and if not, pose your question there. There is a solution. Regards.

isladelobos, 2015-07-02:
Hello.

Yes you can.

You can replace the "long ident" audio file for personalized one.
You can create another audio file and add play timer
You can create a new audio file and pay this via dtmf
You can modify your "short ident" for play morse code instead the fonetic words.

Have many options, but what is your thinking method?

Regards
Roque.
  ----- Original Message ----- 
  From: Emamisu 
  To: sm0svx/svxlink 
  Sent: Wednesday, July 01, 2015 10:36 PM
  Subject: [svxlink] Ident from audio file (#150)

  Hi,
  I'm Ema from Italy and I'm a newbie in the linux world and in the Svxlink software.
  Is possible to play a custom ident from audio file?

  Bye
  E. IU2CIQ

Emamisu, 2015-07-02:
Hi Roque, thanks for the reply. I want send every 15 minute the ident of my
link played from audio file. I already have the audio file .wav but i dont
know how to do...

Emanuele Misurina - IU2CIQ
--------------------------------------------------------------------
Email: emanuele.misurina@gmail.com
Skype: emanuele.misurina
Tel.: 392/97.86.904

Questo messaggio e i suoi allegati sono indirizzati esclusivamente alle
persone indicate. La diffusione, copia o qualsiasi altra azione derivante
dalla conoscenza di queste informazioni sono rigorosamente vietate. Qualora
abbiate ricevuto questo documento per errore siete cortesemente pregati di
darne immediata comunicazione al mittente e di provvedere alla sua
distruzione, Grazie.

*This e-mail and any attachments** is **confidential and may contain
privileged information intended for the addressee(s) only. Dissemination,
copying, printing or use by anybody else is unauthorised. If you are not
the intended recipient, please delete this message and any attachments and
advise the sender by return e-mail, Thanks.*
*Rispetta l'ambiente. Non stampare questa mail se non è necessario.*

Emamisu, 2015-07-03:
Hi everyone!
I have put the audio file iu2ciq_ident.wav in usr/share/svxlink/sounds/en_US/Core.
In the /usr/share/svxlink/events.d/Logic.tcl i have changed so: 

proc send_long_ident {hour minute} {
#  global mycall;
#  global loaded_modules;
#  global active_module;
#  variable CFG_TYPE;
#
#  spellWord $mycall;
#  if {$CFG_TYPE == "Repeater"} {
#    playMsg "Core" "repeater";
#  }
#  playSilence 500;

  playMsg "Core" "iu2ciq_ident";
  playSilence 500;

The changes don't works and the software continues to send the default ident!! :(((

What is the file that I have to change for make my custom identification??

Ciao
Emanuele

dl1hrc, 2015-07-03:
are there some error messages while running the application?
BTW: There is an complete Italian synthetic voice set available for SvxLink (voice "Chiara") so you do not need to use the English version

73s Adi

Emamisu, 2015-07-04:
Hi Adi.
Thanks for the information about the Italian voice.
The software works well and gives no error despite the change of the
logic.tcl file. The software work like a no change had be made!

...
Ciao
Ema

>
> are there some error messages while running the application?
> BTW: There is an complete Italian synthetic voice set available for
SvxLink (voice "Chiara") so you do not need to use the English version

pe1chl, 2015-07-25:
Emamisu,
Please don't use the method of 'put a # in front of it' to remove lines of TCL code you do not want.
It is not valid TCL to do so.  It will work for simple lines like a playmsg call, but when the line includes
braces like { } it will not work like you expect it will.

sm0svx, 2015-09-27:
Is this still an issue?

sm0svx, 2015-10-10:
No answer. Closing issue.

closed 2015-10-10 by sm0svx