sm0svx/svxlink #153: NetTrx and NetRx

f5vmr, 2015-07-18:
There was an issue which has been closed where despite the compatibility between the net connected receiver and its net transmitter, the connection was refused. The query result was never published as the user resolved the issue himself.
I have this issue, where two raspberries on adjacent IP addresses on the same network sharing passwords and ports, refusing their connections despite listing the ports in iptables.
I have followed those relevant pages in the man, but perhaps there is an issue that I have missed, but I think not.

Published here in absence of the the mailing list pages.

f5vmr, 2015-07-18:
In a way you have answered my question but highlighted something that may or not be an issue. I have just installed a 50 MHz repeater running SVXLink, but have tried to use the input of the 145 MHz repeater also running SVXLink as a source of input using the techniques of NetTx and NetRx. I have also also switched in a voting receiver running RemoteTrx. The odd thing is that the remoteTrx unit is not being rejected, but the SvxLink unit is. I wonder if this is an issue.

f5vmr, 2015-07-19:
Rob, you posted a great solution, that I can no longer view due to the mailing list being off line, that of the use of RemoteTrx as a server for linking systems. Can you reproduce it here or mail me directly?
I am doing just that in France, but I am missing something, particularly when connecting to the server. Each of the repeaters has its own port and on the same network, but the full connection isn't happening, and I lost your code to reference it. Should have kept a copy - Doh!
Chris

pe1chl, 2015-07-24:
Hi Chris,
I am not reading the issue tracker all the time, but now I am as I wondered why there is no mail.
But that is cleared up now...
I see you want to use a single receiver as source for two different repeaters.  I have not done
exactly that, but I think you are referring to my alsa "audio plumbling" technique that I used to
connect both remotetrx and svxlink to a single soundcard, but each to a different channel.
The same technique can be used to connect 2 remotetrx instances to the same channel of the
same soundcard, although I am not sure it is necessary for that.  First try to setup a remotetrx.conf
that just defines two instances that are using the same soundcard but on a different LISTEN_PORT,
and connect the two different ports from your two places maybe that works.
When not, you can use the dsnoop plugin.  I use it like this:

    pcm.dsnoopCMI8738 {
      type dsnoop
      ipc_key 884234
      slave {
        pcm "hw:CMI8738,0" 
        channels 2 
        period_size 1024
        buffer_size 2048
        rate 48000
      }
    }

You can then use AUDIO_DEV=alsa:dsnoopCMI8738 in the config file.
Hmmm I now realize that this solves the problem of opening the same audio source in two
different programs (svxlink and remotetrx in my case), but it may do nothing to solve the
problem of having two remotetrx instances in the same remotetrx.conf that want the same card.
(but maybe that problem does not even exist)

sm0svx, 2015-09-27:
Hmmm... It seems to me that there are conversation missing in this thread. Maybe something went directly though email or I just don't understand something.

As Rob say, indeed it should work to share a singe receiver if you run *one* instance of RemoteTrx using *two* NetTrx-sections. You should be able to share a single Rx-configuration section between the two NetTrx:es. If you need to have different config for the Rx of the two NetTrx:es, make sure that the AUDIO_DEV=alsa:plughw:0 config item is *exactly* the same.

Resolved?

pe1chl, 2015-09-27:
You are right, there should be no issue when a single remotetrx opens the soundcard twice for two different channels, we do that all the time.
Normally I prefer to use :hw: instead of :plughw: when possible, to avoid missing mistakes that ALSA "solves" by inserting a conversion, but of course those two should not be mixed in a single config.

sm0svx, 2015-09-27:
For the same receiver it should be possible to use both the same audio device as well as the same audio channel. The SvxLink audio code should be able to handle multiple opens of the same audio device and channel, as long as the AUDIO_DEV specification is exactly the same. Hence, it should be possible to use the same Rx config section from two different NetTrx configs.

At least that should work for the audio device. There may be other configs that will conflict like some squelch types. For example, a serial pin squelch should be able to share but I'm not sure with the other types. There may be other things requiring exclusive access. The easiest way is to just try it out.

pe1chl, 2015-09-27:
Yes, we are often sharing the soundcard for 2 receivers (L and R channel) and the serial port for 3 receivers (CTS, DSR and RI) and 2 transmitters (RTS, DTR) without any problem, while using DCD/RXD/TXD for time synchronization, all on the same port.  As long as all config items are written exactly the same it works fine.
I not so sure what the original issue was about, probably not this thing but something related to trouble he had on the network that we discussed as well.

sm0svx, 2015-09-27:
It was not clear to me either what the problem was but if @f5vmr was trying to connect two nodes to the same RemoteTrx (on the same TCP port) that will not work because the second client will be rejected since one NetTrx can only handle one client. On the client side one would just see a disconnection message but looking at the log on the RemoteTrx side would give some more information.

f5vmr, 2015-09-27:
Hi Everybody,

The current situation with our two repeaters 50 MHz and 145 MHz is that they
are connected permanently using a configuration of SVXServer provided by Adi
DL1HRC and it works 99.5% of the time. Both repeaters have a blanked
repeater session ( no callsign) with a server using the same port in each
instance. The 0.5% is when the server locks up and has to be restarted. In
fact I would say that it has been on continuous operation now for 5 days
with no problem. A user transmits simultaneously on both repeaters when
active on either of them.

Pity the propagation has not been good to test at distance, but the
experiment is very successful.

 

Regards

Chris

sm0svx, 2015-09-27:
So this issue can be closed?  It's important that resolved issues are closed so that we really can concentrate on the real issues.

If you have problems with SVXServer you should clearly indicate that in the title of a new issue. It would also be good to indicate that SVXServer is a branch that at the moment only exist in the @dl1hrc clone of SvxLink.

f5vmr, 2015-09-27:
Thanks Tobias,

The issue can be closed.

 

Regards

Chris

closed 2015-09-27 by sm0svx