sm0svx/svxlink #154: Documentation Error - LinkWithLocalMicAndSpeaker

mmiller7, 2015-07-19:
In the following guide, the location for TCL files is incorrect.
https://github.com/sm0svx/svxlink/wiki/LinkWithLocalMicAndSpeaker

TLDR-version, all the paths in this guide where /usr/share is referenced incorrectly point to svxlik/sound instead of just svxlink directory.

Details:

References to the TCL Logic Core are "/usr/share/svxlink/sounds/events.d/" which does not exist.
I think it should be either "/usr/share/svxlink/events.d" for the source "SimplexLogic.tcl" and either the same or /usr/share/svxlink/events.d/local" for the new "MicSpkrLogic.tcl" file.

Similarly, the EVENT_HANDLER path in the guide for "EVENT_HANDLER=/usr/share/svxlink/sounds/events.tcl" should be fixed to reflect the proper location, which I believe should be "EVENT_HANDLER=/usr/share/svxlink/events.tcl".

sm0svx, 2015-09-27:
Thanks for pointing that out. Better now?

mmiller7, 2015-09-27:
Looks much better now!

closed 2015-09-27 by sm0svx