sm0svx/svxlink #155: Help installing SVXLINK 14.08.1 in Raspberry B + using the GPIO ports

ripardo, 2015-07-27:
Hello dear! My name is
Airam, I am amateur with
the indicative PU8ASR. I
live in Manaus city, in Brazil and I'm trying
to do svxlink work properly
in my raspberry + b. Ja I
used several examples, but
none managed to solve my
problem. Do not blame the
examples, but my lack of
knowledge. Some time
managed to make Linux to
thebridge work and posted
a video / tutorial teaching
to do and this contributed
to other colleagues to gain
performance in his
lectures. Now similarly
seek information that
enables me to do my
raspberry b + function
using the GPIO ports to
power the transceiver PTT
Yaesu FT-1802M. The
interface is already ready.
So I really need to work so
I can provide step-by-step
tutorial in my language to
help other colleagues and
disseminate / popularize
this solution using the
raspberry in my country. If
anyone can help me, you
can contact me by email
or by pu8asr@qsl.net
whatsapp
+55092982763774 Thanks!
73

f5vmr, 2015-08-28:
Help provided - Station is now operational.
Chris F5VMR

pu8asr, 2015-09-17:
Big Chris F5VMR
Thank you for help
Thanks buddy

f5vmr, 2015-09-17:
You are very welcome J

ch0mik, 2015-09-17:
try this (ubuntu @ Cubieboard)

sudo apt-get update
sudo apt-get install -y g++ make libsigc++-dev sigc++ libgsm1-dev
libpopt-dev tcl8.5-dev libgcrypt-dev libspeex-dev libasound2-dev alsa-utils
build-essential cmake

sudo adduser svxlink
sudo adduser svxlink daemon

cd ~/
git clone https://github.com/sm0svx/svxlink.git
cd svxlink/src
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc
-DLOCAL_STATE_DIR=/var -DUSE_QT=no ..
make NO_CHOWN=1 install
sudo make install
sudo ldconfig

#modify first the /etc/svxlink/svxlink.conf and similar conf files
sudo svxlink --config /etc/svxlink/svxlink.conf

Vy 73! de SQ7MRU, Pawel
https://sq7mru.blogspot.com/

closed 2015-09-27 by sm0svx