sm0svx/svxlink #156: Need option to change DTMF commands

mmiller7, 2015-08-04:
I am trying to use svxlink as a link for a repeater, however it seems the DTMF commands for the repeater control and svxlink control are interfering with each other.  For example, bringing up the phone-patch (#) causes echolink users to be disconnected, trying to answer a call (*#) triggers svxlink full long identification and status message.  This particular controller is a non-modifier PROM chip.

In my case, the repeater does not have the ability to change the commands it uses, I am hoping to add an extra command to svxlink.  For example, if there was a way to punch up AB# and have SVXLink ignore all following DTMF commands (but keep doing whatever it is) until CD# was sent this would be great.

pe1chl, 2015-08-08:
You can code that as part of the Logic.tcl script.
Make a copy of that script in the subdirectory local, add a global variable to enable/disable DTMF,
and in proc dtmf_cmd_received add code to manipulate that variable using those special commands
and when something else comes in AND your variable is set, return 1.

sm0svx, 2015-09-27:
Maybe switching the repeater logic to SvxLink is possible?  That would solve your problem.

mmiller7, 2015-09-27:
I've been petitioning them to change to a different setup (or just make their repeater not pass DTMF) but they have a particular repeater controller they really want to use with phone-patch.  I have no idea why since their aim is a portable rig to transport to events.

The Logic.tcl tweak sounds like it could work but I have to learn how to do that first...I didn't realize tcl was an actual programming language, I thought it was some sort of config script specific to this app.

pe1chl, 2015-09-27:
This would be something like:

    variable dtmf_disabled 0;

    proc dtmf_cmd_received {cmd} {
      global dtmf_disabled;

      if {$cmd == "AB"} {
        set dtmf_disabled 1;
        return 1;
      }

      if {$cmd == "CD"} {
        set dtmf_disabled 0;
        return 1;
      }

      if {$dtmf_disabled} {
        return 1;
      }
      ....  here follows the original version of dtmf_cmd_received

This is untested but something like this should work.

mmiller7, 2015-09-27:
Hmmm, cool thanks!  I'll have to play with that a bit.

I don't suppose you have any good references for TCL online I can also refer to?  I'm used to Java and C/C++...this looks more like Javascript (which I absolutely hate).

pe1chl, 2015-09-27:
It is a language with some funny aspects...  one thing to warn against is to use the # (comment)
symbol at the start of a line to "comment out" lines of code.  This works OK when the line is a
simple statement like a playMsg or a set statement, but when it is a compound statement like
an if statement it may fail in a totally unexpected way, leading to unexplainable error messages
about unmatched braces etc.   This is because the # sign is actually treated as a statement.
So it is better not to get into the habit of doing that.

I am no expert either but when googling I found a site that had a programmers manual.  Now
trying that again I don't seem to find the same site, but the primary place to look for is
http://www.tcl.tk/ which also has a WiKi at http://wiki.tcl.tk/ and has a lot of links to docs.

mmiller7, 2015-09-27:
Ah!  Maybe that's part of my problem...coding practices I've learned usually goes like "paste in example commented out and put your own code below each comment to modify it" as a way to learn.  I've had dismal success so far making it work at all.

Thanks so much!

sm0svx, 2015-09-27:
You can disable a block of code in TCL safely using
```
proc COMMENTED_OUT {} {
  ...the disabled code...
}
```
It's kind of strange but it works. See http://stackoverflow.com/questions/3368845/how-put-block-comments-in-tcl.

I've thought many times about replacing TCL with something more modern (like lua or Python) but TCL will have to do for now. It's quite established in the SvxLink community as well so replacing it would probably not make all users happy.

pe1chl, 2015-09-27:
This kind of works, but even with that construct you cannot just disable a block of incorrect code that you are still working on!  Some people want to experiment with code, cannot get it working because they are not fluent in the TCL syntax, then want to disable it again until they find more time to work on it.   Just putting that proc construct around it will not work when the number of braces is unbalanced or there are other syntax errors.
Or sometimes people want to temporarily remove an if statement around some piece of code.  E.g.:

    if {$a == $b} {
       playMsg "a";
       playMsg "b";
    }

It is often customary to just put a comment sign in front of the if and another one in front of the closing brace } and just leave the two playMsg statements indented.  In TCL this leads to disaster...

Anyhow, TCL is fine as it is now, you just need to be careful and always look in the logfile after restarting svxlink (or start it interactively) to check for syntax errors.   And, when the error is about unmatched braces, don't pull your hair out when you have done something like the above.

mmiller7, 2015-09-27:
Ah!  That commenting out the IF check is something I do VERY frequently.

As for changing it...I would say leave it be unless there's something majorly better.  I have worked with Lua before at my software-engineering job, it's made the list of "things I hope I never have to work with again".

pe1chl, 2015-09-27:
Ok so be warned...  another pitfall is that because svxlink always first reads the standard version of the script, and then applies your local copy with mods on top of that.  When you make some mistake that results in a syntax error in your local copy, there is a tendency to automatically fall back to the standard version of the same proc.   When you have made only minor changes, that can lead to very mysterious behaviour as well.  (especially when not checking the log for errors)

closed 2015-09-27 by mmiller7