sm0svx/svxlink #158: Echolink audio muted during ID's + RF activity

pa1okz, 2015-09-15:
In some cases when an Echolink station is 'transmitting', it sometimes happens that a scheduled ID (either spoken or CW) occurs. If at that particular moment a short RF transmission occurs (glitch or maybe unintended TX or so), the ID continues but mutes the Echolink audio until the ID has ended. 

It is not a big deal since ID's are short, but it came to my attention more then once already and therefore reported here as a defect.

sm0svx, 2015-09-27:
Is the EchoLink audio completely muted or just lowered in volume?  If completely muted, does any audio get dropped or is it queued so that nothing is really missed?

pa1okz, 2015-09-27:
Hello Tobias, long time no 'see' ;-). The audio is fully muted. I suppose you mean echolink audio that would be dropped. If that is what you mean, yes, this is the case.