sm0svx/svxlink #159: Remotetrx usb souncard issue

gicardous, 2015-10-03:
Hi,
System running svxlink and two remotetrx-es. Svxlink+remotetrx on same machine, one remotetrx on a raspberry pi.
Sometimes, randomly, remotetrx get stuck on transmit with the following error in log:
Sat Oct  3 09:23:45 2015: *** ERROR: snd_pcm_prepare failed (unrecoverable error): No such device
Sat Oct  3 13:09:17 2015: *** ERROR: snd_pcm_prepare failed (unrecoverable error): No such device
Sat Oct  3 13:24:17 2015: *** ERROR: Transmitter Tx1 have been active for too long. Turning it off...
On svxlink side this signals voter that the squelch is open so second remotetrx is transmitting also.
Restarting only remotetrx on rsp clears the problem.
Usb souncard is on a usb hub on that raspberry pi so I'm pretty sure that is not a remotetrx issue; I will use for audio output rsp's integrated soundcard; however it will be better if remotetrx could handle that problem in a better way.
Thank you.
Catalin yo7gqz

pe1chl, 2016-01-01:
I think it will not be an svxlink issue, as svxlink does nothing more than using soundcards via the Alsa interface.
There may be a bug in the driver you are using, and it may be that it is being stressed by svxlink more than by other programs because svxlink opens and closes sound devices every time something has to be transmitted.   There is an enhancement request #96 to remove that behaviour, maybe your sound device behaves better once that is implemented.

f5vmr, 2017-01-15:
I have seen this error relating to incorrect seating of the USB sound card or it being disconnected or partly dislodged during operation. Correct reseating of the card must be followed by a restart of the software.

sm0svx, 2017-01-16:
Yes, the cause of this problem is very likely a hardware failure. SvxLink could probably handle it better but it should not occur if the hardware is working properly.

sm0svx, 2019-07-15:
I'm closing this issue. Please reopen if you still think it is a bug in SvxLink.

closed 2019-07-15 by sm0svx