sm0svx/svxlink #160: git repo is broken

rneese45, 2015-10-05:
it seems you have missing files and when you do a git pull with any of the clients ast of 2 days ago git ppull fails to complete. it does nto state what files are missing but it failed to complete a full clone.

I am workign to fix a few issues I have found with typos and other issues but if I cant pull the src I cant patch and push a pull request

rneese45, 2015-10-05:
git did not exit cleanly (exit code 128) (20031 ms @ 10/5/2015 12:52:24 PM)

rneese45, 2015-10-05:
OK it seems you have a bad commit somewhere . I pulled the tarball and its even missing files that show  in the git repo. 
the svxlink/src/doc/man/svxlink.conf.5 file is missing.

sm0svx, 2015-10-05:
Cannot see that there is anything wrong. Either it was a temporary GitHub problem or you have some problem on your end. This looked perfectly normal just now:
```
$ git clone git://github.com/sm0svx/svxlink # (or https://github.com/sm0svx/svxlink.git)
Cloning into 'svxlink'...
remote: Counting objects: 23805, done.
remote: Compressing objects: 100% (57/57), done.
remote: Total 23805 (delta 25), reused 0 (delta 0), pack-reused 23748
Receiving objects: 100% (23805/23805), 6.39 MiB | 755.00 KiB/s, done.
Resolving deltas: 100% (16420/16420), done.
Checking connectivity... done.
$ ls svxlink/src/doc/man/svxlink.conf.5
svxlink/src/doc/man/svxlink.conf.5
```
It also builds just fine.

rneese45, 2015-10-05:
I have tried pulling on 5 diff machines
even pulling the zip file from the git hub files are missing

rneese45, 2015-10-05:
root@OdroidC1:~# ls svxlink/src/svxlink/scripts/en_US-heather/Echolink/
ls: cannot access svxlink/src/svxlink/scripts/en_US-heather/Echolink/: No such file or directory

sm0svx, 2015-10-06:
Very strange indeed. I tried the zip as well and it worked just fine as well. Could it be that we access different region servers at GitHub and that one of them have a problem? In what country are you located?

If anyone else have this problem, please write a comment.

sm0svx, 2015-10-06:
Unlikely since you tried on five different machines but..., a 100% filled file system could probably cause that problem. Or maybe a filesystem that does not like the asterisk in the filename (help_subcmd6*.txt). Are you using some remote filesystem?  Windows share or something? Please provide some more information on your setup (linux distro, filesystem config etc).

sm0svx, 2015-10-10:
The git repo is not broken but contains files with filenames that is not compatible with Windows so it cannot be cloned to a Windows machine. This is being addressed by #163 so closing this issue.

closed 2015-10-10 by sm0svx