sm0svx/svxlink #169: make docs issues

rneese45, 2015-10-19:
as of this am . I did a new test build with your current changes. this is the output from make docs.

Warning: Tag `SYMBOL_CACHE_SIZE' at line 310 of file `/usr/src/svxlink/src/build/doc/doxygen.echolib' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `SHOW_DIRECTORIES' at line 519 of file `/usr/src/svxlink/src/build/doc/doxygen.echolib' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `HTML_ALIGN_MEMBERS' at line 901 of file `/usr/src/svxlink/src/build/doc/doxygen.echolib' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `USE_INLINE_TREES' at line 1088 of file `/usr/src/svxlink/src/build/doc/doxygen.echolib' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `XML_SCHEMA' at line 1340 of file `/usr/src/svxlink/src/build/doc/doxygen.echolib' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `XML_DTD' at line 1346 of file `/usr/src/svxlink/src/build/doc/doxygen.echolib' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: doxygen no longer ships with the FreeSans font.
You may want to clear or change DOT_FONTNAME.
Otherwise you run the risk that the wrong font is being used for dot generated graphs.
Notice: Output directory `/usr/src/svxlink/src/build/doc/echolib' does not exist. I have created it for you.
Searching for include files...
Searching for example files...
Searching for files in directory /usr/src/svxlink/src/build/doc/echolib
Searching for images...
Searching for dot files...
Searching for msc files...
Searching for dia files...
Searching for files to exclude
Searching for files to process...
Searching for files in directory /usr/src/svxlink/src/build/include
Reading and parsing tag files
Parsing files
Preprocessing /usr/src/svxlink/src/build/include/EchoLinkDirectory.h...
Parsing file /usr/src/svxlink/src/build/include/EchoLinkDirectory.h...
Preprocessing /usr/src/svxlink/src/build/include/EchoLinkDispatcher.h...
Parsing file /usr/src/svxlink/src/build/include/EchoLinkDispatcher.h...
Preprocessing /usr/src/svxlink/src/build/include/EchoLinkProxy.h...
Parsing file /usr/src/svxlink/src/build/include/EchoLinkProxy.h...
Preprocessing /usr/src/svxlink/src/build/include/EchoLinkQso.h...
Parsing file /usr/src/svxlink/src/build/include/EchoLinkQso.h...
Preprocessing /usr/src/svxlink/src/build/include/EchoLinkStationData.h...
Parsing file /usr/src/svxlink/src/build/include/EchoLinkStationData.h...
Preprocessing /usr/src/svxlink/src/echolib/mainpage.dox...
Parsing file /usr/src/svxlink/src/echolib/mainpage.dox...
Building group list...
Building directory list...
Building namespace list...
Building file list...
Building class list...
Associating documentation with classes...
Computing nesting relations for classes...
Building example list...
Searching for enumerations...
Searching for documented typedefs...
Searching for members imported via using declarations...
Searching for included using directives...
Searching for documented variables...
Building interface member list...
Building member list...
Searching for friends...
Searching for documented defines...
Computing class inheritance relations...
Computing class usage relations...
Flushing cached template relations that have become invalid...
Creating members for template instances...
Computing class relations...
Add enum values to enums...
Searching for member function documentation...
Building page list...
Search for main page...
Computing page relations...
Determining the scope of groups...
[... 108 lines cut ...]
unable to read file loc from[EchoLink::StationData::setId]
unable to read file loc from[EchoLink::StationData::ip]
unable to read file loc from[EchoLink::StationData::setIp]
Generating Sqlite3 output for class sigc::trackable
Generating Sqlite3 output for class sigc::trackable
sqlite3_step failed: NOT NULL constraint failed: derivedcompoundref.base
sqlite3_step failed: NOT NULL constraint failed: derivedcompoundref.base
sqlite3_step failed: NOT NULL constraint failed: derivedcompoundref.base
sqlite3_step failed: NOT NULL constraint failed: derivedcompoundref.base
Generating Sqlite3 output for class EchoLink::Qso::VoicePacket
Generating Sqlite3 output for class EchoLink::Qso::VoicePacket
Generating Sqlite3 output for namespace EchoLink
Generating Sqlite3 output for file EchoLinkDirectory.h
Generating Sqlite3 output for file EchoLinkDispatcher.h
Generating Sqlite3 output for file EchoLinkProxy.h
Generating Sqlite3 output for file EchoLinkQso.h
Generating Sqlite3 output for file EchoLinkStationData.h
Generating Sqlite3 output for file mainpage.dox
Generating Sqlite3 output for dir /usr/src/svxlink/src/build/include/
Generating Sqlite3 output for example EchoLinkDirectory_demo.cpp
Generating Sqlite3 output for example EchoLinkDispatcher_demo.cpp
Generating Sqlite3 output for example EchoLinkProxy_demo.cpp
Generating Sqlite3 output for example EchoLinkQso_demo.cpp
Generating Sqlite3 output for the main page
lookup cache used 437/65536 hits=1935 misses=456

sm0svx, 2015-10-21:
Isn't this the same issue as #168?

sm0svx, 2015-10-21:
It is kind of hard to maintain a Doxygen configuration that work on all ranges of linux distributions. As long as it's only warnings we may have to live with them to be compatible over a wide range of distributions. However, it has been a while since I updated the config so the situation can possibly be somewhat improved.

sm0svx, 2017-11-04:
Doxygen warnings fixed in latest master as far as possible

closed 2017-11-04 by sm0svx