sm0svx/svxlink #171: sound card issue

rneese45, 2015-10-22:
when running svxlink audio randomly jumps left and right channels when in xmit mode.

I have tested 4 diff usb sound cards and 3 pci-e sound cards and it happened on all of them.

sm0svx, 2015-10-24:
Now that's something I have never heard of before. Since you have tested so many different sound devices, that's probably not what's wrong. Have your tried it on another computer?  On another linux distribution?

I'd recommend asking on the svxlink-devel mailing list and maybe you can find someone having the same problem.

rneese45, 2015-10-24:
Have tested on debian jessie. Tested on a odroid c1 plus a raspi and
bbblack. We mainly use arm on this project. As we are working on hardware
that mounts on pi2 and c1 . 3 different cards .

Card 1 echoing and simplex interface
Card 2 a full size board with multi input output
Card 3 board size of c1 and pi but limited in an out puts

sm0svx, 2015-12-01:
Have you found the cause for this issue?

rneese45, 2015-12-03:
Still Having issue with USB sound cards and audio jumping from left to right channel . using armhf boards. not using onboard sound chips.

rneese45, 2015-12-04:
Seems the issue has to do with the sounds cards and how well svxlink interacts with them . I went through testing 11 diff usb sound cars thus far and found svxlink has issues with 7 of them that the audio was distorted or jumped channels or would crash. so I will be listing what usb sound cards I have tested and what chips worked and did not work. once I finish testing.

pe1chl, 2016-01-01:
I think it will not be an svxlink issue, as svxlink does nothing more than using soundcards via the Alsa interface.
There may be a bug in the driver you are using, and it may be that it is being stressed by svxlink more than by other programs because svxlink opens and closes sound devices every time something has to be transmitted.   There is an enhancement request #96 to remove that behaviour, maybe your sound device behaves better once that is implemented.

kb3vgw, 2016-01-01:
It seems to be based on what chip and usb sound card used. And it seems
that on on arm boards it's more finaky.

Most usb 1.o sound cards have issues.  1.1 usb seem to work ok . I am
building a list of cards wich I have had issues with on the pi2 and odroid
c1 and c1+ boards

sm3sgp, 2017-01-25:
Any update on the soundcard testing?

kb3vgw, 2017-01-25:
Cm108 are the best

RichNeese, 2017-10-22:
we are now using boards with sound on them and dont seem to have the issue

closed 2017-10-22 by RichNeese