sm0svx/svxlink #172: svxlink doesn't like arch linux's gcc 5.2 environment

aleksmk, 2015-10-26:
Hi.

I'm trying to install svxlink from AUR but it fails to compile constantly. There are several problems that I've found so far : 

1. c++11 should be enabled during install with setting CXXFLAGS="-std=c++11 ..." before cmake
2. The static const floats should have the constexpr keyword in their definition
3. There is something wrong while linking the LinkManager.cpp.o, this is the output that I get : https://paste.softver.org.mk/95deaf0bcd887af8db9c3ffc2f1bda29

I'm using GCC 5.2.0, libsigc++ 2.6.1, tcl 8.6.4.

Regards and 73,
Z32AL

sm0svx, 2015-11-01:
SvxLink itself does not need C++11 but Arch recently upgraded libsigc++ to a version greater than 2.5.0, which require C++11.

The version in latest git master now support compiling with C++11 support turned on. The CMake scripts will automatically detect if a version greater than 2.5.0 of libsigc++ is installed and enable the -std=c++11 compiler switch.

As for the AUR package, I cannot fix that since it's up to the package maintainer to do that. Also, the AUR package is built from the official releases so I must create a new SvxLink release unless the maintainer want to backport the C++11 patch. It's all in commit e20aa278d82bc6822f3f68bf274e412c5945a064. I'm close to making a release now though so it may not be worth the effort to backport the fix.

aleksmk, 2015-11-03:
Hi,

I can confirm that the new changes fix the issue. I contacted the AUR package maintainer and he updated the package.

73 de Z32AL

closed 2015-11-03 by aleksmk

sm0svx, 2015-11-03:
Perfect!