sm0svx/svxlink #173: GPIO Squelch active low

gicardous, 2015-11-02:
Hi,
In svxlink using this configuration:
[Rx2Local]
TYPE=Local
AUDIO_DEV=alsa:plughw:2
AUDIO_CHANNEL=1
SQL_DET=GPIO
GPIO_SQL_PIN=!gpio24

In this case, I have no indication of squelch open/close.
However if I change config to:
GPIO_SQL_PIN=gpio24
I have this:
01.11.2015 19:07:05: Starting logic: SimplexLogic
01.11.2015 19:07:05: Loading RX: Voter
01.11.2015 19:07:05:    Adding receiver: RxLocal
01.11.2015 19:07:05:    Adding receiver: Rx2Local
01.11.2015 19:07:05: Loading TX: TxLocal
01.11.2015 19:07:05: SimplexLogic: Event handler script successfully loaded.
01.11.2015 19:07:05: 127.0.0.1:5230: RemoteTrx protocol version 2.4
01.11.2015 19:07:05: RxLocal: Connected to remote receiver at 127.0.0.1:5230
01.11.2015 19:07:05: RxLocal: Requesting CODEC "RAW"
01.11.2015 19:07:05: TxLocal: Connected to remote transmitter at 127.0.0.1:5230
01.11.2015 19:07:05: TxLocal: Requesting CODEC "RAW"
01.11.2015 19:07:06: Connected to APRS server 205.233.35.52 on port 14580
01.11.2015 19:07:06: Voter: The squelch is OPEN (Rx2Local=0)
01.11.2015 19:07:08: Voter: The squelch is CLOSED (Rx2Local=0)
01.11.2015 19:07:22: Voter: The squelch is OPEN (Rx2Local=0)
01.11.2015 19:07:24: Voter: The squelch is CLOSED (Rx2Local=0)
01.11.2015 19:07:39: Voter: The squelch is OPEN (Rx2Local=0)
01.11.2015 19:07:41: Voter: The squelch is CLOSED (Rx2Local=0)
01.11.2015 19:07:58: Voter: The squelch is OPEN (Rx2Local=0)

I could easily define inverted pin operation through active_low parameter in rsp's gpio config but for some reason I need to have this in svxlink software.
I know that is implemented for some time but it not seem to be working.
Please check.
Thank you!

sm0svx, 2015-11-02:
I must say it looks correct in the code and I have no test setup to test this on. Is the config above a direct copy/paste from the actual configuration file or did you write it in by hand? I'm wondering if there can be some subtle, non-obvious, error in your configuration file.

What version of the software are you running?

gicardous, 2015-11-03:
Hi,
this is a direct copy/paste from config/logs.

Let me show what I've tested.
Assuming that GPIO_SQL_PIN reads only value and does not care what type of pin is (in or out), I've echoed some values to that pin. First, config without '!'
GPIO_SQL_PIN=gpio24
cat /sys/class/gpio/gpio24/value
1
03.11.2015 20:47:17: SvxLink v1.4.99.29 (Oct 21 2015) Copyright (C) 2003-2015 Tobias Blomberg / SM0SVX
03.11.2015 20:47:17:
03.11.2015 20:47:17: SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
03.11.2015 20:47:17: welcome to redistribute it in accordance with the terms and conditions in the
03.11.2015 20:47:17: GNU GPL (General Public License) version 2 or later.
03.11.2015 20:47:17:
03.11.2015 20:47:17: Using configuration file: /etc/svxlink/svxlink.conf
03.11.2015 20:47:17: --- Using sample rate 16000Hz
03.11.2015 20:47:17:
03.11.2015 20:47:17: Starting logic: SimplexLogic
03.11.2015 20:47:17: Loading RX: Voter
03.11.2015 20:47:17:    Adding receiver: RxLocal
03.11.2015 20:47:17:    Adding receiver: Rx2Local
03.11.2015 20:47:17: Loading TX: TxLocal
03.11.2015 20:47:17: SimplexLogic: Event handler script successfully loaded.
03.11.2015 20:47:17: 127.0.0.1:5230: RemoteTrx protocol version 2.4
03.11.2015 20:47:17: RxLocal: Connected to remote receiver at 127.0.0.1:5230
03.11.2015 20:47:17: RxLocal: Requesting CODEC "RAW"
03.11.2015 20:47:17: TxLocal: Connected to remote transmitter at 127.0.0.1:5230
03.11.2015 20:47:17: TxLocal: Requesting CODEC "RAW"
03.11.2015 20:47:17: Voter: The squelch is OPEN (Rx2Local=0)
03.11.2015 20:47:18: Connected to APRS server 198.137.202.24 on port 14580

echo 0 >/sys/class/gpio/gpio24/value
03.11.2015 20:47:47: Voter: The squelch is CLOSED (Rx2Local=0)
echo 1 >/sys/class/gpio/gpio24/value
03.11.2015 20:48:02: Voter: The squelch is OPEN (Rx2Local=0)

So everything works as expected.
Now, with GPIO_SQL_PIN=!gpio24

03.11.2015 21:02:25: SIGTERM received. Shutting down application...
03.11.2015 21:02:29: SvxLink v1.4.99.29 (Oct 21 2015) Copyright (C) 2003-2015 Tobias Blomberg / SM0SVX
03.11.2015 21:02:29:
03.11.2015 21:02:29: SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
03.11.2015 21:02:29: welcome to redistribute it in accordance with the terms and conditions in the
03.11.2015 21:02:29: GNU GPL (General Public License) version 2 or later.
03.11.2015 21:02:29:
03.11.2015 21:02:29: Using configuration file: /etc/svxlink/svxlink.conf
03.11.2015 21:02:29: --- Using sample rate 16000Hz
03.11.2015 21:02:29:
03.11.2015 21:02:29: Starting logic: SimplexLogic
03.11.2015 21:02:29: Loading RX: Voter
03.11.2015 21:02:29:    Adding receiver: RxLocal
03.11.2015 21:02:29:    Adding receiver: Rx2Local
03.11.2015 21:02:29: Loading TX: TxLocal
03.11.2015 21:02:29: SimplexLogic: Event handler script successfully loaded.
03.11.2015 21:02:29: 127.0.0.1:5230: RemoteTrx protocol version 2.4
03.11.2015 21:02:29: RxLocal: Connected to remote receiver at 127.0.0.1:5230
03.11.2015 21:02:29: RxLocal: Requesting CODEC "RAW"
03.11.2015 21:02:29: TxLocal: Connected to remote transmitter at 127.0.0.1:5230
03.11.2015 21:02:29: TxLocal: Requesting CODEC "RAW"
03.11.2015 21:02:29: Connected to APRS server 129.15.108.113 on port 14580

echo 1 >/sys/class/gpio/gpio24/value
echo 0 >/sys/class/gpio/gpio24/value
echo 1 >/sys/class/gpio/gpio24/value
echo 0 >/sys/class/gpio/gpio24/value

Nothing happens, I have no indication of squelch opening or closing.
I hope that is more clear now.
I also use '!' for ptt but there are not problems.
Thank you.

sm0svx, 2015-11-03:
Ah, thanks! I didn't know one could use "echo" to test this. I'll try that.

sm0svx, 2015-11-03:
Now I see it! The logic was wrong. It should be corrected in Git master now.

gicardous, 2015-11-04:
It works as expected.
Thank you!
All the best! 73's de Catalin yo7gqz.

closed 2015-11-04 by gicardous

kb3vgw, 2017-02-18:
gpio  now use a gpio.conf and svxlink_gpio_up and svxlink_gpio_down scripts for setup and teardown of gpio pins.