sm0svx/svxlink #175: Latest Version - error in compile, version conflict

f5vmr, 2015-11-07:
I have cloned the latest version, but the modules EchoLink etc produce errors indicating that were compiled for version 1.4.1.9 when SVXlink is version 1.4.2.32. Did I miss something vital? Or was this a missed event in the drive to expedite the Debian issue before the deadline of termination?

sm0svx, 2015-11-09:
The installation path for SvxLink modules are now autodetected and depend on the architecture. The installation path has probably changed on your system so you are using the old modules with the new code.

You should remove the old modules in whatever directory MODULE_PATH point to. Then uncomment the MODULE_PATH config variable and SvxLink will find the correct path on its own.

f5vmr, 2015-11-09:
As you correctly pointed out, I deleted the older files, commented out the modules location in SVXlink.conf and rebooted. F5ZGM is in full operation once again. Thanks Tobias.
You may close the issue unless you think it may benefit others who choose to update.

closed 2015-11-09 by sm0svx