sm0svx/svxlink #176: v1.4.99.36 idle_timeout issue

DG1BBQ, 2015-11-16:
I just installed 1.4.99.36.

No matter what value idle_timeout is set to, the acutal timeout is always about 35 seconds.
Had 1.4.99.29 installed before with no issue.
Axel, DG1BBQ (Repeater DB0BHV in Bremerhaven)

sm0svx, 2015-11-16:
I have changed that code recently but it seems to work just fine here. It sounds like it's not picking up your setting of the IDLE_TIMEOUT configuration variable. Do you have something like this in your configuration file:

```
[RepeaterLogic]
...
IDLE_TIMEOUT=10
...
```

No other characters on the same row? All upper case?

DG1BBQ, 2015-11-17:
Hi Tobias.

This is the line containing IDLE_TIMEOUT:

**IDLE_TIMEOUT=10 # Timeout after 10s + IDLE_SOUND_INTERVAL=8000 means caution tone 2s before TX off**

The above is a single line with no line break.

It didn't cause an issue before. I've now put the comment on a separate line and the problem is gone.
So it looks like the changes you made are causing the parser to be less lenient. Now I know not to do that anymore. :)

Axel, DG1BBQ

closed 2015-11-17 by DG1BBQ

sm0svx, 2015-11-20:
Yes, SvxLink has never supported end of line comments but due to how some configuration variables are parsed, it works  anyway. Expect more end of line commented variables to stop working without notice.

Add a feature request if you'd like to see end of line comments implemented.

DG1BBQ, 2015-11-21:
Hi Tobias,
I don't think it's worth the effort. For my part, I can live with it as it is.
Maybe add a warning to the top of svxlink/remotetrx.conf in future releases.

BTW: v1.4.99.36 crashes, when an asterix is received. :)
73s, Axel DG1BBQ