sm0svx/svxlink #179: svxlink as aprs igate using locationinfo pty device

DW8AWC, 2015-11-29:
hello, i managed to use the location info pty path for injecting received aprs packet via rf but there seem to be so much unused message? here is the raw packets transmitted by svxlink

2015-11-29 22:20:08 PHT: DW8AWC-5>APDR13,WIDE1-1,qAS,DX8ZC:=0655.23N/12204.17E$ http://aprsdroid.org/DW8AWC-5>APDR13,WIDE1-1:=0655.23N/12204.17E$ http://aprsdroid.org/ER-DX8ZC>RXTLM-1,TCPIP,qAR,DX8ZC::ER-DX8ZC :UNIT.RX Erlang,TX Erlang,RXcount/10m,TXcount/10m,none1,STxxxxxx,logic
2015-11-29 23:34:32 PHT: DW8AWC-5>APDR13,WIDE1-1,qAS,DX8ZC:=0655.23N/12204.17E$ http://aprsdroid.org/DX8ZC>APRS,WIDE1-1:;ER-DX8ZC *111111z0658.90NE12205.93E0144.740MHz T203 R05k SPARC8 Repeater with APRS IGate (Rx Only) via SvxLink
2015-11-29 23:49:21 PHT: DW8AWC-5>APDR13,WIDE1-1,qAS,DX8ZC:=0655.23N/12204.17E$ http://aprsdroid.org/ER-DX8ZC>RXTLM-1,TCPIP,qAR,DX8ZC::ER-DX8ZC :UNIT.RX Erlang,TX Erlang,RXcount/10m,TXcount/10m,none1,STxxxxxx,logic

sm0svx, 2015-12-01:
@dl1hrc is the APRS guy :-)

dl1hrc, 2015-12-02:
hm, is it possible that you didn't sent \n\r at the end of the aprs message string?

DW8AWC, 2015-12-02:
@dl1hrc still the same even with \n\r at the end of the message string.
this is the format being received by svxlink
DW8AWC-5>APDR13,WIDE1-1:=0655.23N/12204.17E$ test message

sm0svx, 2015-12-02:
Just a thought... shouldn't it be \r\n (CRLF). That's the normal order but maybe it does not matter.

DW8AWC, 2015-12-02:
seems to be ok when receiving via pty. problem only occurs when its updating to the APRS-IS server.

Rx1: The squelch is OPEN (3.04416)
TxSpkr: Turning the transmitter ON
APRSTxSpkr: Turning the transmitter ON
DW8AWC-5>APDR13,WIDE1-1:=0655.23N/12204.17E$

TxSpkr: Turning the transmitter OFF
Rx1: The squelch is CLOSED (7.15264)
APRSTxSpkr: Turning the transmitter OFF

DW8AWC, 2015-12-02:
@sm0svx you are right. it should be \r\n and not \n\r. now everything works.. @sm0svx @dl1hrc thank you so much..73

2015-12-03 02:38:38 PHT: DW8AWC-5>APDR13,WIDE1-1,qAS,DX8ZC:=0655.23N/12204.17E$ 
2015-12-03 02:39:12 PHT: DW8AWC-5>APDR13,WIDE1-1,qAS,DX8ZC:=0655.23N/12204.17E$ aprsdroid
2015-12-03 02:43:02 PHT: DW8AWC-5>APDR13,WIDE1-1,qAS,DX8ZC:=0655.23N/12204.17E$ aprsdroid

closed 2015-12-02 by DW8AWC