sm0svx/svxlink #181: Async Issue in latest version

f5vmr, 2015-12-04:
svxlink: /home/pi/svxlink/src/async/audio/AsyncAudioSource.cpp:168: int Async::AudioSource::sinkWriteSamples(const float*, int): Assertion `len > 0' failed.

Running a new simple repeater, and in testing the audio with ModuleParrot engaged and called for by DTMF and running SvxLink in direct mode (not Daemon) this error is shown in the log list, the repeater locks in transmit and the raspberry no longer responds to command line entries.

The previous versions had no issues in this regard. Any ideas please?

Chris

f5vmr, 2015-12-04:
Today stupidity has entered the game - Somehow RepeaterLogic.tcl got deleted from my local folder.
Ignore this post.

sm0svx, 2015-12-05:
SvxLink should not crash just because a file is missing so I'd say this is still a bug. Is it easy to reproduce?

f5vmr, 2015-12-05:
Oh yes very easy. Especially when one deleted the whole of the svxlink directory in error. Works fine after a rebuild. Silly me!
Chris.

sm0svx, 2015-12-05:
An application should never throw an assertion like that unless there is a programming error in the application. Broken builds could also cause such errors but missing configuration files should never cause an assertion failure.

So, just to be clear, what directory did you delete? There are many directories called svxlink (/etc/svxlink, /usr/share/svxlink, the svxlink source/build directory etc).

Did you delete the directory while SvxLink was running?

f5vmr, 2015-12-05:
Fortunately it was a repeater under construction and not in service. I deleted the whole of the SVXLink directory containing the src etc. So when I set the repeater to run, there were no longer any cpp files to be called. That simple. Complete and utter human error. Marked down as a fail - Doh! 
No worries on the software, as I recompiled from scratch - and it works. I won't be doing that again!
Regards, Chris

sm0svx, 2015-12-05:
OK. We'll call it a user error for now then :-)

As a note, the cpp files are not "called" after SvxLink has been built. The build process translate the cpp source files into machine code, present in the svxlink binary and in the .so files. The source files are not needed after that.

closed 2015-12-05 by sm0svx

f5vmr, 2016-01-25:
Problem has not gone away - Transmitter remains on as the SVXlink crashes with this error. This is the latest git clone on a Raspberry PI B+ on 8 Gb SD Card. I cannot see anything that I have wrong.

svxlink: /home/pi/svxlink/src/async/audio/AsyncAudioSource.cpp:168: int Async::AudioSource::sinkWriteSamples(const float*, int): Assertion `len > 0' failed.

Any pointers would be gratefully received.

Chris

hb9pae, 2016-09-03:
Hello all

I do have the same error as f5mvr noted above.  (Raspberry PI, USB Soundcard, svxlink V 1.5.0)

> _svxlink: /home/pi/svxlink/src/async/audio/AsyncAudioSource.cpp:168: int 
> Async::AudioSource::sinkWriteSamples(const float*, int): Assertion `len > 0' failed.

How to solve this error? 

Thanks for any suggestion.

73 de Peter HB9PAE

f5vmr, 2016-12-03:
Just updating this situation. there is no change with this install. It is the only one that is doing it. When it crashes no errors are reported directly. I had to make a bash script that detected the crash, printing the error and restarting SVXLink. I have swapped USB Audio cards, updated the software, and checked the configuration files. So for the moment this is a bug.