sm0svx/svxlink #182: DTMF / CTCSS Decode problem with 15.11

f5vmr, 2015-12-11:
I have just installed a new repeater running 15.11 on Debian Jessie on an RPi.
No initial probems detected until I tinker with deemphasis in [Rx1].
WIth DEEMPHASIS=1 the ctcss is decoded succesfully but DTMF tones 2,3 and 6 pass the DTMF_MUTING=1 and will not decode. With DEEMPHASIS=0 CTCSS will not decode and DTMF tones are correctly interpreted.

I have tried several radios but the results are identical. My first thoughts were that it was the audio quality/pitch/volume, but I have tried reducing/increasing without any changes. Are there any others experiencing this issue?
Chris

gustavc39, 2015-12-11:
HI Chris,

   The RX audio is fed from the speaker or from the discriminator (flat 
audio)?  In the first case, the low frequencies associated with CTCSS 
tones will be highly filtered (attenuated) resulting in poor decoder 
performance or even preventing the decoder to work. Have you tried 
switching the audio input between MIC and LINE IN? The soundcard MIC 
input generally has an audio filter circuit that cuts audio outside 
vocal frequencies (such as CTCSS).
   As of my own experience, if you want to implement CTCSS in svxlink 
(most radios can do that on hardware too, but you will not have software 
control of that feature) you will need discriminator audio fed into the 
soundcard LINE IN input.  Some radios have a dedicated audio output for 
flat or unfiltered audio.  If your radio has a Packet audio output you 
can try this too.  The flat audio is generally not deemphasized so you 
will need to turn DEEMPHASIS=1 on svxlink.
   In the same way, you will want to connect TX audio, not to the MIC 
input of the transmitter, but to a flat audio input.  If the radio does 
not have that input you can compensate the attenuation introduced by the 
MIC circuits by increasing the CTCSS_LEVEL variable.

   Hope this will be of help.

73!

Gustavo - LU8WFY

f5vmr, 2015-12-11:
Thanks Gustav,

 

But sadly none of this helps in this case. The radio in use is a Kenwood
TK709, already modified to flat audio on receive as part of a repeater
assembly. As for the sound card, there is no line input available as I am
using a USB sound card on a Raspberry Pi. There are no transmitting issues.

The interface I am using has already taken care of filtering, and has
infinitely adjustable input (and output) and I have tweaked these to within
an inch of their lives so to speak. Obviously I need a level of normal audio
to pass through the repeater and reducing these further is
counterproductive. As Tobias has re-configured the DTMF part of the
software, so I was wondering if there was simply an issue with the coding.
As I say only the tones for 2, 3 and 6 on the keypad are affected with this
issue on the preferable setting. The rest of the keypad tones decode
successfully. I have tried other radios with their keypads into this new
repeater and they have an identical result, I can only suggest the receiver
of the software. Maybe it's a combination of both?

 

Chris

f5vmr, 2015-12-11:
Okay, now I am beginning to suspect something may be wrong. I have now checked another repeater (that uses TK809's UHF)), again with an installation of Svxlink 15.11 but using an older version of Raspbian on a Raspberry Pi Model B and an identical interface. The result is identical that DEEMPHASIS=1, CTCSS decodes satisfactorily by DTMF number 2,3 and 6 do not decode. The rest of them do. Yes I have tried different radio to access the repeater but the result is the same. Frustrating as the DTMF codes are necessary for full function of the EchoLink module.

f5vmr, 2015-12-11:
Ok, so I got out my Boafeng UV5R, not renowned for its reliability in my book, and I found that the tones from this portable were acceptable for decoding. It therefore surprises me that my yaesu radios FT60R and FT8900 generate tones are not. Your comments would be appreciated, before I tear the rest of my hair out. :)

gustavc39, 2015-12-11:
I can imagine!!!

Chris, have you tried monitoring the RX audio "by ear"? maybe by turning 
off DTMF_MUTING? Maybe there is some noise/distortion in the incoming 
audio...

DG1BBQ, 2015-12-12:
Salut Chris.
I have the 15.11 installed on our 2m repeater DB0BHV-R (VIA-C7), its remote RX, our 10m repeater DB0BHV (BananaPi, no EchoLink) and its remote RX (no local RX there through obvious reasons). Also on a BananaPi at home (used to access the 10m repeater). I've tested DTMF decoding through all paths using a FT8800 (Identical to FT8900, but only 2m/70cm). Worked whithout an issue in all cases. Doesn't seem to be a general problem.

You might want to use the audio recorder to record the DTMF tones you're sending and analysing the spectrum of the tones using Audacitiy. The low and high tones should idealy have the same levels. If the tones 'lean' significantly to one or the other side (meaning high tone louder than low tone or vice versa), you might want to try tweacking DMTF_MAX_FWD_TWIST and/or DTMF_MAX_REV_TWIST.

73s, Axel DG1BBQ

sm3sgp, 2017-03-08:
@f5vmr did you manage to solve this issue?

dulle, 2017-10-17:
Same here!
Yeasu VX-8G radio DTMF got cut off A,C,D,1,4,7 is no go. But cheap Chinese Boafeng UV5R got all DTMF codes decoded. 
A long time ago when playing with phone patches I had the similar problem that some radios DTMF didn't perform well as others, never resolved it. 
My suspect here would be USB sound card not picking some tones, I have new one on order and will let you know how it goes

sm0svx, 2017-10-21:
Actually my Yaesu VX-6 and VX-3 have a faulty DTMF implementation where the tones seem to be injected after preemphasis on transmission so when received and deemphasized, you will get a significant tone twist, like 6-7dB. Despite that, it works on my link where I use a FT-8900 transceiver. Maybe because they are the same brand :-)  The VX-8 seem to be fine though.

One way to understand what's going on is to set DTMF_DEBUG=1 in your rx config section. Have a look at the man page for svxlink.conf for a full explanation.

mytechguyri, 2019-03-16:
Fwiw, i picked up a $5 mt8870 dtmf decoder board off amazon and feed the audio through that to decode dtmf via hardware instead of software....much more reliable dtmf decoding even with it off spec dtmf implementations.

sm0svx, 2019-07-15:
Software DTMF decoding is usually very reliable, especially with later versions of SvxLink. The decoder was very much improved some years back, especially when it comes to false detections. A mode for low signal strength conditions also was added. When a noisy signal is detected the decoder requires a longer tone detection. This comes natural for most experienced operators, using longer and slower DTMF in weak signal conditions.

Just make sure that the incoming audio is nice and flat and everything should work just fine.

closed 2019-07-15 by sm0svx