sm0svx/svxlink #187: GPIO support with Orange Pi

f5nlg, 2016-01-03:
hello , on ORANGE PI gpio is under /sys/class/gpio_sw
and with svxlink it want /sys/class/gpio 
best information i found : http://www.cnx-software.com/2015/09/26/status-of-orange-pi-boards-gpio-support/

73"

f5nlg, 2016-01-03:
finaly i found how to fix it in /src/svxlink/trx : PttGpio.cpp and SquelchGpio.cpp

stringstream ss;
  ss << "/sys/class/gpio_sw/" << gpio_pin << "/data";
  ofstream gpioval(ss.str().c_str());
  if (gpioval.fail())

and it works !!

DG1BBQ, 2016-01-13:
Wouldn't a symbolic link named gpio_sw in /sys/class/ pointing to gpio have done the job?
Axel, DG1BBQ

f5nlg, 2016-01-13:
Hi,
It's not possible tout make symlink ln /sys and name is différent Also 73

kb3vgw, 2017-01-14:
close

sm0svx, 2017-02-05:
@kb3vgw, why can this issue be closed? Isn't this problem still present on the ORANGE Pi, @f5nlg?

f5nlg, 2017-02-05:
Hello, i found the solution , si i think it,s the reason why someone close
this issue

sm0svx, 2017-02-05:
Was the solution changing the source code and recompile, like you described above, or did you find another way?

f5nlg, 2017-02-05:
Hello no i have to change source code a way to test raspberry or orange pi
then write the good gpio Line code should be  very fine CAT /proc/cpuinfo
should be a way
Raspberry use btoadcom , orange pi h2+ or h3 processor.

73" de f5nlg
PS i investigate orange pi zero who as pin for internal sound card

DG1BBQ, 2017-02-05:
Making the path to the gpio folder configurable would be an option.

f5nlg, 2017-02-05:
No it's not the same path and syntax symbolic link does'nt World

DG1BBQ, 2017-02-17:
Hi there,

I Meant make it configurable in  the GLOBAL section in svxlink.conf.
Maybe like this:
GPIO_PATH=/sys/class/gpio_sw
`When SVXLink needs to access the GPIOs, it could check GPIO_PATH.
@sm0svx  Wouldn't that make sense?

sm0svx, 2017-02-17:
Yes, that is why I asked if a change in the source code was required. If so, there should be a configurable option like you @DG1BBQ describe to set the path so that all different hardware can be supported.

kb3vgw, 2017-02-18:
see pull request 246 this is now fixed to work in the new gpio scripts and gpio.conf.

f5nlg, 2017-04-05:
Sorry but IT DOES'NT Works i test new version but on gpio_sw it's not "value" but "data" so gpio scripts and gpio.conf does'nt work  with any Orange Pi ( lite one pc etc ... )

see my first post about this subject 

stringstream ss;
ss << "/sys/class/gpio_sw/" << gpio_pin << "/data";
and not "/value" as on actual version of svxlink  ....

sample of starting svxlink on orange pi ....

Starting logic: 432Logic
Loading RX: Rx1
Loading TX: Tx1
*** ERROR: Could not open GPIO /sys/class/gpio_sw/PD14/value for writing in transmitter Tx1.
*** ERROR: Could not initialize TX "Tx1"
*** ERROR: Could not initialize Logic object "432Logic". Skipping...

kb3vgw, 2017-04-05:
There is a issue in the systemd startup that configures gpio pins I am working on it 

From: f5nlg
Sent: Tuesday, April 4, 2017 7:54 PM
To: sm0svx/svxlink
Cc: Richard Neese; Mention
Subject: Re: [sm0svx/svxlink] GPIO support with Orange Pi (#187)

Sorry but IT DOES'NT Works i test new version but on gpio_sw it's not "value" but "data" so gpio scripts and gpio.conf does'nt work with any Orange Pi ( lite one pc etc ... )
see my first post about this subject
stringstream ss;
ss << "/sys/class/gpio_sw/" << gpio_pin << "/data";
and not "/value" as on actual version of svxlink ....
sample of starting svxlink on orange pi ....
Starting logic: 432Logic
Loading RX: Rx1
Loading TX: Tx1
*** ERROR: Could not open GPIO /sys/class/gpio_sw/PD14/value for writing in transmitter Tx1.
*** ERROR: Could not initialize TX "Tx1"
*** ERROR: Could not initialize Logic object "432Logic". Skipping...

DG1BBQ, 2017-04-05:
@f5nlg 
I don't know what Linux you are using, but with armbian, the paths and names are just like on the rpi!
echo "3" >/sys/class/gpio/export
echo "in">/sys/class/gpio/gpio3/direction
echo "1">/sys/class/gpio/gpio3/active_low
cat /sys/class/gpio/gpio3/value  gets a 1 as it should be.

I didn't try all possibilities, but this should be proof enough.
So, try out armbian. Runs fine on my OrangePi PC1 and PC2
I personally like the OPi. It has a shutdown button(!) and if you remove the electret Mic, you have an audio-in!
And it's dirt cheap (got mine as a kit for just 25 Euro [SBC, case, 3Amp PSU and 8GB uSD!]). And it runs well.
73s, Axel DG1BBQ

f5nlg, 2017-04-05:
@DG1BBQ

it depend wich orange pi you use , i have Lite/One/PC1/ZERO ...and there 
is no gpio3 or any other port by that name ... i use Armbian , i made 
some distro with svxlink for raspberry and Orange Pi ( in french but 
Debian is Debian , "dpkg-reconfigure locales" do the trick ).

https://f5nlg.wordpress.com/category/orval-orange-pi/ who is the last 
realese from january.

more information about H3 can be found on orange pi website / Sunxi8 manual

GPIO arrangement for lite/PC/One: the picture is at the end of this page 
for H3 processor , for H2+ the GPIO arrangement is not the same :

https://f5nlg.wordpress.com/2016/01/03/orange-pi-pc-et-svxlink/

Orange Pi is realy better than Raspberry, really faster, quad core 
processor, TTY's, etc ... i used some until 2016

the better for us: is orange pi zero, who is about 12€ , audio ports and 
gpio are usable , there is wifi and ethernet. works well with svxlink.

F5ZJX 10m/2m RH2/R2 repeater with orange pi one

F5ZIN 29.130Mhz/50.540Mhz links and RU16H/R0 repeater ( 4 radio way on 2 
orange pi one )

F6KBO-L 2m link on orange pi zero

F5NLG UHF link on svxreflector with orange pi lite

  i also manage RRF network ( French repeater network , about 8 to 10 
RPT/Link connected with EchoLink or SvxReflector now was Svxserver before )

73" de F5NLG.

sm0svx, 2017-04-06:
Ah, did not notice that it was "data" instead of "value". Strange beast that OrangePi but seems useful.

f5nlg, 2017-04-06:
Wiringpi python script on armbian 5.25 solve the problem and do gpio from
orange pi and raspberry much compatible but I don't like to resolv à
problem with a so Big solution ... Python + script as we said on french "
usine à gaz" gas Factory ? In english ...
73"

sq7raf, 2017-04-15:
Hello,
so  armbian 5.25 onboard ( OPI One ) .. but i still don't know how to use GPIO ....
Is there simple instruction/soultion ?
Names of PINs on OPI are called PAx , but it's not working ...

Thanks in advance.
73'

kb3vgw, 2017-04-27:
OK afte testing and reverting I found the iss Tobias. the /lib/systemd/system/svxlink_gpio_setup.service is not getting the override varitable from /etc/default/svxlink file there for it is only looking to load the default /etc/svxlink dir and will not work with a diff directory. I spent 7 days tracking this down step by step . I had to revert systemd back to the old working one and it functioned again like it should so using a helper systemd service does not work as it should.

sp2ong, 2017-09-18:
Hi 
I have a play with Orange Pi Zero to use as svxlink small server. I have used Debian version from https://dl.armbian.com/orangepizero/. I have use development version of ARMBian and stable which base on kernel 3.4.113. When using ARMbian with mainline kernel 4.11.5 the GPIO pins can be accessed from user space using sysfs. 
Device Drivers  ---> GPIO Support  ---> /sys/class/gpio/... 
To access a GPIO pin you first need to export it with
echo XX > /sys/class/gpio/export
with XX being the number of the desired pin. To obtain the correct number you have to calculate it from the pin name (like PA17):
(position of letter in alphabet - 1) * 32 + pin number
E.g for PA17 this would be ( 1 - 1) * 32 + 17 = 0 + 18 = 242 (since 'A' is the first letter).
where the name of GPIO for Orange PI Zero : http://linux-sunxi.org/Xunlong_Orange_Pi_Zero
With legacy kernel 3.4 it is different where gpio exist in /sys/class/gpio_sw and you must use name gpio like PA17 etc.
But I have found on forum ARMbian nice solution for legacy kernel. You need no load module gpio_sunxi and after this you can use gpio numbering and locatiuon like for mainline kernels.

I have put in /etc/modprobe.d/orangepizero.conf
blacklist gpio_sunxi

and remove from /etc/modules.conf

I have observed when I have use internal sound card that sometimes in log is 
Rx1: Distorsion detected! Please lower the input volume!

It very nice small computer with size 4.5x4.5 cm and have ethernet port/wifi / internal sound card 3 ports UART 

=========================================================
It useful is put in /etc/rc.local flash red led with heartbeat system:

echo "heartbeat" > /sys/class/leds/red_led/trigger
=========================================================

f5nlg, 2017-09-18:
> Xunlong_Orange_Pi_Zero

Hello
For my purpose i play with script.fex / script.bin you Can activate exactly
what you need ...

For soundcard you have to fix mic gain to 1 , elsewhere the sound IS Bad
...
Alsamixer does'nt work with mic input ...

For gpio , uart (for configuring sa818) you need to play with script.fex ,
then use fex2bin script.fex script.bin and reboot

Thats the good way for sunxi processor on armbian.

73" de F5NLG Jean-philippe.

sp2ong, 2017-09-18:
Hi Jean-Philippe

Thank you for remarks. Your website about orange pi and svxlink has
inspired me to use Ornage PI Zero :-)

Can you tell more how to fix mic gain to 1 ???

It will be useful information. I have seen in your configuration "spotnik"
only set

# Set audio levels
amixer -c 0 cset numid=6 78%
sleep 1
amixer -c 0 cset numid=4 12%
sleep 1

73 Waldek SP2ONG

f5nlg, 2017-09-18:
You have to fix it in script.fex ,
1st step go go /boot
bin2fex script.bin script.fex
Have a look on script.fex with an editor
Make changes on audio configuration , gpio, uart If you need them
Then :
fex2bin script.fex script.bin
reboot and all mods you do like gpio uart sound etc ... IS fine
Take a look of sunxi manuel for more explain
73" de F5NLG Jean-philippe.

RichNeese, 2017-09-18:
The nanopi neo uses the gpio like pa17  and we just set gpioa17 they work fine
And they work in the gpio.conf no issue

From: f5nlg
Sent: Monday, September 18, 2017 04:17 PM
To: sm0svx/svxlink
Cc: Richard Neese; Mention
Subject: Re: [sm0svx/svxlink] GPIO support with Orange Pi (#187)

You have to fix it in script.fex ,
1st step go go /boot
bin2fex script.bin script.fex
Have a look on script.fex with an editor
Make changes on audio configuration , gpio, uart If you need them
Then :
fex2bin script.fex script.bin
reboot and all mods you do like gpio uart sound etc ... IS fine
Take a look of sunxi manuel for more explain
73" de F5NLG Jean-philippe.

sp2ong, 2017-09-19:
Hi Jean-Philippe

I have to use your IMG "SPOTNIK v1.0-beta on my Orange Pi Zero and when I have run svxlink in log still see in log file

Rx1: Distorsion detected! Please lower the input volume!
it displays only one line no more when svxlink switch TX On/Off but in this test, I don't have connected any interfaces audio to pins MIC and LinuOUTR

Is it  fixed in IMG "SPOTNIK" ( https://www.dropbox.com/s/vpaqgc2d0cg58ng/spotnik-v1-beta.img.7z?dl=0 ) or I must  to fix it in script.fex ??

73 Waldek

sp2ong, 2017-09-19:
Hi Jean-Philippe

I have found in SPOTNIK img  in script.fex

[audio0]
audio_used = 1
lineout_vol = 27
cap_vol = 1
audio_hp_ldo = "none"
adcagc_used = 0
adcdrc_used = 0
dacdrc_used = 0
adchpf_used = 0
dachpf_used = 0
audio_pa_ctrl = port:PC05<1><default><default><0>

sp2ong, 2017-09-21:
Looking in log svxlink I have seen that message with Distortion detected is printed time not always after swithchTX form ON to OFF:
...
Thu Sep 21 05:00:00 2017: SimplexLogic: Sending long identification...
Thu Sep 21 05:00:00 2017: Tx1: Turning the transmitter ON
Thu Sep 21 05:00:14 2017: Tx1: Turning the transmitter OFF
Thu Sep 21 05:30:00 2017: SimplexLogic: Sending short identification...
Thu Sep 21 05:30:00 2017: Tx1: Turning the transmitter ON
Thu Sep 21 05:30:10 2017: Tx1: Turning the transmitter OFF
Thu Sep 21 06:00:00 2017: SimplexLogic: Sending long identification...
Thu Sep 21 06:00:00 2017: Tx1: Turning the transmitter ON
Thu Sep 21 06:00:15 2017: Tx1: Turning the transmitter OFF
Thu Sep 21 06:30:00 2017: SimplexLogic: Sending short identification...
Thu Sep 21 06:30:00 2017: Tx1: Turning the transmitter ON
Thu Sep 21 06:30:10 2017: Tx1: Turning the transmitter OFF
Thu Sep 21 06:30:10 2017: Rx1: Distorsion detected! Please lower the input volume!
Thu Sep 21 07:00:00 2017: SimplexLogic: Sending long identification...
Thu Sep 21 07:00:00 2017: Tx1: Turning the transmitter ON
Thu Sep 21 07:00:15 2017: Tx1: Turning the transmitter OFF
Thu Sep 21 07:00:15 2017: Rx1: Distorsion detected! Please lower the input volume!

I have to check with kernel 3.4.113 and 4.11.12. It looks like internal hardware spike noise on MIC input when svxlink switch TX from ON to OFF.

sp2ong, 2017-09-23:
Hi,
At current, I did not have success use Orange Pi Zero with svxlink . The internal card has bad quality audio for the microphone. When I have USB sound card the output audio is overlapped by noise. 
If anybody has good success used OPIZ will nice when sharing more dipper information how to setup audio etc

73 Waldek SP2ONG

sm3sgp, 2019-09-10:
To my understanding it should be possible to use the GPIO with the Orange Pi.
Have a look at /etc/svxlink/gpio.conf and the systemd script svxlink_gpio_setup.service

closed 2019-09-10 by sm3sgp