sm0svx/svxlink #189: module Metrainfo

kb3vgw, 2016-01-10:
I wanted to know is module metra info being maintained and and used by any one before recording new sounds for it. I have managed to recore new sounds other wise but I have not heard of anyone using metrainfo and if not I think the module should be pulled if its outdated and not maintained.

dl1hrc, 2016-01-10:
The module (MetarInfo not metra) is working and maintained by me (dl1hrc) if an error is reported. If you don't need/like it just don't use it. Why throwing out a working module?

VR2XKP, 2016-08-08:
Hello

I and some ham found the metarInfo seems no longer funtion (prompts "metar_not_valid") . And I search on the web and found this link:
http://svxlink.de/?p=3316

Does it means this feature no longer available on the Svxlink ? If Yes, it is so sad.

VR2XKP, 2016-08-08:
Hello DL1HRC

Thank you for your effort.  Hope the module can resume asap.

Br
Francis / VR2XKP

j69ds, 2016-08-11:
Hi Adi,

I am eagerly looking forward to an update and getting the metar system working here!
It will be asset here for the ham community for sure!

Regards,
Frans J69DS

dl1hrc, 2016-08-12:
Hi Frans,
please checkout my branch: metarinfo_changed_service
git clone https://github.com/dl1hrc/svxlink.git
cd svxlink/src
git checkout metarinfo_changed_service
mkdir build; cd build
...and so on

The branch is even with the trunk. It's not tested well at the moment, please send me error messages if some occur.

vy 73s de Adi / DL1HRC

j69ds, 2016-08-12:
Hi Adi,

I am very happy to see you have it figured out!

Only one problem I am a total noob when it comes to linux.

Do I have to start from scratch to add it in?
I am not sure how to do this at all

Thank you again for a great job!

Frans j69ds

dl1hrc, 2016-08-12:
Hi Frans,

what kind of Linux do you run?
But I think it will burst this frame, send me an email then we can do it 
directly, ok?

vy 73s de Adi / DL1HRC

kb3vgw, 2016-08-12:
We use Debian on arnhf and arm64 . I have a pkg repo for our project called
open repeater @ openrepeater.com. we have a install script and there are a
few boards we develop for it . One I a mm working on now for pine64 board.
We have one for pi 2/3 . We have new sound for English and soon French.

There is a lot going into  it. And loads of new modules.

Kb3vgw
Rich

dl1hrc, 2016-08-12:
...sounds very interesting, thank you much for your efforts. Are the new 
sounds protected or open for everyone?
Especially for metar a lot of new sounds have to be created that it 
works. It will take some weeks I think. The better way maybe to create 
it on-the-fly, e.g. google-api, network-tts,....since there are 
changeing information sometimes.

I've running such own module to give out (thunder)storm warnings from 
the official weather service to alert the om's here in my vicinity. This 
are normal test messages coming via eMail then transfered into a wav an 
sent out as soon as received.

vy 73s de Adi / DL1HRC

j69ds, 2016-08-13:
Hi Adi 

Thank you very very much for the assist in correcting my mistakes.
It is working beautifully now!
Thank you for your time and effort and this great module indeed. 
In the Caribbean it will prove to be a great asset for the operators indeed.

Thank you

regards,
Frans J69DS

kb3vgw, 2016-08-13:
Where to get updated code?

dl1hrc, 2016-08-13:
Hi,

you have to checkout my branch, it's even with the trunk:

git clone http://github.com/dl1hrc/svxlink.git
cd svxlink/src
git checkout metarinfo_changed_service
mkdir build
cd build
...

vy 73s de Adi / DL1HRC

g4usp, 2016-08-14:
Hi Adi,

Many thanks for your response, I thought it might be a change to the service provided by the aviation authorities.

So its a recompile from the latest source rather than changes to the .conf / .tcl files...

I run SVXlink under Raspbian Wheezy on a Raspberry Pi for my analogue gateway (MB7ILH) and the replacement repeater for GB3HT under Raspbian Jessie on a Raspberry Pi which I was hoping to commission later next week...

Again many thanks for all of your work on this...

73's de Simon (G4USP)

f5vmr, 2016-08-14:
You'll be welcome on my Facebook page SVXLink amateur users, if you're not already.
Regards
Chris f5vmr g4nab

s58DB, 2016-11-04:
Hello,

today I tray this command but I get error. 

git clone http://github.com/dl1hrc/svxlink.git 
cd svxlink/src 
git checkout metarinfo_changed_service 
mkdir build 
cd build 
cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc -DLOCAL_STATE_DIR=/var -DUSE_OSS=NO -DUSE_QT=NO ..
make
make doc
sudo make install
sudo ldconfig

Compile finish without any error. 

When I start svxlink I get an error:

Loading module "ModuleMetarInfo" into logic "SimplexLogic"
        Found /usr/lib/arm-linux-gnueabihf/svxlink/ModuleMetarInfo.so
        Module MetarInfo v1.1.1 starting...
**** WARNING: Config variable ModuleMetarInfo/TYPE is not set.
*** ERROR: Initialization failed for module ModuleMetarInfo in logic SimplexLogic

Do you have any suggestion what I am doing wrong.

Bets regards Danilo - S58DB

dl1hrc, 2016-11-04:
Hi Danilo,

please refer to:

https://github.com/dl1hrc/svxlink/blob/metarinfo_changed_service/src/svxlink/modules/metarinfo/ModuleMetarInfo.conf.5

TYPE defines the way of getting the information, as XML or as TXT.

Please take the ModuleMetarInfo.conf from the source code

https://github.com/dl1hrc/svxlink/blob/metarinfo_changed_service/src/svxlink/modules/metarinfo/ModuleMetarInfo.conf

and configure it to your requirements.

vy 73s de Adi / DL1HRC

s58DB, 2016-11-04:
Thenks Adi for your quick answer.  MetarInfo now work great.

kb3vgw, 2016-11-05:
Yes they changed the protocol. There is new code but it's not in the main
repo as of yet. It's in my repo along with the new svxlink server service
my source is setup for Debian pkg build and systemd. What platform u using ?

s58DB, 2016-11-05:
I am using Raspbian Jessie and RPi2.

sm3sgp, 2019-09-10:
The Module MetarInfo is reported working well.

closed 2019-09-10 by sm3sgp