sm0svx/svxlink #190: ModuleTclVoicemail

kb3vgw, 2016-01-24:
Fri Jan 22 12:18:49 2016: /usr/share/svxlink/events.tcl in logic RepeaterLogic: can't read "env(HOME)": no such element in array

when running with the voicemail module loaded I get the following error.seems its not finding the home env properly

pe1chl, 2016-01-28:
It should not matter as it looks in /etc when it cannot find this.
When you don't like to see the error message just customize this script function and remove the check for the cfg_home path.

kb3vgw, 2016-01-28:
The issue is it breaks new installs for newbies. And that's a issue.

pe1chl, 2016-01-28:
Breaks?  Do you mean the program quits or malfunctions, or is it only logging this message and then functioning OK?  (I think the latter)

This issue has been discussed on the Development list.   Background services have no HOME so they will only use the /etc config file.

sm0svx, 2017-01-14:
Yes, it actually breaks SvxLink since it will not start if HOME is missing.

```
Loading module "ModuleTclVoiceMail" into logic "SimplexLogic"
        Found /home/blomman/projs/svxlink/git/src/build/lib/ModuleTcl.so
        Module Tcl v1.0.1 starting...
/home/blomman/.svxlink/events.tcl in logic SimplexLogic: can't read "env(HOME)": no such variable
*** ERROR: Could not initialize Logic object "SimplexLogic". Skipping...
*** ERROR: No logics available. Bailing out...
```
Fixed in Git master now.

closed 2017-01-14 by sm0svx