sm0svx/svxlink #194: SQL open / closed log only when connected to server

pa1okz, 2016-03-06:
In former remotetrx versions, the squelch status could be readout immediately on the screen, even when there was no direct connection with the svxlink server that the system was hooked-up to. In the latest version, such information -only- appears when the remotetrx is really connected to the other end. 

It is highly convenient to have such information locally, enabling tweaking, settings behavior and similar things without having a real connection to the live system. 

Could this change be reversed to the old situation? When preparing new remote stations, I'm currently forced to utilize the live system, which is disturbing for the users...

f5vmr, 2016-03-07:
Personally I log in to the repeater remotely by SSH and tail -f
/var/log/svxlink.log or tail -f /var/log/remotetrx.log depending if I am
looking at the master repeater or the remote receiver. 

As it is only a diagnostic function, if it's not broken I leave it. I don't
need to watch it all the time.

 

Chris

pa1okz, 2017-01-04:
It is only required during setup of new receivers (which occurs regularly in our system), things as SQL open and close indications are useful to have at least some feedback if things fundamentally work and behave as desired after calibration. This cannot be seen any longer without the receiver being 'live' in the system.

sm0svx, 2017-01-14:
You could have worked around this by enabling FALLBACK_REPEATER in your configuration but now I have changed it so that OPEN/CLOSE is displayed always, even when not connected to the main system. It's in latest Git master.

closed 2017-01-14 by sm0svx