sm0svx/svxlink #198: Crashing when APRS LocationInfo is enabled with internet outage

mmiller7, 2016-03-26:
When APRS LocationInfo is enabled and the internet goes out, svxlink crashes.

Symptoms to reproduce:
1. enable LocationInfo
2. Disconnect internet
3. Reboot (no IP addresses on any interface)
4. Start svxlink service
5. Wait <1 minute, process dies

Workaround: Disable (comment out) LocationInfo in the main settings heading.

Here is the configuration I was using for LocationInfo

> [LocationInfo]
#APRS_SERVER_LIST=euro.aprs2.net:14580
APRS_SERVER_LIST=noam.aprs2.net:14580
#STATUS_SERVER_LIST=aprs.echolink.org:5199
#LON_POSITION=12.10.00E
LON_POSITION=77.29.08W
#LAT_POSITION=51.10.00N
LAT_POSITION=38.18.16N
#CALLSIGN=EL-DL0ABC
CALLSIGN=EL-MyCall
#FREQUENCY=438.875
FREQUENCY=446.1
#TX_POWER=8
TX_POWER=10
#ANTENNA_GAIN=6
#ANTENNA_HEIGHT=20m
ANTENNA_HEIGHT=5m
#ANTENNA_DIR=-1
PATH=WIDE1-1
BEACON_INTERVAL=10
TONE=100
COMMENT=svxlink on Linux

dl1hrc, 2016-03-27:
Can not confirm this, have it tested here many times without crash (no network/ip, changing from no ip to with ip and back). It was running for more than 30 minutes. What version do you run?

mmiller7, 2016-03-28:
Both versions v1.4.99.23 and my just upgraded v1.5.0 seem to be exhibiting these symptoms when it's started thru the usual init.d script (boot-time or `service svxlink start` command).  I'm using the init.d script that came with svxlink (distributions/debian/etc/init.d/svxlink) to load the svxlink service.

The host machine is a Raspberry Pi v2 with Raspbian GNU/Linux 7 (wheezy) at runlevel 3

I've been unable to locate any specific errors (the svxlink log just stops abruptly) and there is nothing that I can find in the syslog or dmesg at the time it "dies".

For some reason it seems to work if I "su" to the user and run svxlink manually from a command prompt but that isn't a useful way to operate (I need it to run headless and load the service automatically)

mmiller7, 2016-04-08:
I have some additional details, apparently the specific parameter causing issues is "APRS_SERVER_LIST" and when I comment out this variable I can leave the rest on (including enabling "STATUS_SERVER_LIST".

I can tail -f the logfile and see the last line it prints is either "Disconnected from APRS" or "Reconnecting to APRS" before it dies.

mmiller7, 2016-05-02:
Even after a clean reinstall (searching `find / -name "*svxlink*"` deleting everything, then download/build/install v1.5.0) I'm still seeing this issue when it's started thru the init.d service-start scripts (using the script included with svxlink).

Turning on/off APRS_SERVER_LIST seems to be the "make or break" option.

thasti, 2016-11-12:
I can confirm this issue, although I cannot state our currently used svxlink version (no remote internet access to the repeater ATM).

sm0svx, 2017-01-14:
Can you try the latest master branch from Git? I recently committed a fix to the DNS name lookup code that may have affected this.

sm3sgp, 2017-03-10:
@mmiller7 and @thasti , do you have any update if this problem is still exists in the latest master branch?

pe1chl, 2017-04-04:
It looks like a new problem is introduced in the master branch.
We have updated a couple of repeaters, and the symptom is after a few hours of runtime the
following 2 messages are logged:
Tue Apr  4 10:59:10 2017: *** WARNING: Disconnected from APRS server
Tue Apr  4 10:59:15 2017: *** WARNING: Trying to reconnect to APRS server
This then keeps re-occurring.
No crash but no APRS server either...  it worked OK before this update.

pe1chl, 2017-04-04:
Further debugging: repeater is trying to connect to 93.104.116.162 but gets no response.
Trying from other systems within 44.137.0.0/16 results in no response from this system.
Probably another hamnet-to-internet via NAT goofup in Germany.  The reverse DNS for this
server is ppp-93-104-116-162.dynamic.mnet-online.de
However, svxlink keeps trying this broken server and cannot get connected again.
It should resolve the DNS name (euro.aprs2.net) again and try another IP from the list.

sm3sgp, 2019-09-10:
Is this problem still seen on the latest release?

closed 2025-03-18 by sm3sgp