sm0svx/svxlink #199: svxlink on rpi works but not the squelch detection - siglevdetcal issue

dewilderw, 2016-05-06:
Hi 

I set up a Raspberry Pi running svxlink with C-Media USB sound card, I get good audio
in and output, managed to listen to the radio via echolink and also transmitted to the connectedradio fine,

However I can't get the Squelsh detection to work, neither VOX nor SIGLEV.
What seems to be the issue is when I run siglevdetcal there's no difference between
getting audio to the input or silence to it, it displays the same levels, almost like it's
reading a different channel or card, but it seems to be the correct channel as I need to kill
svxlink that's running and that has the port open ...

What could cause everything to work except the squelch ..

kgrant341, 2016-10-20:
Try changing the audio_channel  to a different one the values are 1 or 0

pa1okz, 2017-01-08:
Is there actual audio coming into the right port as kgrant341 suggests? Probably the level is absent, mixer settings in Alsamixer are wrong (sometimes the driver is dodgy, causing the wrong faders to be present). Furthermore, do you use flat audio - so straight from the FM detector of 9600 baud packet output of the receiver? Otherwise it is never going to work.

sm3sgp, 2017-03-08:
@dewilderw did you manage to solve this problem?

dewilderw, 2017-03-08:
As far as I know I did get the ports set correctly. At one point the receive worked fine, except the squelch part, another time the transmit side worked fin, I could key and unkey the transmitter. But I could not get the audio to squelsh correctly. Maybe too much noise from power supply coming in, there was some hum so perhaps not galvanicly isolated between radio and RPI. I did use audio transformers but had to connect the grounds on both side to get the PTT via transistor to work.

The audio I used was audio out from the radio rear panel external speaker connection..

This issue can be closed now, for now I gave up on getting my setup to work, but the issue 
was perhaps on my side with the audio interface.

closed 2017-03-08 by dewilderw

pa1okz, 2017-03-08:
You should use flat audio out instead of speaker out. Use the 9k6 output of
your trx if you have that, otherwise you should create a detector output.
That solves your issue.

73

Mischa, PA1OKZ