sm0svx/svxlink #203: ModuleFrn

dh9yap, 2016-08-06:
the com / vox delay is too long by 5 seconds.
Have tried the QsoFrn.h of the knobs.
static const int KEEPALIVE_TIMEOUT_TIME = 5000; "
From 5000 changed to 1000 and new Compiled.
Unfortunately, there is no change.

r8oag, 2016-08-13:
I have problem to connect to FRN server

==========
svxlink@frn-echolink:~$ svxlink
SvxLink v1.5.0 Copyright (C) 2003-2015 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /usr/local/etc/svxlink/svxlink.conf
--- Using sample rate 16000Hz

Starting logic: SimplexLogic
Loading RX: Rx1
SimplexLogic: Activating QSO recorder
Loading TX: Tx1
Loading module "ModuleFrn" into logic "SimplexLogic"
	Found /usr/local/lib/i386-linux-gnu/svxlink/ModuleFrn.so
	Module Frn v1.0.0 starting...
frn debugging is enabled
SimplexLogic: Event handler script successfully loaded.
SimplexLogic: Activating module Frn...
state: CONNECTING
connecting to 83.246.188.71:10024
Tx1: Turning the transmitter ON
state: CONNECTED
state: LOGGING_IN_1
state: LOGGIN_IN_2
login stage 1 completed: 2014003
state: ERROR
QSO errored, deactivating module
SimplexLogic: Deactivating module Frn...
state: DISCONNECTED
login stage 2 failed: <MT></MT><SV>2014004</SV><AL>BLOCK</AL><BN>fill in back up server address</BN><BP>10024</BP>
Tx1: Turning the transmitter OFF

==============

r8oag, 2016-08-14:
Sorry, last error message is:
`login stage 2 failed: <MT></MT><SV>2014004</SV><AL>BLOCK</AL><BN>fill in back up server address</BN><BP>10024</BP>`

dh9yap, 2016-08-15:
Hello what is your modules FErn.conf from ... Greetings Jürgen :-)

r8oag, 2016-08-15:
/usr/local/etc/svxlink/svxlink.d
`[ModuleFrn]
NAME=Frn
PLUGIN_NAME=Frn
ID=7
TIMEOUT=300
# Details http://freeradionetwork.eu/frnprotocol.htm
SERVER=83.246.188.71
PORT=10024
VERSION=2014000
EMAIL_ADDRESS=r8oag@mail.ru
DYN_PASSWORD=*************
CLIENT_TYPE=1
CALLSIGN_AND_USER="R8OAG, Andrey"
BAND_AND_CHANNEL="145.350FM"
DESCRIPTION="SvxLink FreeRadioNetwork Station"
COUNTRY="Russian Federation"
CITY_CITY_PART="Novosibirsk - NO14LU"
NET=22dx.ru
FRN_DEBUG=1
#DISABLE_RF=1`

r8oag, 2016-08-15:
`[ModuleFrn]
NAME=Frn
PLUGIN_NAME=Frn
ID=7
TIMEOUT=300
# Details http://freeradionetwork.eu/frnprotocol.htm
SERVER=83.246.188.71
PORT=10024
VERSION=2014000
EMAIL_ADDRESS=r8oag@mail.ru
DYN_PASSWORD=RCZKMJAI
CLIENT_TYPE=1
CALLSIGN_AND_USER="R8OAG, Andrey"
BAND_AND_CHANNEL="145.350FM"
DESCRIPTION="SvxLink FreeRadioNetwork Station"
COUNTRY="Russian Federation"
CITY_CITY_PART="Novosibirsk - NO14LU"
NET=22dx.ru
FRN_DEBUG=1
#DISABLE_RF=1`

r8oag, 2016-08-15:
[ModuleFrn]
NAME=Frn
PLUGIN_NAME=Frn

ID=7
TIMEOUT=300

SERVER=83.246.188.71
PORT=10024
VERSION=2014000
EMAIL_ADDRESS=r8oag@mail.ru
DYN_PASSWORD=********
CLIENT_TYPE=1
CALLSIGN_AND_USER="R8OAG, Andrey"
BAND_AND_CHANNEL="145.350FM"
DESCRIPTION="SvxLink FreeRadioNetwork Station"
COUNTRY="Russian Federation"
CITY_CITY_PART="Novosibirsk - NO14LU"
NET=22dx.ru

FRN_DEBUG=1
#DISABLE_RF=1

dh9yap, 2016-08-16:
Hello ... I have found the error !! :-)
BAND_AND_CHANNEL="145.350FM"       <-------- Wrong
BAND_AND_CHANNEL="145.350 FM"      <-------- Correct
or
BAND_AND_CHANNEL="145.350FM xxxx"

here the korekte ModuleFrn.conf
[ModuleFrn]
NAME=Frn
PLUGIN_NAME=Frn
ID=7
TIMEOUT=300

# Details http://freeradionetwork.eu/frnprotocol.htm
SERVER=83.246.188.71
PORT=10024
VERSION=2014000
EMAIL_ADDRESS=r8oag@mail.ru
DYN_PASSWORD=*********
CLIENT_TYPE=1
CALLSIGN_AND_USER="R8OAG, Andrey"
BAND_AND_CHANNEL="145.350 FM"
DESCRIPTION="SvxLink FreeRadioNetwork Station"
COUNTRY=Russian Federation
CITY_CITY_PART="Novosibirsk - NO14LU"
NET=22dx.ru
#FRN_DEBUG=1
#DISABLE_RF=1

73 es 55 de DH9YAP Jürgen

r8oag, 2016-08-16:
**Thanks, I did for your advice, it worked.**

Maybe these details should be described in more detail in the instructions?

dh9yap, 2016-08-16:
In the configuration files must be no typos make otherwise it does not work.

73 es 55 Jürgen

closed 2016-08-16 by dh9yap

citynetcomputing, 2016-12-14:
There a way to specify multiple NET's that are available on the server you are connected to??