sm0svx/svxlink #205: SVXlink metarinfo stopped working having previously been fully functional...

g4usp, 2016-08-14:
I have SVXlink running as a repeater controller with all modules available. All modules are fully functional with the exception of metarinfo which having previously worked now reports "xxxx metar information not valid" any idea's?

dl1hrc, 2016-08-14:
look here:

https://github.com/sm0svx/svxlink/issues/189

citynetcomputing, 2016-12-03:
Hi,

Yes, noaa changed their website which effects the hard coded "weather.noaa.gov" address in the metarinfo so binary. You have two options:

1. Edit and recompile the SO with the address and sub directories changed to the following:
http://tgftp.nws.noaa.gov/data/observations/metar/stations/

2. This is what i did on raspbian, install apache2 and create the above directory structure in it. Add weather.noaa.gov to /etc/hosts pointing to 127.0.0.1. Write a cron script to grab the http://tgftp.nws.noaa.gov/data/observations/metar/stations/LOC.TXT script and drop it in that folder.

Works perfectly and plays all the metar info from the file. Mine is KBUF.TXT yours is whatever it is..

sm0svx, 2017-01-14:
Adi's updated MetarInfo module is in Git master now

closed 2017-01-14 by sm0svx

dl1hrc, 2017-01-14:
Can I delete the branches "metarinfo_changed_service" and
"AsyncTcpBugfixes"?

sm0svx, 2017-01-14:
Yes, if you are satisfied with what I have merged