sm0svx/svxlink #207: raspberry pi jessie gpio

kgrant341, 2016-08-29:
im getting this error when starting svxlink on my raspberry pi 2 b+ ive tried everthing that i can find on here to get it working im just at a loss here is the error
 ERROR: Could not open GPIO /sys/class/gpio/DTRRTS/value for writing in transmitter Tx1.

f5vmr, 2016-08-29:
It looks like you may have clashing values or you have not configure the Pitt port correctly. Check the svxlink.conf help file available on the site for the options.

kgrant341, 2016-08-29:
here is the svxlink.conf section that is causing the error
[Tx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
PTT_TYPE=GPIO
PTT_PORT=gpio17
PTT_PIN=DTRRTS
#HID_DEVICE=/dev/hidraw3
#HID_PTT_PIN=GPIO3
#SERIAL_SET_PINS=DTR!RTS
#PTT_HANGTIME=1000
TIMEOUT=300
TX_DELAY=500
#CTCSS_FQ=114.8
#CTCSS_LEVEL=9
PREEMPHASIS=0
DTMF_TONE_LENGTH=100
DTMF_TONE_SPACING=50

ive also checked the /sys/class/gpio/gpio17  and this is whats in there  " active_low  device  direction  edge  power  subsystem  uevent  value"
inside direction is out
inside value is 0

f5vmr, 2016-08-29:
Yup, lose the line PTT_PIN=DTRRTS
as that is for serial control. You already have the GPIO pin correctly set otherwise.
Chris F5VMR G4NAB

f5vmr, 2016-08-29:
No in fact I miss read that - your ptt pin should be gpio17 and forget the port line.

kgrant341, 2016-09-03:
Yep that worked

closed 2016-09-03 by kgrant341