sm0svx/svxlink #208: Feature request: multiple proxy server

pe2kmv, 2016-08-31:
Hi Tobias,

Thanks for the great job by providing SVXLink! I really like this software, and I've spentl already quite some time with it last months. There is just one thing I'm bumping into, so I'd like to put this into a feature request.

Is it possible to have multiple proxy servers listed in the config file of the Echolink Module? There are quite a lot of public proxies, but they're statically put into the Echolink Module. So in case the proxy is down, busy or whatever, SVXLink won't connect.

So I could imagine to have a space separated list of proxy servers, an equal space separated list of ports and again a space separated list of passwords.

'73 de Ronald PE2KMV

sm0svx, 2017-01-14:
Yes it would be possible to add. If implementing this I think I'd prefer to put all configuration together on one line like:

```
PROXY_SERVERS=server1:8100:password1 server2:8100:password2
```

cjritola, 2019-03-31:
I vote for this feature.