sm0svx/svxlink #209: Async::AudioSource::sinkWriteSamples(const float*, int): Assertion `len > 0' failed.

hb9pae, 2016-09-03:
Hello all
I do have the same error as f5mvr noted in req #181 (termination svxlink with error msg. 

Error Msg: _svxlink: /home/pi/svxlink/src/async/audio/AsyncAudioSource.cpp:168: int 
Async::AudioSource::sinkWriteSamples(const float*, int): Assertion `len > 0' failed.

My Configuration: Raspberry PI, USB Soundcard, svxlink V 1.5.0

I used the last version from github, compilation and installation OK. After some Tests (RepeaterLogic), svxlinks crashes with error msg.

How to solve this error? Thanks for any suggestion.

73 de Peter HB9PAE

f5vmr, 2016-12-03:
svxlink: /home/pi/svxlink/src/async/audio/AsyncAudioSource.cpp:168: int Async::AudioSource::sinkWriteSamples(const float*, int): Assertion `len > 0' failed.

Just updating this situation. there is no change with this install. It is the only one of several installs that is doing it. When it crashes, no errors are reported directly. I had to make a bash script that detected the crash, printing the error and restarting SVXLink. I have swapped USB Audio cards, updated the software, and checked the configuration files. So for the moment this is a bug. 
The only indication of a problem is that the TX continues with no audio present. The crash occurs just after "squelch is open" and before "transmitter is on" would be displayed if that is of any assistance.
I am suspecting that it may not necessarily be the software, but the raspberry. But it is not a coincidence that Peter has this problem.

f5vmr, 2016-12-18:
I have just put the problematic Talco repeater in service with an error trap in the Raspberry PI that restarts svxlink in the case of a crash. However i have noticed that the crash randomly occurs immediately after the user accessing the repeater drops carrier, ie there is a cessation of audio through the sound card. If this is useful, I am wondering that it might be an audio level problem with this particular repeater. Every other repeater I have (about seven of them) are transceiver pairs of various makes and there are no issues. The Talco is a purpose built repeater and I am beginning to suspect the audio levels are significant, where I have used an identical audio interface to the other repeaters.

hb9pae, 2016-12-19:
Still have the same problem. 

I discovered, that squelched RX-audio does not make problems. If I supply unsquelched RX Audio (RF-Noise) to the same configuration, the program crashes. In both situation I am using an external DCD Signal to open the SVXLINK-Squelch

hb9pae, 2016-12-29:
Closed this tread by mistake, reopened again 
73 de peter, hb9pae

f5vmr, 2016-12-30:
Hi, Still no progress on this issuewhich has been around for ages. It may be the audio level. Are you connected capacitevely or by DC? I suspect that might be the issue.
Chris

pa2dk, 2017-01-29:
Same problem here. I'm also using unsquelshed RX-audio and a usb audio card with cm108 chipset. It seems that squelshed RX-audio also has the same problem. Tests with version 15.11 show that the issue is already present since then. I compiled a debug version of the latest commit on the master branch on the pi and ran it with gdb. A backtrace is attached: [svxlink_debuglog.txt](https://github.com/sm0svx/svxlink/files/737546/svxlink_debuglog.txt)
In noticed an interesting thing in the backtrace. Variables count and len have value 256 at lines #11 to #6
but at line #4 len=-256 so the value of len is negative at that point. I'm not familiar with the source code yet, but i find that a little bit suspicious.

Any help on the issue is appreciated.

Thanks in advance

73 Dennis PA2DK

sm0svx, 2017-01-29:
Thanks! That is interesting. The negative function argument is indeed suspicious. I'll take a look at it.

sm0svx, 2017-01-29:
This has something to do with the CTCSS squelch but after a quick look I cannot tell exactly what the problem is. Can you post your RX configuration section please?

If possible, try to run your system without the CTCSS squelch and see if the problem goes away.

pa2dk, 2017-01-30:
[Rx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
SQL_DET=CTCSS
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=0
#SQL_EXTENDED_HANGTIME=1000
#SQL_EXTENDED_HANGTIME_THRESH=15
#SQL_TIMEOUT=600
VOX_FILTER_DEPTH=20
VOX_THRESH=1000
#CTCSS_MODE=2
CTCSS_FQ=88.5
#CTCSS_SNR_OFFSET=0
CTCSS_SNR_OFFSET=-15.43
#CTCSS_OPEN_THRESH=6
#CTCSS_CLOSE_THRESH=2
#CTCSS_BPF_LOW=50
#CTCSS_BPF_HIGH=300
TX_DELAY=0
SERIAL_PORT=/dev/ttyS0
SERIAL_PIN=CTS
#SERIAL_SET_PINS=DTR!RTS
#EVDEV_DEVNAME=/dev/input/by-id/usb-SYNIC_SYNIC_Wireless_Audio-event-if03
#EVDEV_OPEN=1,163,1
#EVDEV_CLOSE=1,163,0
#GPIO_SQL_PIN=gpio30
#PTY_PATH=/tmp/rx1_sql
#HID_DEVICE=/dev/hidraw3
#HID_SQL_PIN=VOL_UP
#SIGLEV_DET=TONE
#SIGLEV_SLOPE=1
SIGLEV_SLOPE=-255.56
#SIGLEV_OFFSET=0SIGLEV_OFFSET=984.79
#SIGLEV_BOGUS_THRESH=120
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
#SIGLEV_OPEN_THRESH=30
#SIGLEV_CLOSE_THRESH=10
DEEMPHASIS=0
SQL_TAIL_ELIM=500
#PREAMP=6
PEAK_METER=1
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=40
DTMF_SERIAL=/dev/ttyS0
#DTMF_PTY=/tmp/rx1_dtmf
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1
#SEL5_DEC_TYPE=INTERNAL
#SEL5_TYPE=ZVEI1
#FQ=433475000
#MODULATION=FM
#WBRX=WbRx1

sm0svx, 2017-01-31:
I have not been able to reproduce the problem this far. I'm beginning to wonder if this is connected to a specific hardware platform, the Raspberry Pi.  Which version of the Pi are you all using?

Does anyone using some other hardware, like a standard Intel PC, have this problem?

I have never seen this problem on standard Intel hardware. I'm using an ARM based Cubieboard for my own node and that have also never shown this problem so it's not an ARM specific issue.

There could be some subtle difference in configuration that trigger this bug but the RX configuration above look fairly standard.

hb9pae, 2017-02-05:
Hello all

As SM0SVX stated, I belive that this error is related to RPI Plattform only. I'm used many
other installations on x86 (Thin Clients), never seen any problem. 
 
 It seems that the configitem "SQL_DET=CTCSS" is important to fail, after changing to 
 "SQL_DET=Serial" the problem disapears. In both situations, the receiver audio-output
 was not squelched.  
 
 To produce the failure, we "stressed" the repeater  by sending many single DTMF-Signals
 so the Repeater Logic switched between open and closed. 2-3 minutes later, the failure 
 was reported. Sometimes the PTT-Signal (via serial pin RTS) stays on, causing a Carrier
 until restart of the program.

sm0svx, 2017-02-05:
Thanks for the additional information. Which generation of the RPi do you see this on?  Is it an older Pi or is it the latest?

pa2dk, 2017-02-05:
Sorry for my late answer. I didn't had time last days to work on the issue. We use the latest Raspberry Pi 3  here with a soundcard which has a cm108 chipset. I ordered a soundcard with a different type of chip, to see if that maybe causes the problem. It will take some weeks before it arrives, so i will come back on that in a month time. I will test the GPIO squelsh to see if it that not causes the problem and use that for now.

sm0svx, 2017-02-05:
OK, thanks. I have a Pi3 here that I will test to see if I can reproduce the problem.

pa2dk, 2017-02-05:
I used this soundcard: http://www.informatique.nl/536417/ewent-ew3751-usb-audioblaster.html Soundcards like this can be found at many brands

f5vmr, 2017-09-07:
F5ZGM-R has been running with SQL_DET=CTCSS for sometime, and appeared to have no problems, but about three weeks ago I recompiled the logic from the source, changing manually some those files that were not trapped for change on the make install (.so) (sudo git clone --branch svxreflector --single-branch https://github.com/sm0svx/svxlink.git being the source.)
Now the repeater was failing constantly, halting the svxlink program, but leaving the repeater in permanent transmit, immediately after a valid CTCSS detection and opening of the squelch.
So I changed the SQL_DET=GPIO and now there is no problem with the repeater. So is there an issue with the CTCSS detection that causes the svxlink to fall over? As we have some people who like to blip the repeater can we see a possible resolution soon so we can go back to CTCSS detection?

sm0svx, 2017-09-09:
If possible, recompile SvxLink using the cmake -DCMAKE_BUILD_TYPE=Debug parameter and then run it under "gdb". When it stops, type "bt" to get a backtrace.

n7ipb, 2017-10-22:
After the latest round of AsyncAudio patches I'm still seeing this assertion when using CTCSS decode.  I've duplicated it using both CTCSS mode 2 and 3.  It may be there in 1 but I haven't tested it.

It also appears to be happening only when the user stops talking.

 `svxlink: /home/svx_admin/Projects/svxlink/src/async/audio/AsyncAudioSource.cpp:168: int Async::AudioSource::sinkWriteSamples(const float*, int): Assertion `len > 0' failed.
Aborted`

sm0svx, 2017-10-30:
Try latest **master** branch. Look for printouts/log entries looking like: "### AudioProcessor::writeFromBuf: buf_cnt=..."

n7ipb, 2017-10-31:
I just built and tested the latest master branch.  

Attached is a log file showing the results of two consecutive runs of  svxlink with the receiver set for software CTCSS decode.   Both show the writeFromBuf message followed by a sequence that repeats until svxlink is killed.

`Oct 30 2017 18:07:07.204: RepeaterVoter: The squelch is OPEN (RxRepeaterCapPk=51.3636)
Oct 30 2017 18:07:08.266: ### AudioProcessor::writeFromBuf: buf_cnt=0 written=256
Oct 30 2017 18:07:08.271: *** WARNING: 256 samples dropped in squelch detector for receiver RxRepeaterCapPk
Oct 30 2017 18:07:08.271: *** WARNING: 85 samples dropped in squelch detector for receiver RxRepeaterCapPk
Oct 30 2017 18:07:08.293: *** WARNING: 256 samples dropped in squelch detector for receiver RxRepeaterCapPk
`

After the appropriate hang time delay the transmitter will drop and stop responding to receiver inputs and the repeating message continues until svxlink is killed.

[audioprocessor_bug.log](https://github.com/sm0svx/svxlink/files/1429269/audioprocessor_bug.log)

sm0svx, 2017-11-04:
Thanks for feedback. That will probably be helpful.

oh1sdr, 2017-11-06:
Hi!

Not necessarily helpful to this bug, but just wanted to let you know that I installed SVXlink a bit over week ago and have the exactly same issue with CTCSS. The HW is Raspberry Pi 3 model B, with Deltaco USB-dongle sound card (CM108 chipset I think). The error appears randomly, but always when the user stops talking, like already mentioned above.

73 de Janne OH1SDR

sm0svx, 2017-11-27:
Haven't found the cause for this yet but the issue has not been forgotten.

sm0svx, 2017-12-04:
There is hope! I have managed to reproduce the error on my RPi. I get the same behavior that @n7ipb described in his last comment.

sm0svx, 2017-12-04:
And now I can reproduce it on my (Intel) laptop so it's not a RPi only issue. If you want CTCSS really bad I think you may find that CTCSS_MODE=1 will work. That mode is not as good as the default mode though but maybe better than nothing.

sm0svx, 2017-12-18:
The problem should now be fixed in latest master

closed 2017-12-18 by sm0svx

n7ipb, 2017-12-19:
On Monday, December 18, 2017 11:44:41 AM PST Tobias Blomberg 
wrote:
> The problem should now be fixed in latest master

Built, tested and deployed on three systems last night and so far it 
looks very good.

-- 
Ken - N7IPB
Email: n7ipb@wetnet.net
JID: n7ipb@jabber.wetnet.net
PGP Sig: F42B EF90 3CD3 31C7 3056  122E 993A 7B2E 5138 C42A 
I've calculated my velocity with such exquisite precision that I have no 
idea where I am.