sm0svx/svxlink #210: Feature Request:  APRS Digipeater Module

W5BFF, 2016-09-25:
I'm running SVXLink on a Raspberry Pi 2, and it's working quite nicely.  I have it configured to connect to EchoLink and the APRS-IS services.

I have a TNC-Pi (from tnc-x.com) and was wondering how difficult it would be to add full RF APRS Digipeating services to SVXLink... particularly without running other APRS software on the Pi?  

This would allow RF auto-tune services for mobile stations with appropriately capable radios travelling through the repeater coverage area.  I believe it would also help expand the APRS RF coverage network.

W5BFF, 2016-09-29:
No response in 4 days. :/ 

I was looking for various APRS packages for the Raspberry Pi, and APRX looks pretty good to run headless like SVXLink... it doesn't look like it would take much to integrate APRX into SVXLink as some kind of expansion module.  Of course, I'm not a c++ developer, so I can't speak with any kind of real authority on the issue, but it doesn't SEEM like it would be that difficult.

sm3sgp, 2016-10-01:
I am running ARPX on the same coputer as svxlink, works nice. Do you have any example of auto-tune capable radios that can be controlled via APRS?

chakphanu, 2016-10-22:
Any infomation of this branch?
https://github.com/sm0svx/svxlink/tree/afsk_com/src/svxlink/digital

W5BFF, 2016-10-23:
I've not even gotten a reply from the author on my original post. :/

sm3sgp, 2017-03-08:
The afsk_com branch is used here at SK3GW repeater, but it has not much to do with APRS. It is an implementation to send data via radio, in our case, sending SIGLEV reports and other data from remotetrxs to the RepeaterLogic. There are some very good APRS applications for linux already available. What is the use case of having it built into svxlink?

W5BFF, 2017-03-08:
Several:  Repeater control codes sent via APRS, METAR data digipeated via
APRS, Repeater status (beyond just EchoLink connections) beaconed, etc....

...basically, a fully integrated APRS-RF digipeater.  I have the TNC-Pi hat
on my Raspi, and it would be a lot tidier if I could run it with the
SVXLink software instead of running a separate APRS server.

dl1hrc, 2017-03-10:
Hmm,I do not currently see the SvxLink function primarily in the APRS area. There are really better programs available with very good aprs data processing. On almost all of my SvxLink-repeaters I have aprx running, a very good combination I think. The additional cpu load is marginal.
But one application area could be the receipt of short APRS data bursts at the beginning/end of a voice transmission (mic_e) with GPS data on rf repeaters.

W5BFF, 2017-03-10:
What about gating APRS messages to/from the EchoLink Chat Box?  Yeah, that
would be cool, too.

pe1chl, 2017-04-04:
This kind of thing is usually possible already with the right "tricks" to wire things together, so I
am not sure you require specific support for it.
For example, we run remote receivers for our repeater using RTL SDR sticks, that have additional
receivers tuned to the APRS frequencies, the audio from that is sent via the ALSA loopback device
to locally running "direwolf" APRS digipeaters.  It allows an svxlink installation to run as an APRS
digipeater.  And I am sure that it would be possible to transfer info between the two as well.
(we don't do that yet)

sm3sgp, 2019-09-10:
For now, other solutions lite aprx, direwolf etc exist. Will close this issue.

closed 2019-09-10 by sm3sgp