sm0svx/svxlink #211: Module activation on startup and 2 modules at the same time

framar1, 2016-10-10:
would love to have it auto activate a module when it starts 
and would like to run the parrot and the metar at the same time 
maybe metar could announce weather on the long id string

dl1hrc, 2016-10-10:
> would love to have it auto activate a module when it starts
> and would like to run the parrot and the metar at the same time
> maybe metar could announce weather on the long id string

running two module at the same time is not possible, sri.

framar1, 2016-10-10:
what about auto activation of a module on start up IE:i run my node as a parrot repeater so i want parrot to run when the system starts without having to use dtmf to turn it on

dl1hrc, 2016-10-10:
sri, no idea at the moment

kb3vgw, 2016-12-06:
you add them to the svxlink.conf and then you have to use dtmf to activate the module. modules dont just activate .

sm3sgp, 2019-09-10:
Specific announcements at id should be possible to add in the TCL scripts.
Activation of two modules at the same time will require major changes.

closed 2019-09-10 by sm3sgp

hayden-t, 2022-07-12:
A very simple way to do this is to use the injectDtmf at the end of your logic tcls startup{} function , eg if you want to activate module number 1

injectDtmf 1#