sm0svx/svxlink #213: arch install

kgrant341, 2016-10-11:
ok got it to install using yaourt but wouldnt finsh after building
 error: failed to commit transaction (conflicting files)
svxlink: /usr/lib64 exists in filesystem
Errors occurred, no packages were upgraded.

any help on this to fix this error

kgrant341, 2016-10-11:
I don't see a issue in the  PCGBUILD  I think it's in the sorce arch doesn't use /usr/lib64 anymore  it uses /usr/lib

kgrant341, 2016-10-11:
PCGBUILD downloads from source I see that sm0svx wrote in the 15.11 release
 "Another thing that may be good to know is that the default install path for
libraries may have changed since it is now autodetected during the build. For
example, on a x86_64 architecture the libraries and plugins will now end up
under /usr/lib64 instead of /usr/lib. "
I see the issue is in the source any help on finding where it was changed i am running a x86_64 system

kgrant341, 2016-10-11:
I found this in the code
"The install path for libraries are now automatically discovered which may
  make libraries install in a another directory than before. For example, on
  normal 64 bit x86_64 systems they will end up in /usr/lib64 and the plugins
  will be placed in /usr/lib64/svxlink. On other platforms the paths may be a
  bit stranger, like /usr/lib/arm-linux-gnueabihf on the Cubieboard.
  This may require that the MODULE_PATH config variable is changed for the
  plugins to be found. The best is to comment out the MODULE_PATH
  configuration variable and let SvxLink determine what to use.
  The old LIB_SUFFIX CMake variable is not used anymore."

kgrant341, 2016-10-11:
this is where the issue begins after building 
(1/1) checking keys in keyring   100%
(1/1) checking package integrity   100%
(1/1) loading package files   100%
(1/1) checking for file conflicts 100%
error: failed to commit transaction (conflicting files)
svxlink: /usr/lib64 exists in filesystem
errors occurred, no packages were upgraded.

so i think im going to see if i can remove the files

kgrant341, 2016-10-13:
ok had to change stuff to get it to install
pacman -Qo /usr/lib64
then
sudo mv /user/lib64 /usr/lib64.backup 
then it would install

kb3vgw, 2016-12-06:
you should use the defaults and update your config file.

sm0svx, 2017-01-14:
If I understand you correctly this is a problem with the Arch package. You should report this problem to the Arch package maintainer so that they can fix the problem. I don't know if they are monitoring this issue tracker.

closed 2017-01-14 by sm0svx