sm0svx/svxlink #214: METARINFO - NO SUCH AIRPORT

IK1PJL, 2016-10-25:
Hi, I'm new sysop using SVXLINK. I used Echolink for many years before this new installation. The Svxlink is running over a Raspberry PI3 and it is installed inside the repeater cabinet. Everything works perfectly, less the METAR INFO Module. I 'm sure the setting is ok, but after Module activation by 5#, when a call for metar info ( in my module I setup 1, as LIMF, 2 as LIRC etc.. ) svxlink answear me "NO SUCH AIRPORT "... 
Any idea bout that?
Thanks!!!
73, IK1PJL

IK1PJL, 2016-10-31:
Thanks Adì! I'm very sorry for THE post. I saw the post you describe just after sent mine... Anyway thankyou so much for your informations and your attention.
Best 73, Andy, IK1PJL

closed 2016-10-31 by IK1PJL

sm0svx, 2017-01-14:
I cannot find a pull request for this one?

dl1hrc, 2017-01-14:
Hi Tobias,

very nice that you're back :)
I will merge it with your updates today and will update all my other
branches. Then I send a pull request, ok?

vy 73s de Adi / DL1HRC

sm0svx, 2017-01-14:
Thumbs up!