sm0svx/svxlink #215: Feature Request:

citynetcomputing, 2016-12-01:
Hello,

Looking to "Virtualize" the repeater by allowing each "RepeaterLogic" to have its own TX and RX CTCSS tones. This will allow the same repeater "RepeaterLogic1, RepeaterLogic2, RepeaterLogic3" etc to be used as multi-tennant with separate tones ie 123.0, 141.3, 82.0 respectively. Also the use of digital DCS tones. IE: D023N etc.

It looks like in the current branch you can set individual RX tones in the repeater logic but for the TX tone it is set in the TX section which wouldn't work. Haven't tested this.

Thanks!!

sm0svx, 2017-01-14:
I don't understand. Can you explain it in another way. I'm not familiar with the "multi-tennant" expression.

CTCSS squelch tone is set in the RX configuration section and TX CTCSS tone is set in the TX configuration section. In the RepeaterLogic configuration section you can set a tone that will activate the repeater. It does not have to be the same tone but it would be very confusing to have a different tone.

SvxLink do not support DCS so cannot be used unless you use some external detector.

f5vmr, 2017-01-15:
I see what he means, but it is spelt "tenant". To run two repeaters on the same version and instance of EchoLink means duplicating RepeaterLogic.tcl ( and renaming it) and duplicating Logic.tcl (and renaming it), and being very careful to track which of the RX and TX are used with each. And I don't think that will be the end of it, as the control of PTT and squelch will also have to be reconciled. Then there's the sound card situation. I can envisage a UHF and a VHF repeater cohabiting the same environment sharing the logic in this case.

sm0svx, 2017-01-16:
Now I'm even more confused. You wrote "EchoLink". Did you mean to write "SvxLink"?  The original post is mostly about CTCSS. Running multiple repeaters within SvxLink is no problem, either as totally separate repeaters or with possibility to link them together on demand or permanently.

f5vmr, 2017-01-16:
Yes I meant svxlink. A curse on my late night brain 😎

sm3sgp, 2019-09-10:
Not possible to understand this feature request. If needed open a new with a detailed explanation of the use case.

closed 2019-09-10 by sm3sgp