sm0svx/svxlink #218: SQL Continuous Triggering in QSO

tamilnaduhams, 2016-12-09:
Rx1: SQL continuous OPEN/CLOSE while receiving from repeater. and also continuous Tx1: ON/OFF while speaking from echolink. Please help me to correct this issue. Screenshot and syslink.conf are attached

[image]
[image]

[svxlink.conf.txt](https://github.com/sm0svx/svxlink/files/641286/svxlink.conf.txt)

dl1hrc, 2016-12-09:
I had this behaviour when AGC has been enabled on the usb soundcard. Does it happen when the repeater is down and/or during the transmission with no audio?
1st you should try to make it working with a normal microphone, increase the params SQL_START_DELAY and SQL_DELAY a bit (0 to 40-80).

BTW: please correct the default params LON_POSITION=12.10.00E and LAT_POSITION=51.10.00N when [LocationInfo] is enabled. At the moment your position is in DL, next to my village :)

vy 73s de Adi / DL1HRC

tamilnaduhams, 2016-12-09:
AGC is already muted, and also i tried with SQL_START_DELAY=10 and SQL_DELAY=100 but the breaking in audio increased. so i changed back to default.

LAT and LON details are corrected now.

dl1hrc, 2016-12-09:
increase the capture value a bit, maybe the audio input of your receiver is to low, so only the audio peaks trigger the vox. Hard to say what's wrong from the distance.
vy 73s de Adi / DL1HRC

tamilnaduhams, 2016-12-09:
yeah the volume in the receiver is nearly high enough. but when i noticed on QSO it will happend mostly when the SQL Open is below 6.000. and the SQL level in the receiver is 3.

is there anything i have to change in receiver?

dl1hrc, 2016-12-09:
The values in the log file, eg. "The squelch is OPEN (0.343094)" are only interesting if you're using the calibrated siglevel-detector. The SQL level in the receiver has nothing to do with that. Can you increase the capture level? Does it has an influence?
I think it's more important that your sound card feels that the audio level is high enough, you can not decide it by just listening to the speaker output.
vy 73s de Adi / DL1HRC

closed 2016-12-10 by tamilnaduhams