sm0svx/svxlink #221: Where is Tobias?

pa1okz, 2017-01-02:
Although this might not be the optimum location to put this, I still hope to see some renewed activity in the Svxlink development. It's a fantastic environment and is probably the No. #1 repeater controller software worldwide on which many HAM's rely on...

Now that not a single activity has beent visible since october 2015 I wonder what's going on. 

Dear Tobias, is all well? Are you giving up development? Would it be an idea to introduce a different development method, team oriented with strict rules?

73,

Mischa, PA1OKZ

Techinput, 2017-01-02:
Just a quick question Mischa, is SVXLINK a part of the PI2NOS infrastructure?
73's 
Chris, PA7U

pa1okz, 2017-01-02:
Hi Chris,

In fact, Svxlink is the basis for our repeater PI2NOS, though, we have
created a range of features and functions around it. Several of them are
described here as well...

73

Mischa

sm3sgp, 2017-01-02:
I talked to Tobias today, and he is well but has been busy with other things for some time. I don't think he will give up svxlink. Let's hope he find some more time soon, and maybe have some good ideas on how to continue the development.
73, Gunnar

pa1okz, 2017-01-04:
In the case that Tobias lacks time (who doesn't ;-) it might be a good idea to create a small team that collaborates on further improvements. As the project is owned by Tobias, it's up to him of course to decide how one thing could be organized and moreover, to what design rules and principles such a small team should stick. 

There is currently a tremendous list of feature requests and issues to work on and Svxlink pops up time and time again from different perspectives, such as https://openrepeater.com/ etc. In our continuously improving national repeatersystems, we also highly depend on Svxlink. Although we increasingly add our own features and functions, we primarily depend on the fantastic work that Tobias has done...

sm0svx, 2017-01-07:
Yes, I'm still here. I have not decided on how to go forward with the SvxLink project at the moment.
Are there any special issues that you think are more important than others?

pa1okz, 2017-01-08:
Hi Tobias, nice to see you being active here again :-). 

Sure there are issues that we would like to see being implemented, we already raised several and I think it is more up to you which ones should have priority over others. I can see that you are merging several additions now into the branches, there are some contributions from us as well. 

Our priority currently is with issue 132 and in addition, there is a high desire to see the CTCSS detector getting much faster. It is reasonably fast for a SQ open indication, but we run into the issue that the detector causes such a tremendous Squelch tail that we have to introduce about 350ms of squelchtail elimination to get rid of it, highly undesirable...

Hope you stay active Tobias, thank you!

sm0svx, 2017-01-14:
OK about #132. I'll see what I can do about that one.

The CTCSS issue I take as a challenge ;-)  I have some new ideas for how to improve it but there are a lot of compromises between quickness at open/close, sensitivity, false openings, handling flutter etc

closed 2017-01-14 by sm0svx